immortalwrt/immortalwrt #1394: immortalwrt 23.05.3 编译的R5S设备固件无法启动。

lysgwl, 2024-08-01:
### Describe the bug

自编译的r5s设备固件，设置SD卡启动，设备r5s无法启动，插上网线状态灯也不亮。同样编译的r2s设备固件，正常可以运行，r5s无法运行。

### ImmortalWrt version

r27917-81a1f98d5b

### ImmortalWrt release

23.05.3

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R%S

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_friendlyarm_nanopi-r2s=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_friendlyarm_nanopi-r5s=y
CONFIG_COREMARK_OPTIMIZE_O3=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
# CONFIG_LIBCURL_CRYPTO_AUTH is not set
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
# CONFIG_LIBCURL_OPENSSL is not set
# CONFIG_LIBCURL_TFTP is not set
# CONFIG_LIBCURL_THREADED_RESOLVER is not set
# CONFIG_LIBCURL_TLS_SRP is not set
CONFIG_LUCI_CSSTIDY=y
CONFIG_MBEDTLS_AES_C=y
CONFIG_MBEDTLS_ARMV8CE_AES_C=y
CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_DES_C=y
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ENTROPY_FORCE_SHA256=y
CONFIG_MBEDTLS_GCM_C=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_PSK_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
CONFIG_MBEDTLS_NIST_KW_C=y
CONFIG_MBEDTLS_RSA_NO_CRT=y
CONFIG_NGINX_HTTP_ACCESS=y
CONFIG_NGINX_HTTP_AUTH_BASIC=y
CONFIG_NGINX_HTTP_AUTOINDEX=y
CONFIG_NGINX_HTTP_BROWSER=y
CONFIG_NGINX_HTTP_CACHE=y
CONFIG_NGINX_HTTP_CHARSET=y
[... 363 lines cut ...]
CONFIG_PACKAGE_xz-utils=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zram-swap=y
CONFIG_PACKAGE_zsh=y
CONFIG_PACKAGE_zstd=y
CONFIG_PARTED_READLINE=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_TARGET_KERNEL_PARTSIZE=128
CONFIG_TARGET_ROOTFS_PARTSIZE=512
CONFIG_ZRAM_DEF_COMP_ZSTD=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_COREMARK_ENABLE_MULTITHREADING is not set
CONFIG_PACKAGE_kmod-fs-ntfs3=y
# CONFIG_PACKAGE_libgcrypt is not set
# CONFIG_PACKAGE_libgpg-error is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_btrfs_progs is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_ntfs_3g_utils is not set
# CONFIG_PACKAGE_ntfs-3g is not set
# CONFIG_PACKAGE_ntfs-3g-utils is not set
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-08-01:
No log = no solution.

lysgwl, 2024-08-04:
> No log = no solution.

编译日志，还是运行日志？有尝试了几次，依旧是不行，istoreos就没问题。

1715173329, 2024-08-04:
串口日志。你觉得 istoreos 没问题那去用即可。

closed 2024-08-05 by 1715173329

lysgwl, 2024-08-05:
> 如果确认tf卡没有问题 请上串口看看发生了什么，串口有很多种方式，包括但不限于：漏油器的ttl、开发板的ttl、usb2ttl等。

使用的卡刷，sd卡toemmc方式，启动friendlywrt官方固件没问题，说明tf卡不存在问题。至于串口日志，马上不知道怎么搞，新买的设备。开发工具连接需要usb对等线，需要等几天。还有更好方式获取日志吗？

lysgwl, 2024-08-05:
> 串口日志。你觉得 istoreos 没问题那去用即可。

这个是串口日志，麻烦大佬给看看吧。
Boot1 Release Time: Apr 14 2023 10:04:54, version: 1.17
support nand flash type: slc
...nandc_flash_init enter...
No.1 FLASH ID:ff ff ff ff ff ff
sfc nor id: ff ff ff
sfc_nand id: ff ff ff
SD IO init 0
Set SD Clk: 200, 5
Set SD Clk: 200, 5
 SDC_BusRequest:  CMD=8  SDC_RESP_TIMEOUT 1736
 mmc0:cmd8,20
 SDC_BusRequest:  CMD=5  SDC_RESP_TIMEOUT 1736
 mmc0:cmd5,20
 SDC_BusRequest:  CMD=55  SDC_RESP_TIMEOUT 1736
 mmc0:cmd55,20
 SDC_BusRequest:  CMD=1  SDC_RESP_TIMEOUT 1736
 mmc0:cmd1,20
Set SD Clk: 200, 5
 SDC_BusRequest:  CMD=8  SDC_RESP_TIMEOUT 1736
 mmc0:cmd8,20
 SDC_BusRequest:  CMD=5  SDC_RESP_TIMEOUT 1736
 mmc0:cmd5,20
 SDC_BusRequest:  CMD=55  SDC_RESP_TIMEOUT 1736
 mmc0:cmd55,20
 SDC_BusRequest:  CMD=1  SDC_RESP_TIMEOUT 1736
 mmc0:cmd1,20
Set SD Clk: 200, 5
 SDC_BusRequest:  CMD=8  SDC_RESP_TIMEOUT 1736
 mmc0:cmd8,20
 SDC_BusRequest:  CMD=5  SDC_RESP_TIMEOUT 1736
 mmc0:cmd5,20
 SDC_BusRequest:  CMD=55  SDC_RESP_TIMEOUT 1736
 mmc0:cmd55,20
 SDC_BusRequest:  CMD=1  SDC_RESP_TIMEOUT 1736
 mmc0:cmd1,20
SdmmcInit=0 1
Emmc IO init.
EMMC_DLL_RXCLK = 0
Emmc IO init.
EMMC_DLL_RXCLK = 0
mmc_set_bus_width: 1
SetEmmcClk: 375000, 5
mmc_switch index:183, value:0x2
mmc_switch index:185, value:0x2
SetEmmcClk: 375000, 5
mmc_set_bus_width: 8
SetEmmcClk: 200000000, 1
Enable PHY CLK: 200000000
SdmmcInit=2 0
BootCapSize=100000
UserCapSize=29600MB
FwPartOffset=2000 , 100000
UsbBoot ...66891
powerOn 68813