immortalwrt/immortalwrt #1410: 主(拨号)、旁路由均是op时，网关/dns互相指向对方会无法科学上网

kylongteng, 2024-08-10:
### Describe the bug

主(拨号)、旁路由均是op时，网关/dns互相指向对方会无法科学上网。

### ImmortalWrt version

r27946-868b12b200

### ImmortalWrt release

23.05-SNAPSHOT

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Steps to reproduce

设备：ax6s，pve8.2.4虚拟x86(ImmortalWrt 23.05-SNAPSHOT r27946-868b12b200 / LuCI openwrt-23.05 branch git-24.222.22929-9284585)
主、旁网关互相指向对方，旁路由开passwall。测试过以下情形：

主路由开dhcp，旁路由关dhcp
主旁均关闭软硬件流量分载，防火墙均允许入站、出站、转发
每次测试配置好后所有路由器全部重启以排除干扰，旁路由passwall本机测试百度和google都能正常访问。

1、主:ax6s-op，旁:x86-op 不可科学上网
2、主:x86-op，旁:ax6s-op 不可
此时，我印象中以前主路由用ax6s原装系统小米时这样配置是没问题的，所以将ax6s刷回原系统测试
3、主:ax6s-小米, 旁:x86-op 可
4、主:x86-ikuai，旁:ax6s-op 可

不可时，大陆访问正常，只是无法科学上网，请求google时passwall有正常日志，类似这样：2024/08/10 05:25:25 192.168.1.2:57272 accepted tcp:1.1.1.1:53 [tcp_redir -> default]

结论：主路由是硬路由原系统或ikuai时，主、旁网关和dns互相指向对方，大陆和海外访问都是正常的。但只要主路由是op就不行。

起因:昨天使用keepalived虚拟了vip: 192.168.1.1（旁(master，优先级100)/主(backup，优先级50)）作为网关配置在主路由的上游然后出现此问题，接着关闭keepalived，手动将真实的旁路由ip配置到主路由的网关和dns然后重启，问题依旧，折腾了一下午无果。后来还测试了其它几个op分支，效果是一样的：不可。说明这个现象不是ImmortalWrt独有的。

猜测:是不是op的路由策略不允许这样配置？

### Diffconfig

```text
ax6s
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head

x86/64
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

closed 2024-08-10 by 1715173329

kylongteng, 2024-08-10:
> 谁教你这么配的就去问谁。

有人请教到你这里了，在你的专业领域，这个问题也许很蠢；
可以选择回答或者不回答的，删了都行。
我不清楚你这“谁教你这么配的就去问谁”这短短几个字里到底有没有包含傲慢的态度、嘲讽的情绪等贬义的意味
但人生在世，总会进入别人擅长的领域，大概率会遭遇同样的事情

nicholas-opensource, 2024-08-10:
对不起，在这个领域，菜是原罪，你菜你活该。人生道理说给弱者罢了

kylongteng, 2024-08-10:
> 对不起，在这个领域，菜是原罪，你菜你活该。人生道理说给弱者罢了

嘿，在乎才会上心
已取消订阅，我也没其它情绪。再也不见

nicholas-opensource, 2024-08-10:
嗯，祝您一辈子都这么菜