immortalwrt/immortalwrt #1414: rp-cp-03 最新23.05.3分支源代码编译,手动修改断言时间，gpt设置根分区2G,出现下面两个问题

aYangQT, 2024-08-11:
### Describe the bug

大佬您好，我在编译的imm 23..4.3最新固件有两个问题：
1，使用默认源安装应用时，提示内核MD5不匹配，修改md5配置文件就可以了，不过感觉比较繁琐，每次刷固件都要手动改
2，通过书签直接进入实时信息-连接页面，或者登录web后快速打开这个页面，会导致这个页面卡在正在收集中，不能正常显示连接信息，稍后会进入输入密码页面，输入密码也是一样
3，zerotier好像不能用（这个好像是tun模块不能用，恢复出厂了，日志里面可能没有）

### ImmortalWrt version

r27917-81a1f98d5b

### ImmortalWrt release

23.05.3

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

 JDCloud RE-CP-03

### Image kind

Official downloaded image

### Steps to reproduce

rp-cp-03 最新23.05.3分支源代码编译,手动修改断言时间，gpt设置根分区2G,

### Actual behaviour

1，使用默认源安装应用时，提示内核MD5不匹配，修改md5配置文件就可以了，不过感觉比较繁琐，每次刷固件都要手动改
2，通过书签直接进入实时信息-连接页面，或者登录web后快速打开这个页面，会导致这个页面卡在正在收集中，不能正常显示连接信息，稍后会进入输入密码页面，输入密码也是一样
3，zerotier好像不能用（这个好像是tun模块不能用，恢复出厂了，日志里面可能没有）

### Expected behaviour

1，不知道怎么在编译阶段修改md5和默认源匹配，每次需要手动修改/usr/lib/opkg/status中的值
2，实时信息-连接页面可以正常打开，官方op的昨天snapshot版本也是不能正常显示

### Additional info

这里是日志

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

aYangQT, 2024-08-11:
[内核日志_240811.txt](https://github.com/user-attachments/files/16571417/_240811.txt)
[系统日志_240811.txt](https://github.com/user-attachments/files/16571418/_240811.txt)

aYangQT, 2024-08-11:
[image]
[image]
[image]

aYangQT, 2024-08-11:
在实时信息-连接界面，停留一段时间好像也会这样，让输入密码，进入还是这个界面，如果这时候点击首页，会很难打开，然后刷新打开报红（第三张截图），官方op 昨天下载的snapshot，好像输入正确密码也进不去，imm可以正常进去，但是连接那里还是不能正常显示，imm刚开始可以正常显示，过一会儿好像就不行了

1715173329, 2024-08-11:
自己编译固件不能安装内核模块。
大连接数会卡崩 rpcd。

closed 2024-08-11 by 1715173329

aYangQT, 2024-08-11:
> 自己编译固件不能安装内核模块。 大连接数会卡崩 rpcd。

好的，感谢大佬解答，我这边临时先用着，等imm 23.05.4稳定版发布应该可以直接使用官方镜像就可以了