immortalwrt/immortalwrt #1423: 最新源码编译内核时报错

iwowcn, 2024-08-16:
### Describe the bug

  CC [M]  drivers/usb/storage/shuttle_usbat.o
  LD [M]  drivers/usb/storage/ums-alauda.o
  LD [M]  drivers/usb/storage/ums-cypress.o
  LD [M]  drivers/usb/storage/ums-datafab.o
  LD [M]  drivers/usb/storage/ums-freecom.o
  AR      drivers/usb/storage/built-in.a
  LD [M]  drivers/usb/storage/ums-isd200.o
  LD [M]  drivers/usb/storage/ums-karma.o
  LD [M]  drivers/usb/storage/ums-jumpshot.o
  LD [M]  drivers/usb/storage/ums-sddr55.o
  LD [M]  drivers/usb/storage/ums-sddr09.o
  LD [M]  drivers/usb/storage/ums-usbat.o
  AR      drivers/usb/built-in.a
make[7]: *** [scripts/Makefile.build:480: drivers] Error 2
make[6]: *** [/home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/Makefile:1921: .] Error 2
make[5]: *** [Makefile:234: __sub-make] Error 2
make[5]: Leaving directory '/home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45'
make[4]: *** [Makefile:28: /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.modules] Error 2
make[4]: Leaving directory '/home/lede/immortalwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/lede/immortalwrt/target/linux'
time: target/linux/compile#80.16#7.89#12.40
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lede/immortalwrt'
make[1]: *** [target/Makefile:25: /home/lede/immortalwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lede/immortalwrt'
make: *** [/home/lede/immortalwrt/include/toplevel.mk:248: world] Error 2

### ImmortalWrt version

288bb4faccf82671918f85a912c4db1e221408a3

### ImmortalWrt release

288bb4faccf82671918f85a912c4db1e221408a3

### ImmortalWrt target/subtarget

ImmortalWrt v23.05.3 Release

### Device

x86_64

### Image kind

Official downloaded image

### Actual behaviour

Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
在以上环境中编译内核时报错，预编译默认6.6.45内核固件。有科学上网环境

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-08-16:
使用多线程编译且日志不全，没有 diffconfig，无法算命。

iwowcn, 2024-08-16:
> 使用多线程编译且日志不全，没有 diffconfig，无法算命。

我试试单线程 看看报什么错 折腾快两天了~~

iwowcn, 2024-08-16:
大佬给瞅瞅~~
lede@openwrt:~/immortalwrt$ make V=s -j1
make[2]: Entering directory '/home/lede/immortalwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/lede/immortalwrt/scripts/config'
make[1]: Entering directory '/home/lede/immortalwrt'
make[2]: Entering directory '/home/lede/immortalwrt'
+ mkdir -p /home/lede/immortalwrt/staging_dir/target-x86_64_musl
+ cd /home/lede/immortalwrt/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/lede/immortalwrt/build_dir/target-x86_64_musl/stamp
touch /home/lede/immortalwrt/staging_dir/target-x86_64_musl/.prepared
make[3]: Entering directory '/home/lede/immortalwrt/target/linux'
make[4]: Entering directory '/home/lede/immortalwrt/target/linux/x86'
rm -f /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/localversion
/home/lede/immortalwrt/scripts/kconfig.pl  + + /home/lede/immortalwrt/target/linux/generic/config-6.6 /home/lede/immortalwrt/target/linux/x86/config-6.6 /home/lede/immortalwrt/target/linux/x86/64/config-6.6 > /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/lede/immortalwrt/.config >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target      
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target        
/home/lede/immortalwrt/scripts/package-metadata.pl kconfig /home/lede/immortalwrt/tmp/.packageinfo /home/lede/immortalwrt/.config 6.6 > /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.override
/home/lede/immortalwrt/scripts/kconfig.pl 'm+' '+' /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.target /dev/null /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.override > /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set
mv /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.old
grep -v INITRAMFS /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.old > /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set      
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set
rm -rf /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.prev || { cp /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config; cp /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.prev; }     
export MAKEFLAGS= ; [ -d /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/user_headers ] || make -C /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45 KCFLAGS="-fmacro-prefix-map=/home/lede/immortalwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/lede/immortalwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Aug 15 07:55:11 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/lede/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  CC="x86_64-openwrt-linux-musl-gcc" KERNELRELEASE=6.6.45  INSTALL_HDR_PATH=/home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/user_headers headers_install
grep '=[ym]' /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.config.set | LC_ALL=C sort | /home/lede/immortalwrt/staging_dir/host/bin/mkhash md5 > /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.vermagic
touch /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/.configured
rm -f /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/vmlinux /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45/System.map
make -C /home/lede/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.45 KCFLAGS="-fmacro-prefix-map=/home/lede/immortalwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/ [... cut ...]
make[3]: Leaving directory '/home/lede/immortalwrt/target/linux'
time: target/linux/compile#2.51#0.67#3.15
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lede/immortalwrt'
make[1]: *** [target/Makefile:25: /home/lede/immortalwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lede/immortalwrt'
make: *** [/home/lede/immortalwrt/include/toplevel.mk:248: world] Error 2
lede@openwrt:~/immortalwrt$

805828526, 2024-08-17:
一样，我也报着这个错
WARNING: The `change-id` is missing in the `config.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 125535` at the top of `config.toml`
NOTE: this message was printed twice to make it more likely to be seen
Build completed successfully in 0:25:38
touch /home/liuran/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/.built
( cd /home/liuran/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/build/dist ; for targz in *.tar.gz; do /home/liuran/openwrt/staging_dir/host/bin/libdeflate-gzip -dc "$targz" | tar -xf - ; done ; find . -mindepth 2 -maxdepth 2 -type f -name install.sh -execdir bash '{}' --prefix=/home/liuran/openwrt/staging_dir/target-x86_64_musl/host --disable-ldconfig \; ; )
install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'build-manifest'

    build manifest installed.

install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'cargo'

    cargo installed.

install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-src'

    rust src installed.

install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-std-x86_64-unknown-linux-gnu'

    rust std installed.

install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-std-x86_64-unknown-linux-musl'

    rust std installed.

install: creating uninstall script at /home/liuran/openwrt/staging_dir/target-x86_64_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rustc'

    rustc installed.

mkdir -p /home/liuran/openwrt/staging_dir/hostpkg/stamp
touch /home/liuran/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/.built
touch /home/liuran/openwrt/staging_dir/hostpkg/stamp/.rust_installed 
make[3]: Leaving directory '/home/liuran/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#6831.18#1062.60#2079.68
make[2]: Leaving directory '/home/liuran/openwrt'
make[1]: *** [package/Makefile:171: /home/liuran/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/liuran/openwrt'
make: *** [/home/liuran/openwrt/include/toplevel.mk:248: world] Error 2

1715173329, 2024-08-17:
你获取到的内核源码包或文件系统等有问题，函数mpt**S**as_del_end_device 应为全小写。
运行 make distclean 后重新编译即可。

closed 2024-08-17 by 1715173329