immortalwrt/immortalwrt #1425: J4125小主机卡Waiting for root device PARTUUID=xxx.....无法正常引导

asimovo, 2024-08-18:
### Describe the bug

cpu为J4125畅网小主机，用u盘烧录镜像后卡在

Waiting for root device PARTUUID=082c9cb8-4106-95d3-6cf9-b48d2a2b7002.....
[   60.534953] i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
[   60.536048] i915 0000:00:02.0: DMC firmware homepage: [https://git.kernel.org/pub/scm/l ... mware.git/tree/i915](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915)

无法进入系统

### ImmortalWrt version

v23.05.3-SNAPSHOT

### ImmortalWrt release

immortalwrt-23.05.3-x86-64-generic-squashfs-combined-efi.img.gz

### ImmortalWrt target/subtarget

X86/64

### Device

J4125

### Image kind

Self-built image

### Steps to reproduce

使用Rufus烧录immortalwrt-23.05.3-x86-64-generic-squashfs-combined-efi.img.gz到U盘内，开机引导卡住

### Actual behaviour

一直卡住，回车等操作无效，插入键盘有提示

### Expected behaviour

正常引导进入系统

### Additional info

[     9.429377] Waiting for root device PARTUUID=082c9cb8-4106-95d3-6cf9-b48d2a2b7002.....
[   68.215297] i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
[   68.249627] i915 0000:00:02.0: DMC firmware homepage: [https://git.kernel.org/pub/scm/l ... mware.git/tree/i915](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915)

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

asimovo, 2024-08-18:
[image]

1715173329, 2024-08-18:
u 盘有各种奇妙兼容问题，建议换个正常硬盘。

asimovo, 2024-08-18:
> u 盘有各种奇妙兼容问题，建议换个正常硬盘。

j4125使用msata硬盘，手头上没有，更换另一个u盘也不行，但是这几个u盘刷istoreos却都能正常引导

zorrose82, 2024-09-12:
编译lede也出现这个问题，内核改回6.1成功进入了，你也试试

1715173329, 2025-07-04:
fixed by 4a18bb1056c78e1224ae3444f5862f6265f9d91c

closed 2025-07-04 by 1715173329