immortalwrt/immortalwrt #1426: /etc/resolv.conf配置文件有误

bmzk00, 2024-08-18:
### Describe the bug

试了多个版本，均存在以下问题：
/etc/resolv.conf的配置文件中servername:127.0.0.11有误，应该为127.0.0.1。
手动修改后，只要重启又会被重置回11，希望作者能够修复。谢谢！

### ImmortalWrt version

23.05

### ImmortalWrt release

23.05

### ImmortalWrt target/subtarget

23.05

### Device

x86

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-08-18:
immortalwrt 不存在此行为，应检查你安装的软件包。

closed 2024-08-18 by 1715173329

xlighting2017, 2024-08-18:
盲猜玩docker玩的， 去掉就好了

bmzk00, 2024-08-18:
> 盲猜玩docker玩的， 去掉就好了

刚查了资料，应该是docker搞出来的。我每次手动修改后，只要重启就会被重置，请问怎么去掉？谢谢！

bmzk00, 2024-08-18:
> 盲猜玩docker玩的， 去掉就好了

暂时用了个变通的办法，就是增加开机执行修改命令，把resolv.conf中的内容修改了