immortalwrt/immortalwrt #1427: 树莓派4b 用最新的master源码编译，没有 无线 （wireless）选项

rhyolithe, 2024-08-18:
### Describe the bug

树莓派 raspberry 4b，master分支代码，同样代码编译，asus acrh17 所有功能正常，用23.05分支，树莓派，acrh17所有功能正常，只有master 分支给树莓派编译的没有 无线， 我看驱动是有的，是源码里还没添加，还是怎么了呢？

### ImmortalWrt version

r0-288bb4f

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

bcm27xx/bcm271

### Device

Raspberry Pi 4 Model B Rev 1.1

### Image kind

Self-built image

### Additional info

[image]

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

rhyolithe, 2024-08-18:
我看无线芯片cypress43455 驱动是有的，是源码里还没添加界面代码，还是怎么了呢？

1715173329, 2024-08-19:
没日志没 diffconfig，不如上街找人算命。

rhyolithe, 2024-08-19:
抱歉， 第一次弄这个，不太清楚要给什么信息，diffconfig以那个做对比我不清楚，我先补上完整的config

rhyolithe, 2024-08-19:
树莓派的 dmesg 信息：
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.45 (itre@itre-W) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-288bb4f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Thu Aug 15 07:55:11 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000002ac00000..0x000000002ebfffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef643e0..0x000000003ef647cf (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 18 pages/cpu s36264 r8192 d29272 u73728
[    0.000000] pcpu-alloc: s36264 r8192 d29272 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:47:B1:87 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3826164K/4050944K available (8256K kernel code, 920K rwdata, 2336K rodata, 448K init, 539K bss, 159244K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000209] Console: colour dummy device 80x25
[    0.000222] printk: console [tty1] enabled
[    0.000620] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000647] pid_max: default: 32768 minimum: 301
[    0.001044] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001109] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001691] cgroup: Disabling memory control group subsystem
[    0.002917] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.003135] rcu: Hierarchical SRCU implementation.
[    0.003151] rcu: 	Max phase no-delay instances is 1000.
[    0.003829] smp: Bringing up secondary CPUs ...
[    0.004286] Detected PIPT I-cache on CPU1
[    0.004392] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.004890] Detected PIPT I-cache on CPU2
[    0.004970] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.005447] Detected PIPT I-cache on CPU3
[    0.005529] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.005615] smp: Brought up 1 node, 4 CPUs
[    0.005680] SMP: Total of 4 processors activated.
[    0.005694] CPU features: detected: 32-bit EL0 Support
[    0.005708] CPU features: detected: CRC32 instructions
[    0.005790] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[... 322 lines cut ...]
[   10.552583] EXT4-fs (sda1): mounted filesystem 7faa5798-ddc5-4aa0-8e24-5fac795ae618 r/w with ordered data mode. Quota mode: disabled.
[   11.938910] EXT4-fs error (device sda1): ext4_validate_block_bitmap:421: comm ext4lazyinit: bg 18: bad block bitmap checksum
[... cut ...]
[   20.119935] EXT4-fs error (device sda1): ext4_validate_block_bitmap:421: comm ext4lazyinit: bg 12672: bad block bitmap checksum
[   20.929970] EXT4-fs error (device sda1): ext4_validate_block_bitmap:421: comm ext4lazyinit: bg 13984: bad block bitmap checksum
[  315.280774] EXT4-fs (sda1): error count since last fsck: 1849
[  315.286577] EXT4-fs (sda1): initial error at time 1712228790: ext4_validate_block_bitmap:421
[  315.295039] EXT4-fs (sda1): last error at time 1723836387: ext4_validate_block_bitmap:421

rhyolithe, 2024-08-19:
dmesg | grep brcmfmac
[    8.464353] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.473452] usbcore: registered new interface driver brcmfmac
[    8.501165] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[    8.510986] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.txt
[   12.481612] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

rhyolithe, 2024-08-19:
问题解决了，在 /lib/firmware/brcm 文件夹下没有brcmfmac43455-sdio.txt 文件，我把 brcmfmac43455-sdio.raspberrypi,4-model-b.txt 拷贝一份命名为brcmfmac43455-sdio.txt ，重启后无线就好了。
是源码里和驱动文件名没有匹配好的问题吗？

rhyolithe, 2024-08-19:
编译时还有一个 缺少依赖 “libpcre” 的提醒：
在 feeds/telephony/net/freeswitch/Makefile 里改成下面的 “libpcre2” 就好了
@@ -320,7 +320,7 @@
 	  +libcurl \
 	  +libedit \
 	  +libopenssl \
-	  +libpcre \
+	  +libpcre2 \
 	  +libpthread \
 	  +librt \
 	  +libspandsp3 \

还不太清楚怎么提交这种情况，暂时在这里说一下

1715173329, 2024-08-31:
Try revert 87a65ed282ae2052e0942637d280352001d53c87

closed 2025-04-19 by 1715173329