immortalwrt/immortalwrt #1428: 红米AX6000采用最新代码5G信号问题

UR-l, 2024-08-19:
### Describe the bug

红米AX6000采用最新代码5Gwifi无法被其他设备搜索到，有时会显示未关联，有时会正常，但是其他设备搜索不到5g信号

### ImmortalWrt version

r31323-288bb4facc

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000

### Image kind

Self-built image

### Expected behaviour

可以正常打开并被其他设备搜索到5G信号

### Additional info

/etc/config/wireless如下:

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option band '2g'
	option channel 'auto'
	option htmode 'HE40'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'UR'
	option encryption 'psk-mixed'
	option key '123456789'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option band '5g'
	option channel '36'
	option htmode 'HE160'
	option cell_density '0'
	option txpower '10'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'UR-5G'
	option encryption 'psk-mixed'
	option key '123456789'

系统日志如下:
Mon Aug 19 16:12:26 2024 daemon.notice hostapd: Set new config for phy phy1:
Mon Aug 19 16:12:26 2024 daemon.notice hostapd: Remove interface 'phy1'
Mon Aug 19 16:12:26 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DISABLED
Mon Aug 19 16:12:26 2024 daemon.notice hostapd: phy1-ap0: AP-DISABLED
[... 170 lines cut ...]
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_socat=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wifi-scripts=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_zlib=y
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-08-19:
没设置国家代码，并且 dfs 信道可能存在冲突。

UR-l, 2024-08-19:
1

UR-l, 2024-08-19:
我知道为什么了

UR-l, 2024-08-19:
我知道为什么了，重置之后他默认通道宽度是80Mhz，切换成160Mhz就不行了

1715173329, 2024-08-20:
必须设置**有效**国家代码。

closed 2024-08-20 by 1715173329