immortalwrt/immortalwrt #1429: uhttpd服务自动停止

iwowcn, 2024-08-19:
### Describe the bug

uhttpd服务在启动后大概2分钟左右就自动停止，无法访问web服务。

### ImmortalWrt version

r31307-07e476cac8

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Self-built image

### Actual behaviour

Intel(R) Pentium(R) Silver N6005 @ 2.00GHz x 4C 4T，16G内存，x86架构软路由，物理机实装，设备启动后登录进去，大概2分钟左右uhttpd服务就自动停止，ssh进去，手动启动服务，也会在大概2分钟后停止。添加了https证书。

### Expected behaviour

找到故障点，让uhttpd服务恢复正常，可以正常使用web界面，不在自动停止。

### Additional info

Tue Aug 20 05:59:29 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 78 seconds since last crash
Tue Aug 20 06:02:38 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:02:49 2024 authpriv.info dropbear[14072]: Child connection from 192.168.10.200:62664
Tue Aug 20 06:02:54 2024 authpriv.notice dropbear[14072]: Password auth succeeded for 'root' from 192.168.10.200:62664
Tue Aug 20 06:04:31 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:06:07 2024 kern.info kernel: [ 1048.612845] uhttpd[14381]: segfault at 7f1be64f8dd0 ip 00007f1be677f5c0 sp 00007ffe8244b148 error 4 in libcrypto.so.3[7f1be6741000+26a000] likely on CPU 0 (core 0, socket 0)
Tue Aug 20 06:06:07 2024 kern.info kernel: [ 1048.628485] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:06:07 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 7 crashes, 56 seconds since last crash
Tue Aug 20 06:09:50 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:10:09 2024 daemon.err uhttpd[15160]: [info] luci: accepted login on / for root from 192.168.10.200
Tue Aug 20 06:10:36 2024 kern.info kernel: [ 1317.510334] uhttpd[15160]: segfault at 7fb0a55a1050 ip 00007fb0a58205c0 sp 00007fff4df6eff8 error 4 in libcrypto.so.3[7fb0a57e2000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 20 06:10:36 2024 kern.info kernel: [ 1317.525934] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:10:36 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 8 crashes, 37 seconds since last crash
Tue Aug 20 06:11:13 2024 cron.err crond[16799]: crond (busybox 1.36.1) started, log level 5
Tue Aug 20 06:13:27 2024 kern.info kernel: [ 1488.073151] uhttpd[15865]: segfault at 7f37b9181830 ip 00007f37b94025c0 sp 00007ffd54239dd8 error 4 in libcrypto.so.3[7f37b93c4000+26a000] likely on CPU 1 (core 1, socket 0)
Tue Aug 20 06:13:27 2024 kern.info kernel: [ 1488.088834] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:13:27 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 9 crashes, 153 seconds since last crash
Tue Aug 20 06:14:23 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:21:49 2024 kern.info kernel: [ 1990.199453] uhttpd[19035]: segfault at 7fc8e06a95f0 ip 00007fc8e09185c0 sp 00007ffef1d6e268 error 4 in libcrypto.so.3[7fc8e08da000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 20 06:21:49 2024 kern.info kernel: [ 1990.215082] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:21:49 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 10 crashes, 422 seconds since last crash
Tue Aug 20 06:24:09 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:25:10 2024 kern.info kernel: [ 2190.848744] uhttpd[22508]: segfault at 7f3083a837a0 ip 00007f3083c9e5c0 sp 00007ffcb16fa488 error 4 in libcrypto.so.3[7f3083c60000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 20 06:25:10 2024 kern.info kernel: [ 2190.864341] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:25:10 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 11 crashes, 166 seconds since last crash
Tue Aug 20 06:29:23 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Tue Aug 20 06:32:17 2024 kern.info kernel: [ 2618.019262] uhttpd[26282]: segfault at 7f65128ceb00 ip 00007f6512b005c0 sp 00007ffec6927548 error 4 in libcrypto.so.3[7f6512ac2000+26a000] likely on CPU 1 (core 1, socket 0)
Tue Aug 20 06:32:17 2024 kern.info kernel: [ 2618.034891] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:32:17 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 12 crashes, 124 seconds since last crash
Tue Aug 20 06:33:09 2024 kern.info kernel: [ 2670.304021] uhttpd[28545]: segfault at 7ff6cbcc9170 ip 00007ff6cbf385c0 sp 00007fff6d239bd8 error 4 in libcrypto.so.3[7ff6cbefa000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 20 06:33:09 2024 kern.info kernel: [ 2670.319608] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:33:09 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 13 crashes, 34 seconds since last crash
Tue Aug 20 06:35:22 2024 kern.info kernel: [ 2803.314138] uhttpd[28961]: segfault at 7fea60ebc320 ip 00007fea611355c0 sp 00007ffc85a1a328 error 4 in libcrypto.so.3[7fea610f7000+26a000] likely on CPU 0 (core 0, socket 0)
Tue Aug 20 06:35:22 2024 kern.info kernel: [ 2803.329742] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Tue Aug 20 06:35:22 2024 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 14 crashes, 48 seconds since last crash
Tue Aug 20 06:37:26 2024 daemon.warn odhcpd[2533]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!

### Diffconfig

```text
相比自建镜像默认配置，增加了如下包：
luci-i18n-firewall-zh-cn
luci-theme-argon
luci-app-accesscontrol
luci-i18n-accesscontrol-zh-cn
luci-app-openclash
luci-app-ddns-go
luci-i18n-ddns-go-zh-cn
luci-app-autoreboot
luci-i18n-autoreboot-zh-cn
luci-app-arpbind
luci-i18n-arpbind-zh-cn
luci-app-vlmcsd
luci-i18n-vlmcsd-zh-cn
luci-app-wol
luci-i18n-wol-zh-cn
luci-app-attendedsysupgrade
luci-i18n-attendedsysupgrade-zh-cn
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2024-08-20:
添加了https证书。
--就是这个导致的， 盲猜去掉就好了

需要提供  你是如何生成证书，具体证书例子，对应的uhttpd配置及操作步骤，编译时 选中了那些包，去掉了那些包 （这些在你现在提供的信息中完全没有,特别是 ssl 相关的 package）

iwowcn, 2024-08-20:
> 添加了https证书。 --就是这个导致的， 盲猜去掉就好了
> 
> 需要提供 你是如何生成证书，具体证书例子，对应的uhttpd配置及操作步骤，编译时 选中了那些包，去掉了那些包 （这些在你现在提供的信息中完全没有,特别是 ssl 相关的 package）

1、证书是购买的Sectigo单域名证书，将Apache格式的crt和key证书上传到/etc目录，0755权限，重启默认uhttpd服务。没有额外安装过web服务包和ssl软件包。
2、package插件包情况如下：
**********Generic x86/64 SNAPSHOT默认预装软件包**********
alsa-utils
automount
base-files
busybox
ca-bundle
dnsmasq-full
dropbear
fdisk
firewall4
fstools
grub2-bios-setup
ipv6helper
kmod-8139cp
kmod-8139too
kmod-ac97
kmod-amazon-ena
kmod-amd-xgbe
kmod-bnx2
kmod-button-hotplug
kmod-dwmac-intel
kmod-e1000
kmod-e1000e
kmod-forcedeth
kmod-fs-f2fs
kmod-fs-vfat
kmod-i40e
kmod-igb
kmod-igbvf
kmod-igc
kmod-ixgbe
kmod-ixgbevf
kmod-nft-offload
kmod-pcnet32
kmod-r8101
kmod-r8125
kmod-r8126
kmod-r8168
kmod-sound-hda-codec-hdmi
kmod-sound-hda-codec-realtek
kmod-sound-hda-codec-via
kmod-sound-hda-core
kmod-sound-hda-intel
kmod-sound-i8x0
kmod-sound-via82xx
kmod-tg3
kmod-tulip
kmod-usb-audio
kmod-usb-hid
kmod-usb-net
kmod-usb-net-asix
kmod-usb-net-asix-ax88179
kmod-usb-net-rtl8150
kmod-usb-net-rtl8152-vendor
kmod-vmxnet3
libc
libgcc
libustream-openssl
logd
mkf2fs
mtd
netifd
nftables
opkg
partx-utils
ppp
ppp-mod-pppoe
procd
procd-seccomp
procd-ujail
uci
uclient-fetch
urandom-seed
urngd
**********Generic x86/64 SNAPSHOT自定义构建加装软件包**********
autocore
block-mount
default-settings-chn
kmod-nf-nathelper
kmod-nf-nathelper-extra
luci
luci-app-opkg
luci-i18n-opkg-zh-cn
luci-compat
luci-lib-base
luci-lib-ipkg
luci-i18n-firewall-zh-cn
luci-theme-argon
luci-app-accesscontrol
luci-i18n-accesscontrol-zh-cn
luci-app-openclash
luci-app-ddns-go
luci-i18n-ddns-go-zh-cn
luci-app-autoreboot
luci-i18n-autoreboot-zh-cn
luci-app-arpbind
luci-i18n-arpbind-zh-cn
luci-app-vlmcsd
luci-i18n-vlmcsd-zh-cn
luci-app-wol
luci-i18n-wol-zh-cn
luci-app-attendedsysupgrade
luci-i18n-attendedsysupgrade-zh-cn
**********Generic x86/64 SNAPSHOT web界面安装软件包**********
mosdns

1715173329, 2024-08-20:
libcrypto 崩了，可能它不是很喜欢你的证书

iwowcn, 2024-08-20:
> libcrypto 崩了，可能它不是很喜欢你的证书

后期有木有可能修复？换个别的我估计它也不喜欢

nasbdh9, 2024-08-22:
尝试恢复31ec4515c3c14704d669156d87e2af5eeb5420e4试试

smallprogram, 2024-10-11:
+1 ,我今天也出现了
[image]

skefy, 2024-10-26:
[image]

我也出现了

skefy, 2024-10-26:
> 尝试恢复[31ec4515c3c14704d669156d87e2af5eeb5420e4](https://github.com/immortalwrt/immortalwrt/commit/31ec4515c3c14704d669156d87e2af5eeb5420e4)试试
> 
> git revert [31ec451](https://github.com/immortalwrt/immortalwrt/commit/31ec4515c3c14704d669156d87e2af5eeb5420e4)

你好。测试了，没有效果，还是一样的错误。
发现一个小规律：iPhone 的Safari浏览器登陆后台更容易出现

skefy, 2024-11-05:
> +1 ,我今天也出现了 ![ed6065a1c1bb63d3fb8ec3d05860dbd](https://private-user-images.githubusercontent.com/6050360/375688004-9cea2cf4-652b-4a03-bbba-3feb7796907d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3ODkwNDEsIm5iZiI6MTczMDc4ODc0MSwicGF0aCI6Ii82MDUwMzYwLzM3NTY4ODAwNC05Y2VhMmNmNC02NTJiLTRhMDMtYmJiYS0zZmViNzc5NjkwN2QuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDVUMDYzOTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDMzNzNiYmY5ZTkyY2M2NThhMWJkYmRkYTY2YjQ5Zjc5MGRkNDFiMzNiMGRiMDkyYWRjNTVhZDgyMzZjNDVhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.8UA28coV_W8dEycubccdreF8Ai0kR3Qvi1NS1j0UYhU)

兄弟，这个问题有解决吗？ 试了各种尝试，一直找不到解决方案😭

skefy, 2024-12-21:
更新：

我编译配置添加了下面两项，现在固件使用半天了正常。OP要不也验证一下：

CONFIG_OPENSSL_ENGINE_BUILTIN=y
CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG=y

magic-monkey, 2025-02-16:
> 更新：
> 
> 我编译配置添加了下面两项，现在固件使用半天了正常。OP要不也验证一下：
> 
> CONFIG_OPENSSL_ENGINE_BUILTIN=y CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG=y

本地编译这个配置写哪里？

1447262390, 2025-02-16:
我也遇到这个问题，不知道该如何解决

wtfr-dot, 2025-03-11:
> 更新：
> 
> 我编译配置添加了下面两项，现在固件使用半天了正常。OP要不也验证一下：
> 
> CONFIG_OPENSSL_ENGINE_BUILTIN=y CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG=y

我也遇到这个问题，把这两样选上也不行

leeonfu, 2025-03-31:
As you said, the issue it seems like caused by OpenSSL, suggest you can update the OpenSSL version.
And to check the uhttpd binary whether link correct , such as `ldd`

wtfr-dot, 2025-03-31:
我是自编译的op，改为使用Mbed-TLS就解决问题了，另外我因为是把配置文件一起编译进去的，这个配置文件是从23.05来，发现里面有一个openssl的配置文件，估计跟这个有关，不过我没有测试

danbao, 2025-05-15:
``` bash
uci delete uhttpd.main.listen_https
uci delete uhttpd.main.cert
uci delete uhttpd.main.key
uci commit uhttpd
/etc/init.d/uhttpd restart
```

我先移除 https来解决了

smallprogram, 2025-05-20:
我把config删了，重新make menuconfig，选择需要的东西，编译，从前天到现在再没有出现过这个问题了

KKatationer, 2025-06-17:
> 我是自编译的op，改为使用Mbed-TLS就解决问题了，另外我因为是把配置文件一起编译进去的，这个配置文件是从23.05来，发现里面有一个openssl的配置文件，估计跟这个有关，不过我没有测试

确实是这个原因。我贴一下修改过程：

* 先看看基本环境
```
root@OpenWrt:~# ldd $(which uhttpd)
        /lib/ld-musl-x86_64.so.1 (0x7f5e1e6a0000)
        libubox.so.20240329 => /lib/libubox.so.20240329 (0x7f5e1e67c000)
        libjson_script.so.20240329 => /lib/libjson_script.so.20240329 (0x7f5e1e677000)
        libblobmsg_json.so.20240329 => /lib/libblobmsg_json.so.20240329 (0x7f5e1e672000)
        libjson-c.so.5 => /usr/lib/libjson-c.so.5 (0x7f5e1e65f000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f5e1e645000)
        libc.so => /lib/ld-musl-x86_64.so.1 (0x7f5e1e6a0000)

root@OpenWrt:~# opkg list-installed | grep uhttpd
luci-app-uhttpd - 25.154.30888~023b69b
luci-i18n-uhttpd-zh-cn - 25.154.30888~023b69b
uhttpd - 2023.06.25~34a8a74d-r4
uhttpd-mod-ubus - 2023.06.25~34a8a74d-r4

root@OpenWrt:~# opkg list-installed | grep libustream
libustream-openssl20201210 - 2024.07.28~99bd3d2b-r1

root@OpenWrt:~# opkg status libustream-openssl20201210
Package: libustream-openssl20201210
Version: 2024.07.28~99bd3d2b-r1
Depends: libc, libubox20240329, libopenssl3
Provides: libustream-openssl
Status: install ok installed
Architecture: x86_64
Installed-Time: 1748769034
```

* 去除openssl
```
root@OpenWrt:~# opkg remove libustream-openssl20201210
Removing package libustream-openssl20201210 from root...
```

* **手动下载ipk后上传至tmp目录下，opkg当前无法操作SSL相关内容**
<https://mirrors.vsean.net/openwrt/releases/24.10-SNAPSHOT/packages/x86_64/base/libustream-mbedtls20201210_2024.07.28~99bd3d2b-r1_x86_64.ipk>

```
root@OpenWrt:~# opkg install /tmp/libustream-mbedtls20201210_2024.07.28\~99bd3d2b-r1_x86_64.ipk
Installing libustream-mbedtls20201210 (2024.07.28~99bd3d2b-r1) to root...
Configuring libustream-mbedtls20201210.
```

* 重启uhttpd
```
root@OpenWrt:~# /etc/init.d/uhttpd restart
```

MichaelEdson, 2025-07-06:
> 确实是这个原因。我贴一下修改过程：
>
> * 先看看基本环境
>
> ```
> root@OpenWrt:~# ldd $(which uhttpd)
>         /lib/ld-musl-x86_64.so.1 (0x7f5e1e6a0000)
>         libubox.so.20240329 => /lib/libubox.so.20240329 [...]

按你的方式试了下，外网使用safari登录Web页面正常，看上去有效果，感谢。