immortalwrt/immortalwrt #1431: 6.6.45在virtio net网卡上存在BUG

ZqinKing, 2024-08-21:
### Describe the bug

会出现大量错误日志，会影响网络：
eth0: bad gso: type: 1, size: 1452

### ImmortalWrt version

master

### ImmortalWrt release

master

### ImmortalWrt target/subtarget

x86_64

### Device

pve 虚拟机

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

ZqinKing, 2024-08-21:
[https://lore.kernel.org/all/60bc20c5-7512-44f7-88cb-abc540437ae1@heusel.eu/#r](https://lore.kernel.org/all/60bc20c5-7512-44f7-88cb-abc540437ae1@heusel.eu/#r)

qingtian110, 2024-08-21:
这个问题在  6.6.46 中已经得到修复、。。

closed 2024-08-22 by 1715173329