immortalwrt/immortalwrt #1436: 使用immortalwrt master分支编译之后启动不了

wixxm, 2024-08-24:
### Describe the bug

如图启动之后到这一步就停止了，编译了两次都是在这一步停止。

### ImmortalWrt version

immortalwrt最新的master分支

### ImmortalWrt release

immortalwrt最新的master分支

### ImmortalWrt target/subtarget

immortalwrt最新的master分支

### Device

x86

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

wixxm, 2024-08-24:
[image]

wixxm, 2024-08-24:
图片上传不上用图床了
[](https://img.imgdd.com/f210f3.4777d8f7-b675-4bf1-9dbd-a44757fe7df4.png)

wixxm, 2024-08-24:
https://img.imgdd.com/f210f3.4777d8f7-b675-4bf1-9dbd-a44757fe7df4.png

wixxm, 2024-08-24:
换了全新的编译环境master分支6.6内核编译完成依旧无法启动，编译过程没有报错，乌班图22.04系统。

skefy, 2024-08-25:
最新代码编译后在R4S 也是无法启动，换成6月份的代码分支编译出来的OK（配置没有变化）。

wixxm, 2024-08-25:
> 最新代码编译后在R4S 也是无法启动，换成6月份的代码分支编译出来的OK（配置没有变化）。

是的同样的配置，我换了23.05.3分支编译没问题

fuliaojiyue, 2024-08-31:
> > 最新代码编译后在R4S也无法启动，换成6月份的代码分支编译出来的OK（配置没有变化）。
> 
> 是同样的配置，我换了23.05.3分支编译没问题

感谢，编译23.05分支成功了

sst311212, 2024-09-03:
嘗試給 R4S 編譯了幾份不同核心版本的韌體
測試的結果是包含 6.6.43 以後的開機都 ping 不到
目前用 6.6.40 的可以正常使用，應該是41到42中的commit有問題

1715173329, 2024-09-03:
r4s 请单开 issue，并提供串口日志。

wixxm, 2024-09-05:
估计应该是内核版本问题，目前最新的6.6.47以上的内核编译都没有问题，都可以成功。

closed 2024-09-06 by 1715173329

fuliaojiyue, 2024-11-26:
> > > 最新代码编译后在R4S也无法启动，换成6月份的代码分支编译出来的OK（配置没有变化）。
> > 
> > 
> > 是同样的配置，我换了23.05.3分支编译没问题
> 
> 感谢，编译23.05分支成功了

https://github.com/immortalwrt/user-FAQ/blob/main/%E5%B0%8F%E7%99%BD%E9%97%AE%E9%A2%98%E9%80%9F%E6%9F%A5.md#%E8%87%AA%E7%BC%96%E8%AF%91%E7%9A%84%E5%9B%BA%E4%BB%B6%E6%97%A0%E6%B3%95%E8%BF%9B%E5%85%A5%E7%BD%91%E9%A1%B5luci