immortalwrt/immortalwrt #1445: iNet GL-MT3000使用最新master分支编译后lan口存在睡死不再分配ip的情况

liyuhaolol, 2024-09-02:
### Describe the bug

常用笔记本为macbook，使用有线连接的MT3000，如果拔掉lan口，不会有问题。但是如果连着lan口，电脑合上进入休眠以后，大概30分钟左右，重新打开macbook。则无法获取ip，整个lan口没有任何响应。此时无线是可以连接到路由器的，概览那里lan口显示已连接。就算拔掉网线，依旧显示已连接。感觉是僵住了。
```
Mon Sep  2 11:26:14 2024 kern.info kernel: [ 7740.805068] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Sep  2 11:26:14 2024 kern.info kernel: [ 7740.811716] br-lan: port 1(eth1) entered disabled state
Mon Sep  2 11:26:14 2024 daemon.notice netifd: Network device 'eth1' link is down
Mon Sep  2 11:26:26 2024 kern.info kernel: [ 7752.678285] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
Mon Sep  2 11:26:26 2024 kern.info kernel: [ 7752.686787] br-lan: port 1(eth1) entered blocking state
Mon Sep  2 11:26:26 2024 kern.info kernel: [ 7752.692058] br-lan: port 1(eth1) entered forwarding state
Mon Sep  2 11:26:26 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Sep  2 11:26:26 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:27:17 2024 kern.info kernel: [ 7803.588524] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Sep  2 11:27:17 2024 kern.info kernel: [ 7803.595108] br-lan: port 1(eth1) entered disabled state
Mon Sep  2 11:27:17 2024 daemon.notice netifd: Network device 'eth1' link is down
Mon Sep  2 11:27:24 2024 kern.info kernel: [ 7810.735778] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 100Mbps/Full - flow control off
Mon Sep  2 11:27:24 2024 kern.info kernel: [ 7810.736091] br-lan: port 1(eth1) entered blocking state
Mon Sep  2 11:27:24 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Sep  2 11:27:24 2024 kern.info kernel: [ 7810.749447] br-lan: port 1(eth1) entered forwarding state
Mon Sep  2 11:33:25 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:39:01 2024 daemon.info hostapd: phy0-ap0: STA 00:87:36:20:4e:19 IEEE 802.11: authenticated
Mon Sep  2 11:39:01 2024 daemon.info hostapd: phy0-ap0: STA 00:87:36:20:4e:19 IEEE 802.11: associated (aid 1)
Mon Sep  2 11:39:01 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 00:87:36:20:4e:19 auth_alg=open
Mon Sep  2 11:39:01 2024 daemon.info hostapd: phy0-ap0: STA 00:87:36:20:4e:19 RADIUS: starting accounting session E192CA34533DAB31
Mon Sep  2 11:39:01 2024 daemon.info hostapd: phy0-ap0: STA 00:87:36:20:4e:19 WPA: pairwise key handshake completed (RSN)
Mon Sep  2 11:39:01 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 00:87:36:20:4e:19
Mon Sep  2 11:39:01 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:39:02 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:39:03 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 00:87:36:20:4e:19
Mon Sep  2 11:39:03 2024 daemon.info hostapd: phy0-ap0: STA 00:87:36:20:4e:19 IEEE 802.11: disassociated
Mon Sep  2 11:41:44 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:47:25 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:53:03 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:56:10 2024 kern.info kernel: [ 9536.679000] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Sep  2 11:56:10 2024 kern.info kernel: [ 9536.685452] br-lan: port 1(eth1) entered disabled state
Mon Sep  2 11:56:10 2024 daemon.notice netifd: Network device 'eth1' link is down
Mon Sep  2 11:56:20 2024 kern.info kernel: [ 9546.787717] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
Mon Sep  2 11:56:20 2024 kern.info kernel: [ 9546.787748] br-lan: port 1(eth1) entered blocking state
Mon Sep  2 11:56:20 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Sep  2 11:56:20 2024 kern.info kernel: [ 9546.801393] br-lan: port 1(eth1) entered forwarding state
Mon Sep  2 11:56:44 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 11:57:00 2024 kern.info kernel: [ 9586.274231] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Sep  2 11:57:00 2024 kern.info kernel: [ 9586.280685] br-lan: port 1(eth1) entered disabled state
Mon Sep  2 11:57:00 2024 daemon.notice netifd: Network device 'eth1' link is down
Mon Sep  2 11:57:07 2024 kern.info kernel: [ 9593.423298] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 100Mbps/Full - flow control off
Mon Sep  2 11:57:07 2024 kern.info kernel: [ 9593.423329] br-lan: port 1(eth1) entered blocking state
Mon Sep  2 11:57:07 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Sep  2 11:57:07 2024 kern.info kernel: [ 9593.436977] br-lan: port 1(eth1) entered forwarding state
Mon Sep  2 12:03:22 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 12:07:10 2024 daemon.warn odhcpd[2052]: No default route present, setting ra_lifetime to 0!
Mon Sep  2 12:14:29 2024 kern.info kernel: [10634.373684] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Sep  2 12:14:29 2024 kern.info kernel: [10634.380146] br-lan: port 1(eth1) entered disabled state
Mon Sep  2 12:14:29 2024 daemon.notice netifd: Network device 'eth1' link is down
```
这大概是我离开，然后回来系统记录的日志，大概可以看出来eth1 link down以后，我无论怎么插网线，他就是永远都不link up了。重启lan口才能解决这个问题。并且再次休眠macbook，还可以稳定再次触发这个问题。由于是刚刚刷的系统，并没有什么自定义修改，全默认用的。所以来反馈一下问题。

### ImmortalWrt version

r31414-37bca1c75e

### ImmortalWrt release

r31414-37bca1c75e

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Self-built image

### Steps to reproduce

插上网线在lan口，电脑盒盖子休眠，等个半小时，就会一定断网，获取不到ip

### Actual behaviour

lan口僵住了，不再分配ip，路由器本身没有问题。wifi可以进后台，网络一切正常

### Expected behaviour

长时间电脑休眠恢复以后，lan口可以正常分配ip

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

liyuhaolol, 2024-09-02:
> 请提供 MacBook 系统版本和网卡型号。

- 系统版本：macOS Moterey 12.7.6
- 网卡型号：Apple 57762-A0 或者Apple Thunderbolt 2 to Gigabit Ethernet Adapter MD463LL/A A1433苹果官方的雷电2转千兆有线适配器

1715173329, 2024-09-02:
参阅 https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969

liyuhaolol, 2024-09-02:
> 参阅 [openwrt/openwrt#16281 (comment)](https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969)

已看到，按照说明进行了修改，刚刚编译好，需要睡醒后进行一段时间测试，尽量今天下午告诉您结果

liyuhaolol, 2024-09-03:
> 参阅 [openwrt/openwrt#16281 (comment)](https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969)

已经测试到快2点了，笔记本休眠一段时间再恢复，不会再出现不分配ip的情况了，看来本方法确实有效。不过导致这个问题的原因是openwrt上游的问题吧，不归本项目改。我就把issues关闭了吧。

closed 2024-09-03 by liyuhaolol

1715173329, 2024-09-14:
最新 commit 应该已经修好了。