immortalwrt/immortalwrt #1464: R5S设备固件其中一LAN口无法识别

lysgwl, 2024-09-17:
### Describe the bug

23.05.3版本编译的r5s设备固件，两个LAN口其中一个LAN口无法识别，中间的2.5GLAN口无法识别，另外一个可以使用，查询使用的驱动是R8125驱动，什么原因呢？使用[immortalwrt]提供的镜像也是如此。

### ImmortalWrt version

r27917-81a1f98d5b

### ImmortalWrt release

23.05.3

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

r5s

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-09-17:
No log = no solution.

lysgwl, 2024-09-17:
> No log = no solution.
以下时内核日志，没有检测到网卡eth2

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.162 (root@ubuntu22) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r27917-81a1f98d5b) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT Thu Jul 18 06:35:54 2024
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x00000000ee800000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s36504 r8192 d29032 u73728
[    0.000000] pcpu-alloc: s36504 r8192 d29032 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: coherent_pool=2M console=ttyS2,1500000 console=tty1 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3821484K/3930112K available (8768K kernel code, 2380K rwdata, 2568K rodata, 1664K init, 734K bss, 92244K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @440000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @450000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000000460000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000470000
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001450] Console: colour dummy device 80x25
[    0.001901] printk: console [tty1] enabled
[    0.002351] printk: bootconsole [uart8250] disabled
[    0.002883] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002916] pid_max: default: 32768 minimum: 301
[    0.003492] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.003547] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 412 lines cut ...]
[   16.171644] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   16.174143] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   16.721851] eth1: 0xffff800009fd0000, aa:ae:b5:02:67:82, IRQ 76
[   16.769755] br-lan: port 1(eth1) entered blocking state
[   16.770255] br-lan: port 1(eth1) entered disabled state
[   16.771093] device eth1 entered promiscuous mode
[   16.778417] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   16.868999] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   16.876944] dwmac4: Master AXI performs any burst length
[... cut ...]
[   40.490818] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   81.928990] r8125: eth1: link up
[   81.929371] br-lan: port 1(eth1) entered blocking state
[   81.929851] br-lan: port 1(eth1) entered forwarding state
[   81.930705] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   83.472956] r8125: eth1: link down
[   83.490155] br-lan: port 1(eth1) entered disabled state
[   85.581005] r8125: eth1: link up
[   85.581380] br-lan: port 1(eth1) entered blocking state
[   85.581860] br-lan: port 1(eth1) entered forwarding state

lysgwl, 2024-09-17:
> No log = no solution.

这个是启动日志，同时也没有发现lan口网卡2

Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Thu Jul 18 14:36:00 2024 kern.notice kernel: [    0.000000] Linux version 5.15.162 (root@ubuntu22) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r27917-81a1f98d5b) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT Thu Jul 18 06:35:54 2024
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Machine model: FriendlyElec NanoPi R5S
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Zone ranges:
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000]   DMA32    empty
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000]   Normal   empty
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Early memory node ranges
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x00000000ee800000
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.2
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s36504 r8192 d29032 u73728
Thu Jul 18 14:36:00 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s36504 r8192 d29032 u73728 alloc=18*4096
Thu Jul 18 14:36:00 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] CPU features: detected: GIC system register CPU interface
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] CPU features: detected: Virtualization Host Extensions
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] CPU features: detected: Hardware dirty bit management
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] alternatives: patching kernel code
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
Thu Jul 18 14:36:00 2024 kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=2M console=ttyS2,1500000 console=tty1 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Memory: 3821484K/3930112K available (8768K kernel code, 2380K rwdata, 2568K rodata, 1664K init, 734K bss, 92244K reserved, 16384K cma-reserved)
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] 	Trampoline variant of Tasks RCU enabled.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] GICv3: 320 SPIs implemented
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] GICv3: 0 Extended SPIs implemented
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] GICv3: Distributor has no Range Selector support
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
Thu Jul 18 14:36:00 2024 kern.info kernel: [    0.000000] GICv3: 16 PPIs implemented
[... cut ...]
Thu Jul 18 14:52:17 2024 daemon.err uwsgi[1306]: Thu Jul 18 14:52:17 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is now loyal
Thu Jul 18 14:52:42 2024 daemon.info wsdd2: samba36/4 or ksmbd is not running
Thu Jul 18 14:52:48 2024 daemon.info wsdd2: samba36/4 or ksmbd is not running
Thu Jul 18 14:52:49 2024 daemon.warn odhcpd[2341]: No default route present, setting ra_lifetime to zero!
Thu Jul 18 14:53:08 2024 authpriv.info dropbear[6991]: Child connection from 192.168.2.5:3513
Thu Jul 18 14:53:09 2024 authpriv.notice dropbear[6991]: Password auth succeeded for 'root' from 192.168.2.5:3513

lysgwl, 2024-09-18:
> No log = no solution.

能看出什么原因吗？拜托🙏大佬

1715173329, 2024-09-18:
运行 lspci 看看，内核认不出你的网卡

lysgwl, 2024-09-18:
> 运行 lspci 看看，内核认不出你的网卡

root@wl-openwrt:~# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:11:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)

这是运行后的结果，认出一个2.5G设备。测试友善提供的固件，运行就正常 了。
还有一个奇怪问题，简单替换了rkbin和uboot版本，第一次运行设备网卡依然检测失败，启动后再次检测竟然可以了。

lysgwl, 2024-09-18:
> 运行 lspci 看看，内核认不出你的网卡
[image]
[image]

大佬，还有一个问题，状态灯的显示也存在问题，lan口插在第一个口，按理说应该是指示灯1亮，但是显示在第二个指示灯。截图给你看看。

lysgwl, 2024-09-26:
> No log = no solution.

大神，问题能确认吗？需要我协助做点什么呢？看最新代码更新好多友善固件信息，最新代码已解决我的故障问题吗？

qingtian110, 2024-09-28:
> > No log = no solution.
> 
> 大神，问题能确认吗？需要我协助做点什么呢？看最新代码更新好多友善固件信息，最新代码已解决我的故障问题吗？

试试 openwrt： https://r5s.cooluc.com/minimal/openwrt-23.05/v23.05.5/openwrt-23.05.5-rockchip-armv8-friendlyarm_nanopi-r5s-squashfs-sysupgrade.img.gz

lysgwl, 2024-09-29:
> > > No log = no solution.
> > 
> > 
> > 大神，问题能确认吗？需要我协助做点什么呢？看最新代码更新好多友善固件信息，最新代码已解决我的故障问题吗？
> 
> 试试 openwrt： https://r5s.cooluc.com/minimal/openwrt-23.05/v23.05.5/openwrt-23.05.5-rockchip-armv8-friendlyarm_nanopi-r5s-squashfs-sysupgrade.img.gz

这个固件测试过了，还是不行，跟以前的状况是一样的。中间Lan的网口还是不行，查看内核日志也没有认出，以下时日志情况：

[系统日志.log](https://github.com/user-attachments/files/17177054/default.log)
[内核日志.log](https://github.com/user-attachments/files/17177055/default.log)

qingtian110, 2024-10-01:
> 这个固件测试过了，还是不行，跟以前的状况是一样的。中间Lan的网口还是不行，查看内核日志也没有认出，以下时日志情况：
>
> [系统日志.log](https://github.com/user-attachments/files/17177054/default.log) [...]

看了一下你上传的内核日志，设备故障的可能性比较大。。。

你的内核上只检测到并驱动了一个 R8125 网卡，一个正常的 R5S 设备，内核是会显示两个 R8125 驱动信息的。

这是我的 NanoPI R5S 内核日志
```
[    8.687660] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded
[    8.688425] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[    8.705517] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    8.706404] r8125 0000:01:00.0 (unnamed net_device) (uninitialized): Random ether addr f6:5b:b8:0b:10:98
[    8.707897] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    8.711009] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[    8.711009]  This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>. 
[    8.711009]  This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>. 
[    8.713921] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded
[    8.714600] r8125 0001:11:00.0: enabling device (0000 -> 0003)
[    8.731520] r8125 0001:11:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    8.732392] r8125 0001:11:00.0 (unnamed net_device) (uninitialized): Random ether addr 72:51:61:96:e7:10
[    8.733782] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    8.736825] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
```

你应该有个网卡是坏掉了。。。

lysgwl, 2024-10-02:
> 看了一下你上传的内核日志，设备故障的可能性比较大。。。
>
> 你的内核上只检测到并驱动了一个 R8125 网卡，一个正常的 R5S 设备，内核是会显示两个 R8125 驱动信息的。
>
> 这是我的 NanoPI R5S 内核日志
>
> ```
> [    8.687660] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded
> [ [...]

坏掉的可能性不大，测试用友善官方的固件，是正常的。网卡也能驱起来。目前刷的是istoreOS，正常使用。你说是哪的问题，坏没坏？过几天给上传istoreos的内核日志。

lysgwl, 2024-10-04:
> 看了一下你上传的内核日志，设备故障的可能性比较大。。。
>
> 你的内核上只检测到并驱动了一个 R8125 网卡，一个正常的 R5S 设备，内核是会显示两个 R8125 驱动信息的。
>
> 这是我的 NanoPI R5S 内核日志
>
> ```
> [    8.687660] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded
> [ [...]

[系统日志.log](https://github.com/user-attachments/files/17261102/default.log)
[内核日志.log](https://github.com/user-attachments/files/17261104/default.log)
以上日志是istoreos正常使用的日志信息，网卡是可以正常使用的。

qingtian110, 2024-10-05:
那就真玄学！

我也是 NanoPI R5S，但是我用 openwrt(Snapshot)、immortalwrt、第三方 openwrt 23.05 (kernel 6.6 / 6.12-rc1)，三个网口都是正常的，并且你说的那个 LED 灯也是正确的。甚至网口的灯也正常闪烁。

[image]

lysgwl, 2024-10-07:
> 那就真玄学！
>
> 我也是 NanoPI R5S，但是我用 openwrt(Snapshot)、immortalwrt、第三方 openwrt 23.05 (kernel 6.6 / 6.12-rc1)，三个网口都是正常的，并且你说的那个 LED 灯也是正确的。甚至网口的灯也正常闪烁。 [...]

所以说，就奇了怪了吗，折腾了好久第三方固件都不行，测试过lede也不行。之前一直在用immortalwrt，然后有一次把友善的uboot和rkbin替换后，倒是成功过一次，因为对immortalwrt结构代码不是熟，只是简单替换。
下面这图是运行istoreos网卡识别成功的，也是现在在用的版本：
[image]
感觉是不是immortalwrt在适配r5s上，uboot和rkbin存在点问题，能帮忙替换成友善最新版本，我来测试下吗？

closed 2024-10-18 by lysgwl