immortalwrt/immortalwrt #1466: FriendlyElec NanoPi R5C: Kernel call trace when pppd wants to connect to LTE via Huawei E3372-s153 USB dongle.

dreirund, 2024-09-17:
### Describe the bug

As soon as pppd wants to connect to 4G/LTE via a connected USB Huawei E3372-s153 USB dongle, I get a kernel call trace, a pppd crash, and also some other network messup since only ssh, but not the LUCI gui, works anymore until a restart:

```
[  666.369633] 3g-cellwan0: renamed from ppp0
[  666.391242] Unable to handle kernel paging request at virtual address 0000000000005451
[  666.391985] Mem abort info:
[  666.392243]   ESR = 0x0000000096000004
[  666.392583]   EC = 0x25: DABT (current EL), IL = 32 bits
[  666.393061]   SET = 0, FnV = 0
[  666.393339]   EA = 0, S1PTW = 0
[  666.393626]   FSC = 0x04: level 0 translation fault
[  666.394065] Data abort info:
[  666.394327]   ISV = 0, ISS = 0x00000004
[  666.394725]   CM = 0, WnR = 0
[  666.395004] user pgtable: 4k pages, 48-bit VAs, pgdp=000000000d930000
[  666.395584] [0000000000005451] pgd=0000000000000000, p4d=0000000000000000
[  666.396301] Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP
[  666.396882] Modules linked in: dm_crypt nf_nat_amanda nf_conntrack_amanda encrypted_keys rtw88_8822ce rtw88_8822c rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option mt7921u mt7921_common mt76x0u mt76x0_common ipw gspca_m5602 gspca_konica gspca_jeilinj gspca_gl860 gspca_finepix gspca_etoms gspca_conex cdc_mbim carl9170 ath6kl_usb ath6kl_core ath ar5523 wireguard uvcvideo usb_wwan usb_serial_simple ueagle_atm ti_usb_3410_5052 sr9700 speedtch snd_usb_audio smsc95xx smsc75xx sierra_net sierra rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl8812au rt2x00usb rt2x00lib rsi_usb rsi_91x rndis_host r8723bs qmi_wwan pwc pppoe plusb pl2303 oti6858 nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mt792x_usb mt792x_lib mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7601u mt76_usb mt76_connac_lib mt76 mos7840 mos7720 mmc_spi mct_u232 mcs7830 mac80211 lz4 libchacha20poly1305 l2tp_ppp keyspan kalmia ipt_REJECT io_edgeport huawei_cdc_ncm gspca_zc3xx gspca_vc032x
[  666.397243]  gspca_tv8532 gspca_t613 gspca_sunplus gspca_stv06xx gspca_stk014 gspca_sq930x gspca_sq905c gspca_sq905 gspca_spca561 gspca_spca508 gspca_spca506 gspca_spca505 gspca_spca501 gspca_spca500 gspca_sonixj gspca_sonixb gspca_sn9c20x gspca_se401 gspca_pac7311 gspca_pac207 gspca_ov534_9 gspca_ov534 gspca_ov519 gspca_mr97310a gspca_mars gspca_main garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cxacru cp210x chacha_neon ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix ark3116 aqc111 zstd yealink xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_physdev xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipv4options xt_iprange xt_iface xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_condition xt_comment xt_TCPMSS xt_REDIRECT xt_PROTO xt_NETMAP xt_MASQUERADE xt_LOG xt_LED xt_HL xt_FLOWOFFLOAD xt_DSCP xt_DHCPMAC xt_CT xt_CLASSIFY
[  666.405189]  xr_usb_serial_common xfrm_interface visor videobuf2_v4l2 vhci_hcd usbserial usbnet usbmon usblp usbip_host usbip_core usbatm ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uinput ts_kmp ts_fsm ts_bm tcp_scalable tcp_hybla tcp_bbr spi_gpio snd_usbmidi_lib smsc sch_mqprio sch_cake rtl8150 rfcomm r8152 r8125 pptp pppox ppp_synctty ppp_mppe ppp_async poly1305_neon pegasus of_mmc_spi nlmon nft_xfrm nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fullcone nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_log_syslog nf_flow_table nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp
[  666.412923]  nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge nf_conncount netatop mdio_netlink macvlan macremapper lzo_rle lzo lz4_decompress lz4_compress libcurve25519_generic libchacha lan78xx kaweth ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables hso hidp hid_cp2112 hci_uart ezusb ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 drm_kms_helper dmx_usb crc8 crc7 crc_itu_t cpia2 cordic compat_xtables compat cm109 cls_flower cdc_wdm cdc_acm btusb btrtl btintel br_netfilter bnep bluetooth ax88796b atusb arptable_filter arpt_mangle arp_tables act_vlan mac802154 crc_ccitt ieee802154_socket ieee802154 fuse ntfs3 sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_codel em_text em_nbyte em_meta em_cmp act_simple
[  666.420629]  act_pedit act_csum em_ipset cls_bpf act_bpf act_ctinfo act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs videobuf2_vmalloc videobuf2_memops videobuf2_common sg videodev drivetemp lp ledtrig_usbport trelay ppdev parport ledtrig_pattern ledtrig_oneshot ledtrig_gpio ledtrig_audio ledtrig_activity gpio_beeper cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink sr_mod ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 pppoatm
[  666.428335]  ppp_generic bonding tls ip6_gre ip_gre gre ifb l2tp_ip6 l2tp_ip l2tp_eth ip6_vti ip_vti sit sctp l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore slip slhc ovpn_dco xfrm_user xfrm_ipcomp af_key xfrm_algo isofs cdrom cifs cifs_md4 cifs_arc4 autofs4 dns_resolver br2684 atm nls_utf8 nls_iso8859_15 nls_cp850 nls_cp437 zram zsmalloc vxlan udp_tunnel ip6_udp_tunnel macsec bfq wp512 twofish_generic twofish_common tea serpent_generic khazad cast6_generic cast5_generic cast_common camellia_generic blowfish_generic blowfish_common anubis ecdh_generic ecc xxhash_generic xts xcbc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv rmd160 pcbc michael_mic md5 md4 kpp fcrypt echainiv ecb
[  666.436068]  des_generic libdes deflate cts cmac chacha20poly1305 cbc blake2b_generic authencesn authenc arc4 crypto_acompress fbtft(C) sysimgblt sysfillrect syscopyarea fb_sys_fops cfbimgblt cfbfillrect cfbcopyarea drm fb font drm_panel_orientation_quirks uas uleds dwc2 uhci_hcd ledtrig_transient fsl_mph_dr_of ehci_fsl gpio_button_hotplug vfat fat exfat btrfs xor zstd_decompress zstd_compress xor_neon raid6_pq lzo_decompress lzo_compress dm_mirror dm_region_hash dm_log dm_mod dax button_hotplug gpio_cascade mux_core udc_core microchip trusted tpm oid_registry asn1_encoder asn1_decoder [last unloaded: ksmbd]
[  666.448407] CPU: 1 PID: 23926 Comm: pppd Tainted: G         C        5.15.162 #0
[  666.449060] Hardware name: FriendlyElec NanoPi R5C (DT)
[  666.449521] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  666.450135] pc : rtw_adapter_proc_replace+0x24/0xac [rtl8812au]
[  666.450715] lr : rtw_cbuf_free+0x118/0x11c [rtl8812au]
[  666.451216] sp : ffff80000d473b00
[  666.451510] x29: ffff80000d473b00 x28: dead000000000122 x27: ffff000001acc550
[  666.452140] x26: 0000000000000001 x25: ffff800009563de8 x24: ffff000007d03cc0
[  666.452771] x23: 0000000000000001 x22: ffff000001acc000 x21: 000000000000000b
[  666.453403] x20: ffff000001acc000 x19: 000000000000000b x18: 0000000000000000
[  666.454034] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[  666.454665] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[... cut ...]
[  666.462529]  el0t_64_sync_handler+0xe8/0x114
[  666.462907]  el0t_64_sync+0x184/0x188
[  666.463238] Code: aa0003f6 b0000620 f9400818 f944a2d7 (f96a2af5) 
[  666.463776] ---[ end trace ab485869c1b629f4 ]---
```

### ImmortalWrt version

`r27917-81a1f98d5b`

### ImmortalWrt release

`23.05.3`

### ImmortalWrt target/subtarget

`rockchip/armv8`

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-09-18:
I pretty doubt if it's "official downloaded image".
Remove kmod-rtl8812au-ac and all other rtk craps from the system, and try again.

dreirund, 2024-09-18:
> *I pretty doubt if it's "official downloaded image".*

It was build by your official tools by your official system on

https://firmware-selector.immortalwrt.org/

with custom specified packages to bake into the system, and thendownloaded from there.

My custom list was this: [`custom_packages.txt`](https://github.com/user-attachments/files/17039997/custom_packages.txt)

*(this would be similar with downloading a minimal official image and then do from this `opkg install`, I think. So in my opinion it counts as official image(?))*

> Remove `kmod-rtl8812au-ac` and all other rtk craps from the system, and try again.

I did remove all packages containing `rtl` except of `rtl8822ce-firmware`, `kmod-rtlwifi`, `kmod-rtlwifi-usb`, `kmod-usb-net-rtl8150`, `kmod-usb-net-rtl8152`.

This did make it work.

Thanks for the hint!

closed 2024-09-18 by dreirund

dreirund, 2024-09-18:
> *I pretty doubt if it's "official downloaded image". Remove kmod-rtl8812au-ac and all other rtk craps from the system, and try again.*

`rtl8822ce` is the driver for the onboard WiFi.