immortalwrt/immortalwrt #1468: 系统进程页面不显示任何信息

smallprogram, 2024-09-18:
### Describe the bug

系统进程页面不显示任何信息，如图：
[image]

我检查了相应的源码

https://github.com/immortalwrt/luci/blob/898781f90ed2683295cb84c7f38c3a80bc7bfadd/modules/luci-mod-status/htdocs/luci-static/resources/view/status/processes.js#L7-L11

我发现使用rpc远程调用了luci的一个getProcessList的方法，我找了luci中的这个方法：

https://github.com/immortalwrt/luci/blob/898781f90ed2683295cb84c7f38c3a80bc7bfadd/modules/luci-base/root/usr/share/rpcd/ucode/luci#L540-L544

而它调用了process_list()方法：

https://github.com/immortalwrt/luci/blob/898781f90ed2683295cb84c7f38c3a80bc7bfadd/modules/luci-base/ucode/sys.uc#L6-L30

而当我手工在openwrt执行命令`/bin/busybox top -bn1`，又可以返回正常的数据：

[image]

碍于功力不够，我无法尝试修复或者找到这个问题的解决方案。故提出这个issue,希望有人能够帮助我

### ImmortalWrt version

r31654-1ed559c77c

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

x86/64

### Device

iKOOLCORE TECHNOLOGY iKOOLCORE R2

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-09-19:
应该走 ubus 调用看有没有响应，另外用 f12 查看加载情况。

1715173329, 2024-09-19:
使用 istore 的东西不受支持。

smallprogram, 2024-09-19:
没有使用istore，纯纯的源码编译，查看了网络，ubus的request正常返回200，但是response中没有数据：
[image]
[image]

1715173329, 2024-09-19:
这是你装的 battstatus 的数据，不是 processes 的数据。