immortalwrt/immortalwrt #1470: 树莓派5 刷入后通电无http服务启动

breaking153, 2024-09-19:
### Describe the bug

树莓派5 刷入后通电无http服务启动

### ImmortalWrt version

r31537-3b243119ad

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-09-19:
snapshot 固件不自带 luci。

closed 2024-09-19 by 1715173329

libiunc, 2024-09-19:
[参考这个](https://github.com/immortalwrt/user-FAQ/blob/main/%E5%B0%8F%E7%99%BD%E9%97%AE%E9%A2%98%E9%80%9F%E6%9F%A5.md#%E8%87%AA%E7%BC%96%E8%AF%91%E7%9A%84%E5%9B%BA%E4%BB%B6%E6%97%A0%E6%B3%95%E8%BF%9B%E5%85%A5%E7%BD%91%E9%A1%B5luci)