immortalwrt/immortalwrt #1490: MediaTek MT7981 RFB Z8103AX，编译23.05.4后路由器无法正常启动

Mrbug001, 2024-10-01:
### Describe the bug

Z8103AX路由器，正常编译23.05.4系统，编译后安装，路由器无法正常启动

### ImmortalWrt version

23.05.4

### ImmortalWrt release

immortalwrt-mediatek-filogic-zbtlink_zbt-z8103ax-squashfs-sysupgrade.bin

### ImmortalWrt target/subtarget

immortalwrt-mediatek-filogic-zbtlink_zbt-z8103ax-squashfs-sysupgrade.bin

### Device

Z8103AX

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-01:
No log = no solution.

Mrbug001, 2024-10-01:
> No log = no solution.

路由器不能启动啊！看不了日志，只是几个lan口灯频繁闪动

oenhu, 2024-10-29:
接TTL 看日志