immortalwrt/immortalwrt #1493: 使用数天后 WIFI 出现无法连接的情况

TomCzHen, 2024-10-06:
### Describe the bug

使用一周左右之后，会突然出现 WIFI 无法连接的问题，手机和 PC症状一致，可以搜索到 SSID，但是无法连接上。 PC 端 dmesg 日志显示是认证超时，有线可以正常连接。

```
[47918.618098] [   T1576] wlo1: authenticate with a4:a9:30:71:20:b8 (local address=60:e9:aa:0a:c1:05)
[47918.627729] [   T1576] wlo1: send auth to a4:a9:30:71:20:b8 (try 1/3)
[47918.632355] [ T102737] wlo1: send auth to a4:a9:30:71:20:b8 (try 2/3)
[47918.636933] [ T102737] wlo1: send auth to a4:a9:30:71:20:b8 (try 3/3)
[47918.641468] [ T102737] wlo1: authentication with a4:a9:30:71:20:b8 timed out
[48171.801909] [   T1576] wlo1: authenticate with a4:a9:30:71:20:b8 (local address=60:e9:aa:0a:c1:05)
[48171.811443] [   T1576] wlo1: send auth to a4:a9:30:71:20:b8 (try 1/3)
[48171.818277] [  T92130] wlo1: send auth to a4:a9:30:71:20:b8 (try 2/3)
[48171.822850] [  T92130] wlo1: send auth to a4:a9:30:71:20:b8 (try 3/3)
[48171.827459] [  T92130] wlo1: authentication with a4:a9:30:71:20:b8 timed out
[48184.412545] [   T1576] wlo1: authenticate with a4:a9:30:71:20:b8 (local address=60:e9:aa:0a:c1:05)
[48184.422167] [   T1576] wlo1: send auth to a4:a9:30:71:20:b8 (try 1/3)
[48184.426866] [  T99540] wlo1: send auth to a4:a9:30:71:20:b8 (try 2/3)
[48184.431411] [  T99540] wlo1: send auth to a4:a9:30:71:20:b8 (try 3/3)
[48184.435928] [  T99540] wlo1: authentication with a4:a9:30:71:20:b8 timed out
[48220.807514] [   T1576] wlo1: authenticate with a4:a9:30:71:20:b8 (local address=60:e9:aa:0a:c1:05)
[48220.817148] [   T1576] wlo1: send auth to a4:a9:30:71:20:b8 (try 1/3)
[48220.821690] [  T48310] wlo1: send auth to a4:a9:30:71:20:b8 (try 2/3)
[48220.826325] [  T48310] wlo1: send auth to a4:a9:30:71:20:b8 (try 3/3)
[48220.830961] [  T48310] wlo1: authentication with a4:a9:30:71:20:b8 timed out
```

### ImmortalWrt version

r27917-81a1f98d5b

### ImmortalWrt release

23.05.3

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000

### Image kind

Official downloaded image

### Steps to reproduce

正常使用数天概率触发，尚未发现可以主动触发问题的方式。根据同型号设备（固件版本也相同）在朋友公司的使用情况，有时一天之类就会出现该问题，推测和终端数量相关。

### Additional info

路由器上的日志没有看出wifi 相关的日志，有 hostapd 相关的日志，但是 mac 是其他设备，没有 pc 无线网卡 mac 相关设备。

```
Sun Oct  6 23:31:45 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 08:e9:f6:62:89:9e
Sun Oct  6 23:31:45 2024 daemon.info hostapd: phy0-ap0: STA 08:e9:f6:62:89:9e IEEE 802.11: disassociated due to inactivity
Sun Oct  6 23:31:46 2024 daemon.info hostapd: phy0-ap0: STA 08:e9:f6:62:89:9e IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

尝试重启无线设备，仍然无法连接，日志如下

```
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: Set new config for phy phy1:
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: Remove interface 'phy1'
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Sun Oct  6 23:48:14 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Oct  6 23:48:14 2024 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sun Oct  6 23:48:14 2024 kern.info kernel: [770308.965648] device phy1-ap0 left promiscuous mode
Sun Oct  6 23:48:14 2024 kern.info kernel: [770308.970540] br-lan: port 5(phy1-ap0) entered disabled state
Sun Oct  6 23:48:14 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
Sun Oct  6 23:48:15 2024 daemon.notice wpa_supplicant[1743]: Set new config for phy phy1
Sun Oct  6 23:48:15 2024 daemon.notice netifd: Wireless device 'radio1' is now down
Sun Oct  6 23:48:15 2024 daemon.notice netifd: radio1 (22955): WARNING: Variable 'data' does not exist or is not an array/object
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: Set new config for phy phy1:
Sun Oct  6 23:48:15 2024 daemon.notice wpa_supplicant[1743]: Set new config for phy phy1
Sun Oct  6 23:48:15 2024 daemon.notice wpa_supplicant[1743]: Set new config for phy phy1
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: Restart interface for phy phy1
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CN ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=42 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_ed
Sun Oct  6 23:48:15 2024 kern.info kernel: [770309.924742] br-lan: port 5(phy1-ap0) entered blocking state
Sun Oct  6 23:48:15 2024 kern.info kernel: [770309.930401] br-lan: port 5(phy1-ap0) entered disabled state
Sun Oct  6 23:48:15 2024 kern.info kernel: [770309.936272] device phy1-ap0 entered promiscuous mode
Sun Oct  6 23:48:15 2024 kern.info kernel: [770309.941463] br-lan: port 5(phy1-ap0) entered blocking state
Sun Oct  6 23:48:15 2024 kern.info kernel: [770309.947112] br-lan: port 5(phy1-ap0) entered forwarding state
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Oct  6 23:48:15 2024 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Sun Oct  6 23:48:15 2024 kern.info kernel: [770310.130943] br-lan: port 5(phy1-ap0) entered disabled state
Sun Oct  6 23:48:15 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Sun Oct  6 23:48:17 2024 kern.info kernel: [770311.264975] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Sun Oct  6 23:48:17 2024 kern.info kernel: [770311.271847] br-lan: port 5(phy1-ap0) entered blocking state
Sun Oct  6 23:48:17 2024 kern.info kernel: [770311.277515] br-lan: port 5(phy1-ap0) entered forwarding state
Sun Oct  6 23:48:17 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Sun Oct  6 23:48:17 2024 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->ENABLED
Sun Oct  6 23:48:17 2024 daemon.notice hostapd: phy1-ap0: AP-ENABLED
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-06:
Upgrade to latest stable first.

142857X1, 2024-10-11:
遇到过类似无线问题，wifi丢失重启后恢复正常，

libriunk, 2024-10-11:
> 遇到过类似无线问题，wifi丢失重启后恢复正常，

下次出现这种情况 请用网口连上路由器，然后把日志po在这里

TomCzHen, 2024-10-11:
> 遇到过类似无线问题，wifi丢失重启后恢复正常，

我在23.05.2版本遇到过，症状是WiFi搜索不到SSID，进后台查看无线设备状态显示不可用。但是也只遇到过一次。

目前我遇到最多的是可以搜到WiFiSSID但是无法连接，已经连接上的设备虽然现实是连接的，但是实际网络是不通的。

yang15333, 2024-10-11:
这个问题我也遇到过，现象是所有设备断开wifi连接，并且搜索不到ssid，我用的解决方案是禁用不活动轮询，这个问题就没有再出现了，怀疑是当有设备变为不活动状态是触发的bug

yang15333, 2024-10-14:
> > 这个问题我也遇到过，现象是所有设备断开wifi连接，并且搜索不到ssid，我用的解决方案是禁用不活动轮询，这个问题就没有再出现了，怀疑是当有设备变为不活动状态是触发的bug
> 
> 请问老哥是怎么具体解决的，我也复现这个问题了，重刷了好几次

我是在网络-无线-高级设置-勾选禁用不活动轮询，没再出现过，2.4g和5g的都勾选了

libriunk, 2024-10-14:
> > 这个问题我也遇到过，现象是所有设备断开wifi连接，并且搜索不到ssid，我用的解决方案是禁用不活动轮询，这个问题就没有再出现了，怀疑是当有设备变为不活动状态是触发的bug
> 
> 请问老哥是怎么具体解决的，我也复现这个问题了，重刷了好几次

方便提供一下日志吗，在无线无法使用的时候用有线连上去  在'状态-系统日志'
把日志都发一下 谢谢

BlueCitizens, 2024-11-10:
遇到了疑似相同的故障。

### 固件版本

ImmortalWrt 23.05.4 r28061-399f9a1db3 / LuCI openwrt-23.05 branch git-24.265.44782-0ff45d8

### 目标平台
mediatek/filogic

### 内核版本
5.15.167

### 设备
Xiaomi Redmi Router AX6000

### 镜像
Official downloaded image

### 故障描述
使用2天左右会出现无线无法连接的情况，原本连着的设备也会断连。有线连接可以正常上网，重启后会恢复正常，约2天左右会再次出现，已循环多次，触发条件不明。

### 已安装的插件
OpenClash

### 系统日志
故障大约发生于Sun Nov 10 22:00:00
```
Sat Nov  9 18:03:11 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:10:00 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:14:44 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:16:17 2024 daemon.info hostapd: phy0-ap0: STA 2a:97:d4:92:f6:a0 IEEE 802.11: associated (aid 2)
Sat Nov  9 18:16:17 2024 daemon.notice hostapd: phy1-ap0: Prune association for 2a:97:d4:92:f6:a0
Sat Nov  9 18:16:17 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 2a:97:d4:92:f6:a0
Sat Nov  9 18:16:19 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 2a:97:d4:92:f6:a0 auth_alg=sae
Sat Nov  9 18:16:19 2024 daemon.info hostapd: phy0-ap0: STA 2a:97:d4:92:f6:a0 RADIUS: starting accounting session 5263959BA26F6F2F
Sat Nov  9 18:16:19 2024 daemon.info hostapd: phy0-ap0: STA 2a:97:d4:92:f6:a0 WPA: pairwise key handshake completed (RSN)
Sat Nov  9 18:16:19 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 2a:97:d4:92:f6:a0
Sat Nov  9 18:16:47 2024 daemon.info hostapd: phy1-ap0: STA 2a:97:d4:92:f6:a0 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Nov  9 18:22:19 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:23:45 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:27:28 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:27:40 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:32:46 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:36:20 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:38:02 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:42:18 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:42:20 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:42:24 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:43:19 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:48:28 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:49:56 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:53:46 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 18:55:57 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 18:58:48 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:04:00 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:04:21 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 19:07:43 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 19:09:09 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:13:13 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 19:14:23 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:19:39 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:22:08 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sat Nov  9 19:24:50 2024 daemon.notice hostapd: phy0-ap0: AP-STA-POLL-OK 2a:97:d4:92:f6:a0
Sat Nov  9 19:26:22 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
[... cut ...]
Sun Nov 10 22:35:50 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sun Nov 10 22:35:51 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sun Nov 10 22:35:52 2024 daemon.warn odhcpd[1963]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Sun Nov 10 22:36:27 2024 daemon.err uhttpd[2082]: [info] luci: accepted login on / for root from 192.168.123.241
```

OpenSourceUser2021, 2024-12-22:
1-2 天遇到一次，WIFI 不通，有线正常

immortalwrt 最新mater 和24.10 分支都i有
换成openwrt 官方的也一样

<html>
<body>

主机名 | OpenWrt-24.10
-- | --
型号 | JDCloud RE-CP-03
架构 | ARMv8 Processor rev 4
目标平台 | mediatek/filogic
固件版本 | OpenWrt 24.10-SNAPSHOT r28201-5289460fd0 / LuCI openwrt-24.10 branch 24.355.00587~f92e072
内核版本 | 6.6.66

</body>
</html>

[wifi_log.txt](https://github.com/user-attachments/files/18221368/wifi_log.txt)

rbqvq, 2025-03-10:
Newifi3有的时候WLAN也抽风 莫名其妙死机或者WLAN不见了
目前解决方法是定时任务每天重启😇

xixky, 2026-01-09:
> newifi d1也会出现，断开wifi之后，重新连接会提示拒绝接入。一天出现一次。

现在解决办法是编译大雕的固件