immortalwrt/immortalwrt #1496: 求助make menuconfig时，菜单默认已选项太多，导致编译固件过大。

kygsmsc, 2024-10-08:
### Describe the bug

翻出家里一台老路由，只有16M闪存，用make menuconfig编译，自己什么都没有选择，就只是改变成路由器对应的7620A和路由器型号，结果编译出来的固件大于16M，在make menuconfig 已默认勾选的配置中排查了半天，也没有发现是勾选了什么造成最后编译的固件过大的，望能给于帮助。（L大的源码也有同样问题，官方源码却无此问题， make menuconfigi改变默认项编译出来人固件小于6M。

### ImmortalWrt version

无

### ImmortalWrt release

无

### ImmortalWrt target/subtarget

无

### Device

hc5761

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-08:
https://downloads.immortalwrt.org/releases/23.05.4/targets/ramips/mt7620/
本项目默认配置编译出来只有 8M 左右，哪儿来的 16M？

closed 2024-10-08 by 1715173329

kygsmsc, 2024-10-08:
在线github云编译，拉取地址：https://github.com/immortalwrt/immortalwrt，master版，只是改变路由器对应的7620A和路由器型号大麦DW22D，生成固件大于16M。
[image]