immortalwrt/immortalwrt #1503: make defconfig后出现递归错误！

netlovehf, 2024-10-10:
### Describe the bug

tmp/.config-package.in:127810:error: recursive dependency detected!
tmp/.config-package.in:127810:	symbol PACKAGE_wpad-mesh-openssl is selected by PACKAGE_luci-app-easymesh
tmp/.config-package.in:64449:	symbol PACKAGE_luci-app-easymesh depends on PACKAGE_libopenssl
tmp/.config-package.in:53714:	symbol PACKAGE_libopenssl is selected by PACKAGE_wpa-supplicant-mesh-openssl
tmp/.config-package.in:127284:	symbol PACKAGE_wpa-supplicant-mesh-openssl depends on PACKAGE_wpad-mesh-openssl
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

### ImmortalWrt version

23.5

### ImmortalWrt release

tmp/.config-package.in:127810:error: recursive dependency detected!

### ImmortalWrt target/subtarget

tmp/.config-package.in:127810:error: recursive dependency detected!

### Device

R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2024-10-11:
哪里来的 easymesh 
wifi都没有的板子，要这个干啥……

暴风影音xx特别版 无法安装到windows上，找微软客服可不行

1715173329, 2024-10-11:
使用第三方插件，请联系第三方插件开发者。

closed 2024-10-11 by 1715173329