immortalwrt/immortalwrt #1509: 设置IP集合出现错误

cnkmmk, 2024-10-12:
### Describe the bug

防火墙中设置IP集合，IP地址设置为 192.168.4.80/29，之后在 /etc/config/firewall 中查看，确实是 list entry '192.168.4.80/29',
但在nftables 中查看时，却变成了 elements = { 192.168.4.80 } 。

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

bcm53xx/generic

### Device

Phicomm K3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-12:
This issue should be sent to upstream.

bsdcpp, 2024-10-20:
一样的问题，但是好像不影响使用

cnkmmk, 2024-10-20:
> 一样的问题，但是好像不影响使用

只能一个地址一个地址的设置，没办法一个网段的搞起。
强迫症看着就是不舒服。

bsdcpp, 2024-10-21:
> > 一样的问题，但是好像不影响使用
> 
> 只能一个地址一个地址的设置，没办法一个网段的搞起。 强迫症看着就是不舒服。

确实，我之前测速的可能是tcp连接没有断，通过文件添加的网段其实是不行的

bsdcpp, 2024-10-21:
@cnkmmk 发现问题了，在页面添加的时候要选择src_net，但是很奇怪保存/提交无法生效，需要重启防火墙
/etc/init.d/firewall restart
```bash
table inet fw4 {
	set white_ip_list {
		type ipv4_addr
		flags interval
		auto-merge
		comment ""
		elements = { 27.0.204.0-27.0.215.255, 27.109.124.0/22,
			     27.113.128.0/18, 27.115.0.0/17,
			     36.111.0.0/19, 36.111.192.0/21,
			     36.113.0.0/16, 36.133.24.0-36.133.35.255,
```

cnkmmk, 2024-10-23:
> @cnkmmk 发现问题了，在页面添加的时候要选择src_net，但是很奇怪保存/提交无法生效，需要重启防火墙 /etc/init.d/firewall restart
> 
> ```shell
> table inet fw4 {
> 	set white_ip_list {
> 		type ipv4_addr
> 		flags interval
> 		auto-merge
> 		comment ""
> 		elements = { 27.0.204.0-27.0.215.255, 27.109.124.0/22,
> 			     27.113.128.0/18, 27.115.0.0/17,
> 			     36.111.0.0/19, 36.111.192.0/21,
> 			     36.113.0.0/16, 36.133.24.0-36.133.35.255,
> ```

确实是，看着集合是通过了。

bsdcpp, 2024-10-23:
> 确实是，看着集合是通过了。

嗯，手头没有原版openwrt，没法给大佬们测试。