immortalwrt/immortalwrt #1512: SNAPSHOT X86 版本无法通过 attendedsysupgrade 升级

Aethersailor, 2024-10-13:
构建固件镜像时出错
`
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://192.168.49.10/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_core
Downloading http://192.168.49.10/snapshots/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_base
Downloading http://192.168.49.10/snapshots/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/targets/x86/64/kmods/6.6.54-1-dfb624cc34d09fc7d8013b7343ec0dc5/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_kmods
Downloading http://192.168.49.10/snapshots/targets/x86/64/kmods/6.6.54-1-dfb624cc34d09fc7d8013b7343ec0dc5/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/packages/x86_64/luci/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_luci
Downloading http://192.168.49.10/snapshots/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_packages
Downloading http://192.168.49.10/snapshots/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_routing
Downloading http://192.168.49.10/snapshots/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/immortalwrt_telephony
Downloading http://192.168.49.10/snapshots/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/build/asu/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../home/build/asu/cache/SNAPSHOT/x86/64/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_install_cmd: Cannot install package ipv6helper.
 * opkg_install_cmd: Cannot install package ipv6helper.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2`

1715173329, 2024-10-13:
现在没有 ipv6helper 这个包了。

closed 2024-10-13 by 1715173329

Aethersailor, 2024-10-14:
> 现在没有 ipv6helper 这个包了。

感谢大佬解答！
我移除ipv6helper（未移除相关依赖）后可以正常更新升级了，可是ipv6helper应该用什么替代呢，我的wan6口现在可选协议中没有DHCPv6了…