immortalwrt/immortalwrt #1534: Photonicat 光影猫 USB不供电

bloodopensun, 2024-10-21:
### Describe the bug

使用电池供电时  usb口不工作  没有电输出
使用外置电源时  usb口正常

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Ariaboard Photonicat

### Image kind

Official downloaded image

### Steps to reproduce

插上usb设备 如u盘 usb无线等 设备不识别
此时插上电源 内核日志能看到新的usb设备
此时拔掉电源 内核日志能看到新的usb设备断开连接

### Actual behaviour

使用电池 无法使用usb接口

### Expected behaviour

使用电池 应该可以使用usb接口

### Additional info

Photonicat官方固件可以在使用电池供电的情况下使用usb
经对比DTS文件  发现少了

`
&gpio4 {
	aux-pwren {
		gpio-hog;
		gpios = <RK_PD2 GPIO_ACTIVE_HIGH>;
		output-high;
		line-name = "aux-pwren";
        };
};`

`
gmac0_xpcsclk: xpcs-gmac0-clock {
		compatible = "fixed-clock";
		clock-frequency = <125000000>;
		clock-output-names = "clk_gmac0_xpcs_mii";
		#clock-cells = <0>;
	};
`

手动加上后  可以在电池供电下使用usb 但是不清楚会不会引发其它问题

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-21:
https://github.com/immortalwrt/immortalwrt/blob/0b7d76327b34f6262729c3fdf0095ed80e6130c0/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts#L402

这里改成 &pcfg_output_high 看看

bloodopensun, 2024-10-21:
试了 表现一样  电池供电时  插上usb设备完全没反应
插上电源  就能在内核日志看到usb设备连上  拔掉电源就看到断开连接

bloodopensun, 2024-10-21:
> https://github.com/immortalwrt/immortalwrt/blob/0b7d76327b34f6262729c3fdf0095ed80e6130c0/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts#L402
> 
> 这里改成 &pcfg_output_high 看看

他不是说usb供电不足  是完全没有电输出
上面说的电源 不是指usb设备的外接电源  是光影猫的充电接口

我插着usb设备不拔的情况下  插拔光影猫的充电线  usb设备就连上/断开
插上充电线 usb设备连上  拔掉充电线 usb设备断开

1715173329, 2024-10-21:
把 vcc5v0_usb_otg 的 vin-supply 改成 vcc5v0_boost 试试呢。手上没这板子了

bloodopensun, 2024-10-21:
我把 光影猫官方那边的 &gpio4 复制过来  就可以用了  只是我不知道它是干啥用的  dts上有4个地方都有用到
不知道会不会有什么不良影响

&gpio4 { aux-pwren { gpio-hog; gpios = <RK_PD2 GPIO_ACTIVE_HIGH>; output-high; line-name = "aux-pwren"; }; };

gmac0_xpcsclk: xpcs-gmac0-clock { compatible = "fixed-clock"; clock-frequency = <125000000>; clock-output-names = "clk_gmac0_xpcs_mii"; #clock-cells = <0>; };

gmac0_xpcsclk 是我对比的时候发现也没有的  dts也有一个地方有用到

bloodopensun, 2024-10-21:
> 把 vcc5v0_usb_otg 的 vin-supply 改成 vcc5v0_boost 试试呢。手上没这板子了

改这个的话 编译失败

Error: arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:406.65-66 syntax error
FATAL ERROR: Unable to parse input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb] Error 1
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1450: dtbs] Error 2
make[5]: Leaving directory '/home/openwrt/immortalwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.167'
make[4]: *** [Makefile:24: /home/openwrt/immortalwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.167/.modules] Error 2
make[4]: Leaving directory '/home/openwrt/immortalwrt/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/openwrt/immortalwrt/target/linux'
time: target/linux/compile#536.51#52.11#588.46
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/openwrt/immortalwrt'
make[1]: *** [target/Makefile:25: /home/openwrt/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/openwrt/immortalwrt'
make: *** [/home/openwrt/immortalwrt/include/toplevel.mk:233: world] Error 2

1715173329, 2024-10-21:
`vin-supply = <&vcc5v0_boost>;`

bloodopensun, 2024-10-21:
> `vin-supply = <&vcc5v0_boost>;`

        vcc5v0_usb_otg: vcc5v0-usb-otg-regulator {
                compatible = "regulator-fixed";
                enable-active-high;
                gpio = <&gpio0 RK_PA5 GPIO_ACTIVE_HIGH>;
                pinctrl-names = "default";
                pinctrl-0 = <&vcc5v0_usb_otg_en>;
                regulator-name = "vcc5v0_usb_otg";
                regulator-min-microvolt = <5000000>;
                regulator-max-microvolt = <5000000>;
                vin-supply = <&vcc5v0_boost>;
        };

bloodopensun, 2024-10-22:
> `vin-supply = <&vcc5v0_boost>;`

不好意思 可能是前面改的时候动到了406行
我把整个文件还原
然后只修改了vin-supply = <&vcc5v0_boost>;
可以编译成功

今天出行需要用到设备  暂时没测固件

1715173329, 2024-10-22:
```patch
diff --git a/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts b/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
index 93b1ec8753..d694a16596 100644
--- a/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
+++ b/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
@@ -57,6 +57,15 @@
 		reset-gpios = <&gpio2 RK_PB1 GPIO_ACTIVE_LOW>;
 	};
 
+	vccin_5v: regulator-vccin-5v {
+		compatible = "regulator-fixed";
+		regulator-name = "vccin_5v";
+		regulator-always-on;
+		regulator-boot-on;
+		regulator-min-microvolt = <5000000>;
+		regulator-max-microvolt = <5000000>;
+	};
+
 	vcc_sysin: vcc-sysin-regulator {
 		compatible = "regulator-fixed";
 		regulator-name = "vcc_sysin";
@@ -64,6 +73,7 @@
 		regulator-boot-on;
 		regulator-min-microvolt = <5000000>;
 		regulator-max-microvolt = <5000000>;
+		vin-supply = <&vccin_5v>;
 	};
 
 	vcc_syson: vcc-syson-regulator {
@@ -113,6 +123,7 @@
 		regulator-name = "vcc3v3_ngff";
 		regulator-min-microvolt = <3300000>;
 		regulator-max-microvolt = <3300000>;
+		vin-supply = <&vccin_5v>;
 	};
 
 	/* pi6c pcie clock generator */
@@ -151,27 +162,29 @@
 		vin-supply = <&vcc_3v3>;
 	};
 
-	vcc5v0_boost: vcc5v0-boost-regulator {
+	vcc5v0_usb_otg: vcc5v0-usb-otg-regulator {
 		compatible = "regulator-fixed";
 		enable-active-high;
-		gpio = <&gpio4 RK_PD2 GPIO_ACTIVE_HIGH>;
+		gpio = <&gpio0 RK_PA5 GPIO_ACTIVE_HIGH>;
 		pinctrl-names = "default";
-		pinctrl-0 = <&vcc5v0_boost_en>;
-		regulator-name = "vcc5v0_boost";
+		pinctrl-0 = <&vcc5v0_usb_otg_en>;
+		regulator-name = "vcc5v0_usb_otg";
 		regulator-min-microvolt = <5000000>;
 		regulator-max-microvolt = <5000000>;
+		vin-supply = <&vccin_5v>;
 	};
 
-	vcc5v0_usb_otg: vcc5v0-usb-otg-regulator {
+	/* actually fed by vccin_5v, dependent on vcc5v0_usb_otg */
+	vcc5v0_boost: vcc5v0-boost-regulator {
 		compatible = "regulator-fixed";
 		enable-active-high;
-		gpio = <&gpio0 RK_PA5 GPIO_ACTIVE_HIGH>;
+		gpio = <&gpio4 RK_PD2 GPIO_ACTIVE_HIGH>;
 		pinctrl-names = "default";
-		pinctrl-0 = <&vcc5v0_usb_otg_en>;
-		regulator-name = "vcc5v0_usb_otg";
+		pinctrl-0 = <&vcc5v0_boost_en>;
+		regulator-name = "vcc5v0_boost";
 		regulator-min-microvolt = <5000000>;
 		regulator-max-microvolt = <5000000>;
-		vin-supply = <&vcc_sysin>;
+		vin-supply = <&vcc5v0_usb_otg>;
 	};
 
 	vcca_1v8: vcca-1v8-regulator {
@@ -570,7 +583,7 @@
 };
 
 &usb2phy1_otg {
-	phy-supply = <&vcc5v0_usb_otg>;
+	phy-supply = <&vcc5v0_boost>;
 	status = "okay";
 };
 
```
有时间的话试试这个 patch

bloodopensun, 2024-10-24:
> 把 vcc5v0_usb_otg 的 vin-supply 改成 vcc5v0_boost 试试呢。手上没这板子了

使用23.05.4 dts只修改vcc5v0_usb_otg 的 vin-supply usb电池状态可以使用了 有供电了

bloodopensun, 2024-10-24:
> ```diff
> diff --git a/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts b/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
> index 93b1ec8753..d694a16596 100644
> --- a/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
> +++ b/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts
> @@ -57,6 +57,15 @@
>  		reset-gpios = <&gpio2 RK_PB1 GPIO_ACTIVE_LOW>;
>  	};
>  
> +	vccin_5v: regulator-vccin-5v {
> +		compatible = "regulator-fixed";
> +		regulator-name = "vccin_5v";
> +		regulator-always-on;
> +		regulator-boot-on;
> +		regulator-min-microvolt = <5000000>;
> +		regulator-max-microvolt = <5000000>;
> +	};
> +
>  	vcc_sysin: vcc-sysin-regulator {
>  		compatible = "regulator-fixed";
>  		regulator-name = "vcc_sysin";
> @@ -64,6 +73,7 @@
>  		regulator-boot-on;
>  		regulator-min-microvolt = <5000000>;
>  		regulator-max-microvolt = <5000000>;
> +		vin-supply = <&vccin_5v>;
>  	};
>  
>  	vcc_syson: vcc-syson-regulator {
> @@ -113,6 +123,7 @@
>  		regulator-name = "vcc3v3_ngff";
>  		regulator-min-microvolt = <3300000>;
>  		regulator-max-microvolt = <3300000>;
> +		vin-supply = <&vccin_5v>;
>  	};
>  
>  	/* pi6c pcie clock generator */
> @@ -151,27 +162,29 @@
>  		vin-supply = <&vcc_3v3>;
>  	};
>  
> -	vcc5v0_boost: vcc5v0-boost-regulator {
> +	vcc5v0_usb_otg: vcc5v0-usb-otg-regulator {
>  		compatible = "regulator-fixed";
>  		enable-active-high;
> -		gpio = <&gpio4 RK_PD2 GPIO_ACTIVE_HIGH>;
> +		gpio = <&gpio0 RK_PA5 GPIO_ACTIVE_HIGH>;
>  		pinctrl-names = "default";
> -		pinctrl-0 = <&vcc5v0_boost_en>;
> -		regulator-name = "vcc5v0_boost";
> +		pinctrl-0 = <&vcc5v0_usb_otg_en>;
> +		regulator-name = "vcc5v0_usb_otg";
>  		regulator-min-microvolt = <5000000>;
>  		regulator-max-microvolt = <5000000>;
> +		vin-supply = <&vccin_5v>;
>  	};
>  
> -	vcc5v0_usb_otg: vcc5v0-usb-otg-regulator {
> +	/* actually fed by vccin_5v, dependent on vcc5v0_usb_otg */
> +	vcc5v0_boost: vcc5v0-boost-regulator {
>  		compatible = "regulator-fixed";
>  		enable-active-high;
> -		gpio = <&gpio0 RK_PA5 GPIO_ACTIVE_HIGH>;
> +		gpio = <&gpio4 RK_PD2 GPIO_ACTIVE_HIGH>;
>  		pinctrl-names = "default";
> -		pinctrl-0 = <&vcc5v0_usb_otg_en>;
> -		regulator-name = "vcc5v0_usb_otg";
> +		pinctrl-0 = <&vcc5v0_boost_en>;
> +		regulator-name = "vcc5v0_boost";
>  		regulator-min-microvolt = <5000000>;
>  		regulator-max-microvolt = <5000000>;
> -		vin-supply = <&vcc_sysin>;
> +		vin-supply = <&vcc5v0_usb_otg>;
>  	};
>  
>  	vcca_1v8: vcca-1v8-regulator {
> @@ -570,7 +583,7 @@
>  };
>  
>  &usb2phy1_otg {
> -	phy-supply = <&vcc5v0_usb_otg>;
> +	phy-supply = <&vcc5v0_boost>;
>  	status = "okay";
>  };
>  
> ```
> 
> 有时间的话试试这个 patch

我对比了一下 23.05.4 除了红绿色的项 五色的项也跟23.05.4 值不一样  我不太清楚怎么改
因为我没找到a/target 跟 b/target 能否发一下完整的dts?

1715173329, 2024-10-24:
基于最新的 master 分支

1715173329, 2024-10-24:
> 使用23.05.4 dts只修改vcc5v0_usb_otg 的 vin-supply usb电池状态可以使用了 有供电了

确定是 regulator 配置问题。

另外上面的那个补丁最后一处改错了，应改 usb2phy0_otg 的 vin-supply 而不是 usb2phy1_otg。

bloodopensun, 2024-10-24:
> > 使用23.05.4 dts只修改vcc5v0_usb_otg 的 vin-supply usb电池状态可以使用了 有供电了
> 
> 确定是 regulator 配置问题。
> 
> 另外上面的那个补丁最后一处改错了，应改 usb2phy0_otg 的 vin-supply 而不是 usb2phy1_otg。

上面那个是你发的  让我有时间测一下的  我没找到完整的文件内容
因为一个是a/xx 一个是b/xx
我尝试对比23.05.4的dts修改  发现不止是红色 绿色 有改动  包括部份没有背景颜色的都跟23.05.4的dts不一样
我不知道需要改那些  所以想看看你有没有完整的dts文件

另外 如下修改usb可以在电池供电情况下使用

> 我把 光影猫官方那边的 &gpio4 复制过来 就可以用了 只是我不知道它是干啥用的 dts上有4个地方都有用到 不知道会不会有什么不良影响
>
> &gpio4 { aux-pwren { gpio-hog; gpios = <RK_PD2 GPIO_ACTIVE_HIGH>; output-high; line-name = "aux-pwren"; }; };
>
> gmac0_xpcsclk: [...]

以及如下修改  也可以使用

> 把 vcc5v0_usb_otg 的 vin-supply 改成 vcc5v0_boost 试试呢。手上没这板子了

如下修改没有效果

> https://github.com/immortalwrt/immortalwrt/blob/0b7d76327b34f6262729c3fdf0095ed80e6130c0/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts#L402
> 
> 这里改成 &pcfg_output_high 看看

bloodopensun, 2024-10-24:
> 基于最新的 master 分支

master 分支的话 pcat-manager 编译失败  原因是gpio(libgpiod)的包 版本过高  有些方法弃用了
手动降级后 它调用 pmu 又有问题  但是pcat-manager没有具体的日志输出  不清楚是什么原因

bloodopensun, 2024-10-24:
有群友反馈说 hdmi也有问题 没有输出
我试了23.05.4 确实没有输出 光影猫官方固件是有输出的
不过他说替换成光影猫官方的dts  就可以解决
我对比了一下两边的gpu hdmi配置  是一样的
我替换dts后 hdmi也还是没有输出

bloodopensun, 2024-10-25:
> 有群友反馈说 hdmi也有问题 没有输出 我试了23.05.4 确实没有输出 光影猫官方固件是有输出的 不过他说替换成光影猫官方的dts 就可以解决 我对比了一下两边的gpu hdmi配置 是一样的 我替换dts后 hdmi也还是没有输出

LZMA=staging_dir/host/bin/lzma
LZMA e ariaboard_photonicat-kernel.bin -lc1 -lp2 -pb2 ariaboard_photonicat-kernel.bin.new
MV ariaboard_photonicat-kernel.bin.new ariaboard_photonicat-kernel.bin
PATH=build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150/scripts/dtc:$PATH
MKIMAGE=./staging_dir/host/bin/mkimage
MKIMAGE -f ariaboard_photonicat-kernel.bin.its ariaboard_photonicat-kernel.bin.new

他说他是这么解决hdmi问题的  dtc这块完全没了解过

RabbitYilia, 2024-10-25:
> > 使用23.05.4 dts只修改vcc5v0_usb_otg 的 vin-supply usb电池状态可以使用了 有供电了
> 
> 确定是 regulator 配置问题。
> 
> 另外上面的那个补丁最后一处改错了，应改 usb2phy0_otg 的 vin-supply 而不是 usb2phy1_otg。

该补丁能合并至主线吗？另外，感谢您的工作，我这里有闲置的光影猫，可提供测试。

bloodopensun, 2024-10-25:
> > > 使用23.05.4 dts只修改vcc5v0_usb_otg 的 vin-supply usb电池状态可以使用了 有供电了
> > 
> > 
> > 确定是 regulator 配置问题。
> > 另外上面的那个补丁最后一处改错了，应改 usb2phy0_otg 的 vin-supply 而不是 usb2phy1_otg。
> 
> 该补丁能合并至主线吗？另外，感谢您的工作，我这里有闲置的光影猫，可提供测试。

目前有两种办法可以在使用电池的情况下使用usb

第一种是在dts文件末尾增加 

&gpio4 {
	aux-pwren {
		gpio-hog;
		gpios = <RK_PD2 GPIO_ACTIVE_HIGH>;
		output-high;
		line-name = "aux-pwren";
        };
};

第二种是把 vcc5v0_usb_otg 的 vin-supply 改成 vcc5v0_boost 

两种都测试过可用  

第一种来源是对比 https://github.com/photonicat/rockchip_rk3568_openwrt/blob/r2404/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-photonicat-usb30.dts

第二种是天灵提供的

RabbitYilia, 2024-10-25:
谢谢，一会儿我把机器翻出来试一下

closed 2024-10-25 by 1715173329

1715173329, 2024-10-25:
至于所谓 HDMI，我觉得你不应该期待路由器系统会默认启用视频输出，尽管目前有一些例外。

bloodopensun, 2024-10-26:
使用master的dts文件 会导致sd卡协商失败

[  143.433711] mmc_host mmc1: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
[  143.712457] mmc1: error -110 whilst initialising SD card
[  144.213896] dwmmc_rockchip fe2b0000.mmc: Busy; trying anyway
[  144.714422] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[  144.729010] mmc_host mmc1: Bus speed (slot 0) = 375000Hz (slot req 375000Hz, actual 375000HZ div = 0)
[  145.256285] dwmmc_rockchip fe2b0000.mmc: Busy; trying anyway
[  145.756816] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)

去掉dts文件中 sdmmc0 的 sd-uhs-sdr50; 就可以使用sd卡
&sdmmc0 {
	bus-width = <4>;
	cap-sd-highspeed;
	cd-gpios = <&gpio0 RK_PB5 GPIO_ACTIVE_LOW>;
	disable-wp;
	no-1-8-v;
	pinctrl-names = "default";
	pinctrl-0 = <&sdmmc0_bus4 &sdmmc0_clk &sdmmc0_cmd>;
	sd-uhs-sdr50;
	vmmc-supply = <&vcc3v3_sd>;
	vqmmc-supply = <&vcc_3v3>;
	status = "okay";
};

[   93.292312] mmc_host mmc1: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
[   93.544743] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[   93.546070] mmc1: new high speed SDXC card at address aaaa
[   93.547922] mmcblk1: mmc1:aaaa SR256 238 GiB 
[   93.550366]  mmcblk1: p1
[   93.874855] EXT4-fs (mmcblk1p1): mounted filesystem with ordered data mode. Opts: . Quota mode: none.

1715173329, 2024-10-26:
fixed

commit 76fc121e (2024-10-25): rockchip: fix usb power for ariaboard photonicat

Fixes: #1534
commit a3403720 (2024-10-25): rockchip: fix usb power for ariaboard photonicat

Fixes: #1534

(cherry picked from commit 76fc121e8f40c1da7008ccfd04462cc8b9d5e3e2)