immortalwrt/immortalwrt #1535: 网卡频繁掉线

WangNian007, 2024-10-21:
### Describe the bug

网卡断联，系统日志有网卡的down和up日志

Mon Oct 21 20:18:28 2024 kern.info kernel: [3236341.778824] br-lan: port 1(eth1) entered blocking state
Mon Oct 21 20:18:28 2024 kern.info kernel: [3236341.784807] br-lan: port 1(eth1) entered forwarding state
Mon Oct 21 20:18:28 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Oct 21 21:02:00 2024 daemon.notice netifd: Interface 'lan' is now down
Mon Oct 21 21:02:00 2024 daemon.notice ttyd[16577]: [2024/10/21 21:02:00:7751] N: rops_handle_POLLIN_netlink: DELADDR
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238953.888928] br-lan: port 3(eth3) entered disabled state
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238953.894800] br-lan: port 2(eth2) entered disabled state
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238953.900634] br-lan: port 1(eth1) entered disabled state
Mon Oct 21 21:02:00 2024 daemon.notice ttyd[16577]: [2024/10/21 21:02:00:7945] N: rops_handle_POLLIN_netlink: DELADDR
Mon Oct 21 21:02:00 2024 daemon.notice ttyd[16577]: [2024/10/21 21:02:00:7945] N: rops_handle_POLLIN_netlink: DELADDR
Mon Oct 21 21:02:00 2024 daemon.notice ttyd[16577]: [2024/10/21 21:02:00:7946] N: rops_handle_POLLIN_netlink: DELADDR
Mon Oct 21 21:02:00 2024 daemon.warn odhcpd[2926]: No default route present, setting ra_lifetime to zero!
Mon Oct 21 21:02:00 2024 daemon.err odhcpd[2926]: Failed to send to ff02::1%lan@br-lan (Network unreachable)
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238953.907559] device eth1 left promiscuous mode
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238953.912579] br-lan: port 1(eth1) entered disabled state
Mon Oct 21 21:02:00 2024 user.notice mwan3-hotplug[2512]: mwan3 hotplug on lan not called because interface disabled
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238954.052436] device eth2 left promiscuous mode
Mon Oct 21 21:02:00 2024 kern.info kernel: [3238954.057566] br-lan: port 2(eth2) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.232488] device eth3 left promiscuous mode
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.237682] br-lan: port 3(eth3) entered disabled state
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' is disabled
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.657366] br-lan: port 1(eth1) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.663316] br-lan: port 1(eth1) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.669284] device eth1 entered promiscuous mode
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.751652] br-lan: port 2(eth2) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.757634] br-lan: port 2(eth2) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.763737] device eth2 entered promiscuous mode
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.853451] br-lan: port 3(eth3) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.859442] br-lan: port 3(eth3) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.865507] device eth3 entered promiscuous mode
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.871554] br-lan: port 3(eth3) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.877480] br-lan: port 3(eth3) entered forwarding state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.883661] br-lan: port 2(eth2) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.889613] br-lan: port 2(eth2) entered forwarding state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.895713] br-lan: port 1(eth1) entered blocking state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238954.901594] br-lan: port 1(eth1) entered forwarding state
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' is enabled
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' is setting up now
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' is now up
Mon Oct 21 21:02:01 2024 daemon.notice netifd: bridge 'br-lan' link is down
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Network device 'eth1' link is down
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Network device 'eth2' link is down
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Network device 'eth3' link is down
Mon Oct 21 21:02:01 2024 daemon.notice netifd: bridge 'br-lan' link is up
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238955.017799] br-lan: port 1(eth1) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238955.023975] br-lan: port 2(eth2) entered disabled state
Mon Oct 21 21:02:01 2024 kern.info kernel: [3238955.030160] br-lan: port 3(eth3) entered disabled state
Mon Oct 21 21:02:01 2024 daemon.notice netifd: bridge 'br-lan' link is down
Mon Oct 21 21:02:01 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Oct 21 21:02:02 2024 user.notice mwan3-hotplug[2749]: mwan3 hotplug on lan not called because interface disabled
Mon Oct 21 21:02:02 2024 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Oct 21 21:02:02 2024 authpriv.info dropbear[3894]: Early exit: Terminated by signal
Mon Oct 21 21:02:02 2024 authpriv.warn dropbear[3526]: Failed listening on '22': Error listening: Address not available
Mon Oct 21 21:02:02 2024 authpriv.warn dropbear[3526]: Failed listening on '22': Error listening: Address not available
Mon Oct 21 21:02:02 2024 authpriv.info dropbear[3526]: Not backgrounding
Mon Oct 21 21:02:04 2024 kern.info kernel: [3238957.827847] igc 0000:02:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Mon Oct 21 21:02:04 2024 kern.info kernel: [3238957.837096] br-lan: port 1(eth1) entered blocking state
Mon Oct 21 21:02:04 2024 kern.info kernel: [3238957.842887] br-lan: port 1(eth1) entered forwarding state
Mon Oct 21 21:02:04 2024 daemon.notice netifd: Network device 'eth1' link is up
Mon Oct 21 21:02:04 2024 daemon.notice netifd: bridge 'br-lan' link is up
Mon Oct 21 21:02:04 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Oct 21 21:02:05 2024 kern.info kernel: [3238958.297894] igc 0000:04:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Mon Oct 21 21:02:05 2024 kern.info kernel: [3238958.307152] br-lan: port 3(eth3) entered blocking state
Mon Oct 21 21:02:05 2024 kern.info kernel: [3238958.312967] br-lan: port 3(eth3) entered forwarding state
[... 621 lines cut ...]
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PARTED_READLINE=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_SING_BOX_BUILD_CLASH_API=y
CONFIG_SING_BOX_BUILD_ECH=y
CONFIG_SING_BOX_BUILD_GVISOR=y
CONFIG_SING_BOX_BUILD_QUIC=y
CONFIG_SING_BOX_BUILD_UTLS=y
CONFIG_SING_BOX_BUILD_WIREGUARD=y
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
# CONFIG_SING_BOX_BUILD_ACME is not set
# CONFIG_SING_BOX_BUILD_REALITY_SERVER is not set
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

winner890302, 2024-11-29:
我也是4125 ，这边的23.05 也会掉。    换了LEDE 固件，  一切正常

WangNian007, 2024-11-29:
我是因为网线的问题，换了网线再也没掉线过

WangNian007, 2024-11-29:
> 我也是4125 i225网卡 ，这边的23.05 也会掉。 换了其他 固件， 一切正常
我是因为网线的问题，换了网线再也没掉线过

closed 2024-11-29 by WangNian007