immortalwrt/immortalwrt #1540: 树莓派cm4无法使用mt7921

1967937965, 2024-10-24:
### Describe the bug

已经尝试了更换硬件，重新编译固件均无效，已经排除硬件故障，根据系统日志mt7921e: probe of 0000:04:00.0 failed with error -12显示wifi设备初始化失败 [日志.txt](https://github.com/user-attachments/files/17508814/default.txt)

### ImmortalWrt version

master

### ImmortalWrt release

master

### ImmortalWrt target/subtarget

master

### Device

树莓派CM4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

libriunk, 2024-10-24:
[参考这个了吗](https://github.com/immortalwrt/user-FAQ/blob/main/%E5%B0%8F%E7%99%BD%E9%97%AE%E9%A2%98%E9%80%9F%E6%9F%A5.md#%E5%8B%BE%E9%80%89%E6%97%A0%E7%BA%BF%E7%BD%91%E5%8D%A1%E9%A9%B1%E5%8A%A8%E4%BB%A5mt7922%E5%92%8Cmt7921%E4%B8%BA%E4%BE%8B)

编译或构建时请带上pciutils 开机ssh进去看看lspci

1967937965, 2024-10-24:
在x86架构使用是没有问题的，编译时已经勾选7921e，lspci可以正常识别7921网卡

> [参考这个了吗](https://github.com/immortalwrt/user-FAQ/blob/main/%E5%B0%8F%E7%99%BD%E9%97%AE%E9%A2%98%E9%80%9F%E6%9F%A5.md#%E5%8B%BE%E9%80%89%E6%97%A0%E7%BA%BF%E7%BD%91%E5%8D%A1%E9%A9%B1%E5%8A%A8%E4%BB%A5mt7922%E5%92%8Cmt7921%E4%B8%BA%E4%BE%8B)
> 
> 编译或构建时请带上pciutils 开机ssh进去看看lspci

在x86架构使用是没有问题的，编译时已经勾选7921e，lspci可以正常识别7921网卡

closed 2024-10-24 by 1967937965