immortalwrt/immortalwrt #1544: The Wlan module on BPI M7(Armsom Sige 7) cannot work

RabbitYilia, 2024-10-25:
### Describe the bug

The Wlan module on BPI M7(Armsom Sige 7) cannot work

Wed Oct 23 21:20:32 2024 kern.info kernel: [   12.995444] brcmfmac 0003:31:00.0: enabling device (0000 -> 0002)
Wed Oct 23 21:20:32 2024 kern.info kernel: [   13.104153] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43752-pcie for chip BCM43752/2
Wed Oct 23 21:20:32 2024 kern.warn kernel: [   13.105563] brcmfmac 0003:31:00.0: Direct firmware load for brcm/brcmfmac43752-pcie.armsom,sige7.bin failed with error -2
Wed Oct 23 21:20:32 2024 kern.warn kernel: [   13.106530] brcmfmac 0003:31:00.0: Falling back to sysfs fallback for: brcm/brcmfmac43752-pcie.armsom,sige7.bin
Wed Oct 23 21:20:32 2024 user.info kernel: [   13.109320] kmodloader: done loading kernel modules from /etc/modules.d/*

Wed Oct 23 21:21:19 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.bin: No such file or directory
Wed Oct 23 21:21:43 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.clm_blob: No such file or directory
Wed Oct 23 21:21:43 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.txcap_blob: No such file or directory
Wed Oct 23 21:21:43 2024 kern.warn kernel: [   85.246351] brcmfmac 0003:31:00.0: Direct firmware load for brcm/brcmfmac43752-pcie.txcap_blob failed with error -2
Wed Oct 23 21:21:43 2024 kern.warn kernel: [   85.247288] brcmfmac 0003:31:00.0: Falling back to sysfs fallback for: brcm/brcmfmac43752-pcie.txcap_blob
Wed Oct 23 21:21:43 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.txcap_blob: No such file or directory
Wed Oct 23 21:21:43 2024 kern.info kernel: [   85.527351] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Wed Oct 23 21:21:43 2024 kern.info kernel: [   85.530147] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43752/2 wl0: Jul 12 2022 18:54:54 version 18.35.387.23.146 (g412cc5ec) FWID 01-93c53be6
Wed Oct 23 21:22:14 2024 daemon.notice netifd: radio0 (3444): WARNING: Variable 'data' does not exist or is not an array/object
Wed Oct 23 21:22:14 2024 daemon.notice hostapd: Set new config for phy phy0:
Wed Oct 23 21:22:14 2024 daemon.notice wpa_supplicant[1575]: Set new config for phy phy0
Wed Oct 23 21:22:14 2024 daemon.notice wpa_supplicant[1575]: Set new config for phy phy0
Wed Oct 23 21:22:14 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Wed Oct 23 21:22:14 2024 daemon.notice hostapd: Restart interface for phy phy0
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.184788] brcmfmac 0003:31:00.0 phy0-ap0: renamed from wlan0
Wed Oct 23 21:22:14 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[SHORT-GI-80][SU-BEAMFORMER][MU-BEAMFORMEE][SOUNDING-DIMENSION-2][MAX-A-MPDU-LEN-EXP0][VHT-LINK-ADAPT-3] channel=36  interface=phy0-ap0 bssid=f2:23:ae:09:84:7d ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 auth_algs=1 wpa=0 ssid=ImmortalWrt bridge=br-lan wds_bridge= snoop_iface=br-lan qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56 #default_macaddr nas_identifier=f223ae09847d  (phy phy0) --> new PHY
Wed Oct 23 21:22:14 2024 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->HT_SCAN
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.258917] br-lan: port 2(phy0-ap0) entered blocking state
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.259415] br-lan: port 2(phy0-ap0) entered disabled state
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.259911] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.260546] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.267622] brcmfmac 0003:31:00.0 phy0-ap0: left allmulticast mode
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.268180] brcmfmac 0003:31:00.0 phy0-ap0: left promiscuous mode
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.268764] br-lan: port 2(phy0-ap0) entered disabled state
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.291606] br-lan: port 2(phy0-ap0) entered blocking state
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.292102] br-lan: port 2(phy0-ap0) entered disabled state
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.292600] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
Wed Oct 23 21:22:14 2024 kern.info kernel: [  116.293263] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
Wed Oct 23 21:22:14 2024 daemon.notice netifd: Wireless device 'radio0' is now up

.___                               __         .__
|   | _____   _____   ____________/  |______  |  |
|   |/     \ /     \ /  _ \_  __ \   __\__  \ |  |
|   |  Y Y  \  Y Y  (  <_> )  | \/|  |  / __ \|  |__
|___|__|_|  /__|_|  /\____/|__|   |__| (____  /____/
          \/      \/  BE FREE AND UNAFRAID  \/
 -----------------------------------------------------
 ImmortalWrt SNAPSHOT, r32193-fc6edad566
 -----------------------------------------------------
root@OpenWrt:~# cd /lib/firmware/brcm/
root@OpenWrt:/lib/firmware/brcm# ls
BCM4362A2.hcd                        brcmfmac43236b.bin                   brcmfmac43752-pcie.bin               brcmfmac43752-pcie.txt
brcmfmac43143.bin                    brcmfmac43752-pcie.armsom,sige7.txt  brcmfmac43752-pcie.clm_blob
root@OpenWrt:/lib/firmware/brcm# 

It seems that the filename incorrect.I tried fix the bug,but not work.
root@OpenWrt:/lib/firmware/brcm# ln -s brcmfmac43752-pcie.bin brcmfmac43752-pcie.armsom,sige7.bin
root@OpenWrt:/lib/firmware/brcm# ln -s brcmfmac43752-pcie.clm_blob brcmfmac43752-pcie.armsom,sige7.clm_blob

After fix it shows

Wed Oct 23 21:22:31 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.txcap_blob: No such file or directory
Wed Oct 23 21:22:31 2024 kern.warn kernel: [   75.937831] brcmfmac 0003:31:00.0: Direct firmware load for brcm/brcmfmac43752-pcie.txcap_blob failed with error -2
Wed Oct 23 21:22:31 2024 kern.warn kernel: [   75.938759] brcmfmac 0003:31:00.0: Falling back to sysfs fallback for: brcm/brcmfmac43752-pcie.txcap_blob

but there is no file named brcmfmac43752-pcie.txcap_blob.

### ImmortalWrt version

r32193-fc6edad566

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

ArmSoM Sige7

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-25:
I have no idea what do you mean by "cannot work". The log you provided showed the interface was already up.

RabbitYilia, 2024-10-25:
Wed Oct 23 21:21:19 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.bin: No such file or directory
Wed Oct 23 21:21:43 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.clm_blob: No such file or directory
Wed Oct 23 21:21:43 2024 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43752-pcie.armsom,sige7.txcap_blob: No such file or directory
这里，编译出来的固件寻找43752驱动找的是brcmfmac43752-pcie.armsom,sige7* 但是在lib/firmware/brcm目录里只有BCM4362A2.hcd brcmfmac43236b.bin brcmfmac43752-pcie.bin brcmfmac43752-pcie.txt brcmfmac43143.bin brcmfmac43752-pcie.armsom,sige7.txt brcmfmac43752-pcie.clm_blob ，找不到固件文件所以无法驱动，日志报
Wed Oct 23 21:20:32 2024 kern.warn kernel: [ 13.105563] brcmfmac 0003:31:00.0: Direct firmware load for brcm/brcmfmac43752-pcie.armsom,sige7.bin failed with error -2
Wed Oct 23 21:20:32 2024 kern.warn kernel: [ 13.106530] brcmfmac 0003:31:00.0: Falling back to sysfs fallback for: brcm/brcmfmac43752-pcie.armsom,sige7.bin
而且接口页面没有无线选项，采取方法1修复后出现。

解决办法有以下几种
1、做软连接（我目前的解决办法）
 ln -s brcmfmac43752-pcie.bin brcmfmac43752-pcie.armsom,sige7.bin
 ln -s brcmfmac43752-pcie.clm_blob brcmfmac43752-pcie.armsom,sige7.clm_blob
2、Sige7、Bananapi M7 虽然是一个东西，这里明显看到，brcmfmac在找驱动的时候在路径里加了型号，但是bin文件却没加，需要同步一下。

顺便这个板子USB Type-C和光影猫一样似乎是没有供电输出。

RabbitYilia, 2024-10-25:
启动了也好奇怪的，感觉是驱动有问题？
  951.340350] br-lan: port 2(phy0-ap0) entered blocking state
[  951.340846] br-lan: port 2(phy0-ap0) entered disabled state
[  951.341341] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
[  951.341998] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
[  951.362248] cdc_ether 1-1:1.0 usb0: entered allmulticast mode
[  955.971470] ieee80211 phy0: brcmf_configure_wpaie: Invalid key mgmt info
[  955.972066] ieee80211 phy0: brcmf_configure_wpaie: Invalid key mgmt info
[  956.047172] br-lan: port 2(phy0-ap0) entered blocking state
[  956.047667] br-lan: port 2(phy0-ap0) entered forwarding state
[  956.050912] ieee80211 phy0: send_key_to_dongle: wsec_key error (-52)
[  969.033938] brcmfmac 0003:31:00.0 phy0-ap0: left allmulticast mode
[  969.034493] brcmfmac 0003:31:00.0 phy0-ap0: left promiscuous mode
[  969.035074] br-lan: port 2(phy0-ap0) entered disabled state
[  969.470248] ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
[  969.471241] ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
[  970.444885] br-lan: port 2(phy0-ap0) entered blocking state
[  970.445384] br-lan: port 2(phy0-ap0) entered disabled state
[  970.445900] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
[  970.446543] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
[  975.071560] ieee80211 phy0: brcmf_configure_wpaie: Invalid key mgmt info
[  975.072156] ieee80211 phy0: brcmf_configure_wpaie: Invalid key mgmt info
[  975.147579] br-lan: port 2(phy0-ap0) entered blocking state
[  975.148077] br-lan: port 2(phy0-ap0) entered forwarding state
[  975.151842] ieee80211 phy0: send_key_to_dongle: wsec_key error (-52)
[ 1013.940851] brcmfmac 0003:31:00.0 phy0-ap0: left allmulticast mode
[ 1013.941407] brcmfmac 0003:31:00.0 phy0-ap0: left promiscuous mode
[ 1013.942008] br-lan: port 2(phy0-ap0) entered disabled state
[ 1014.782083] ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
[ 1014.783103] ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
[ 1016.199084] br-lan: left allmulticast mode
[ 1016.245560] br-lan: entered allmulticast mode
[ 1016.246119] cdc_ether 1-1:1.0 usb0: left allmulticast mode
[ 1016.293648] cdc_ether 1-1:1.0 usb0: entered allmulticast mode
[ 1042.836060] br-lan: left allmulticast mode
[ 1042.885544] br-lan: entered allmulticast mode
[ 1042.886099] cdc_ether 1-1:1.0 usb0: left allmulticast mode
[ 1042.917533] cdc_ether 1-1:1.0 usb0: entered allmulticast mode
[ 1045.023463] br-lan: port 2(phy0-ap0) entered blocking state
[ 1045.023963] br-lan: port 2(phy0-ap0) entered disabled state
[ 1045.024458] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
[ 1045.025102] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
[ 1049.694260] ieee80211 phy0: brcmf_configure_wpaie: Invalid key mgmt info
[ 1049.772113] br-lan: port 2(phy0-ap0) entered blocking state
[ 1049.772607] br-lan: port 2(phy0-ap0) entered forwarding state
[ 1055.417515] brcmfmac 0003:31:00.0 phy0-ap0: left allmulticast mode
[ 1055.418067] brcmfmac 0003:31:00.0 phy0-ap0: left promiscuous mode
[ 1055.418647] br-lan: port 2(phy0-ap0) entered disabled state
[ 1055.869874] ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
[ 1056.560346] br-lan: port 2(phy0-ap0) entered blocking state
[ 1056.560843] br-lan: port 2(phy0-ap0) entered disabled state
[ 1056.561351] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
[ 1056.561990] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
[ 1061.251331] br-lan: port 2(phy0-ap0) entered blocking state
[ 1061.251832] br-lan: port 2(phy0-ap0) entered forwarding state
[ 1223.186238] brcmfmac 0003:31:00.0 phy0-ap0: left allmulticast mode
[ 1223.186796] brcmfmac 0003:31:00.0 phy0-ap0: left promiscuous mode
[ 1223.187381] br-lan: port 2(phy0-ap0) entered disabled state
[ 1223.595115] br-lan: port 2(phy0-ap0) entered blocking state
[ 1223.595613] br-lan: port 2(phy0-ap0) entered disabled state
[ 1223.596117] brcmfmac 0003:31:00.0 phy0-ap0: entered allmulticast mode
[ 1223.596802] brcmfmac 0003:31:00.0 phy0-ap0: entered promiscuous mode
[ 1223.597397] br-lan: port 2(phy0-ap0) entered blocking state
[ 1223.597887] br-lan: port 2(phy0-ap0) entered forwarding state
[ 1225.449916] br-lan: left allmulticast mode
[ 1225.484758] br-lan: entered allmulticast mode

RabbitYilia, 2024-10-25:
[   89.058616] brcmfmac 0003:31:00.0: Direct firmware load for brcm/brcmfmac43752-pcie.txcap_blob failed with error -2
[   89.059568] brcmfmac 0003:31:00.0: Falling back to sysfs fallback for: brcm/brcmfmac43752-pcie.txcap_blob
[  150.765482] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[  150.768317] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43752/2 wl0: Jul 12 2022 18:54:54 version 18.35.387.23.146 (g412cc5ec) FWID 01-93c53be6
经过测试，驱动加载要等一会儿才能通过重启wifi开启不带密码的AP也是在这里等超时。

RabbitYilia, 2024-10-25:
另外这个板子TypeC OTG口,不能受电，也不能对外供电，是不是和光影猫一样，dts配置不对？