immortalwrt/immortalwrt #1559: k3 划分vlan后导致WiFi无法连接，提示密码错误

fanite, 2024-11-01:
### Describe the bug

k3 划分vlan后导致WiFi无法连接，提示密码错误，去掉WiFi密码可以正常连接和上网。

### ImmortalWrt version

r20074-a8bbadefaf

### ImmortalWrt release

21.02.7

### ImmortalWrt target/subtarget

bcm53xx/generic

### Device

Phicomm K3

### Image kind

Official downloaded image

### Steps to reproduce

1. 设置vlan如下：

   | vlan id | 本地 | lan1 | lan2 | lan3 | wan |
   | :---:      | :----:| :---:  | :---:  | :---:  | ---: |
   |   1       |   √    | u*    |    u*  |  u  * |   -- |
   |   41     |   √    |   --   |   --    |   --  |   t   |
   |  48      |   √    |   --   |   --    |   --  |   t   |

2. 接口处，设置lan的设备为br-lan.1, 设置wan的设备为br-lam.41, iptv接口设备为br-lan.48后电脑连接lan口能成功上网
3. 设置WiFi密码，加密方式为WPA2-PSK
4. 连接WiFi，提示密码错误

### Actual behaviour

```
Fri Nov  1 19:14:20 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
Fri Nov  1 19:14:21 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Fri Nov  1 19:14:21 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Fri Nov  1 19:14:21 2024 daemon.notice netifd: wan (9263): udhcpc: sending renew to server 192.168.1.1
Fri Nov  1 19:14:21 2024 daemon.notice netifd: wan (9263): udhcpc: lease of 192.168.1.9 obtained from 192.168.1.1, lease time 86400
Fri Nov  1 19:14:25 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Fri Nov  1 19:14:29 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Fri Nov  1 19:14:32 2024 daemon.err procd: Got unexpected signal 1
Fri Nov  1 19:14:32 2024 authpriv.info dropbear[8390]: Early exit: Terminated by signal
Fri Nov  1 19:14:32 2024 authpriv.info dropbear[9651]: Not backgrounding
Fri Nov  1 19:14:33 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio0 (9461): Command failed: ubus call hostapd config_set { "phy": "phy0", "config": "", "prev_config": "/var/run/hostapd-phy0.conf" } (Request timed out)
Fri Nov  1 19:14:49 2024 daemon.notice wpa_supplicant[9457]: Set new config for phy phy0
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio1 (9482): Command failed: ubus call hostapd config_set { "phy": "phy1", "config": "", "prev_config": "/var/run/hostapd-phy1.conf" } (Request timed out)
Fri Nov  1 19:14:49 2024 daemon.notice wpa_supplicant[9457]: Set new config for phy phy1
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio0 (9461): command failed: No error information (-524)
Fri Nov  1 19:14:49 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio1 (9482): command failed: No error information (-524)
Fri Nov  1 19:14:49 2024 daemon.notice netifd: Wireless device 'radio1' is now down
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio0 (10107): WARNING: Variable 'data' does not exist or is not an array/object
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio1 (10115): WARNING: Variable 'data' does not exist or is not an array/object
Fri Nov  1 19:14:49 2024 daemon.notice hostapd: Set new config for phy phy0:
Fri Nov  1 19:14:49 2024 daemon.notice wpa_supplicant[9457]: Set new config for phy phy0
Fri Nov  1 19:14:49 2024 daemon.notice hostapd: Set new config for phy phy1:
Fri Nov  1 19:14:49 2024 daemon.notice wpa_supplicant[9457]: Set new config for phy phy1
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio0 (10107): command failed: I/O error (-5)
Fri Nov  1 19:14:49 2024 daemon.notice netifd: radio1 (10115): command failed: I/O error (-5)
Fri Nov  1 19:14:49 2024 daemon.notice wpa_supplicant[9457]: Set new config for phy phy0
Fri Nov  1 19:14:49 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
[... 35 lines cut ...]
Fri Nov  1 19:15:01 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: disassociated
Fri Nov  1 19:15:01 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: disassociated
Fri Nov  1 19:15:02 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: associated
Fri Nov  1 19:15:06 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: disassociated
Fri Nov  1 19:15:37 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: associated
Fri Nov  1 19:15:41 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: disassociated
Fri Nov  1 19:15:42 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: associated
Fri Nov  1 19:15:46 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: disassociated
Fri Nov  1 19:15:46 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: disassociated
Fri Nov  1 19:15:46 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: associated
Fri Nov  1 19:15:50 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: disassociated
Fri Nov  1 19:15:51 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: associated
Fri Nov  1 19:15:55 2024 daemon.info hostapd: phy1-ap0: STA 8e:56:a8:7d:08:93 IEEE 802.11: disassociated
Fri Nov  1 19:16:06 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: associated
Fri Nov  1 19:16:10 2024 daemon.info hostapd: phy0-ap0: STA ce:14:01:45:a4:0d IEEE 802.11: disassociated
Fri Nov  1 19:20:24 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
Fri Nov  1 19:28:30 2024 daemon.warn odhcpd[2048]: A default route is present but there is no public prefix on lan thus we don't announce a default route by setting ra_lifetime to zero!
```

详细系统日志：
[system.log](https://github.com/user-attachments/files/17602711/system.log)

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

initzhu, 2024-11-03:
同志 21.02已经停止维护啦，

请升级到受维护的版本再测试😋

1715173329, 2024-11-07:
大概是上游问题。请测试官方 openwrt 23.05 是否存在此问题。

fanite, 2024-11-14:
> 大概是上游问题。请测试官方 openwrt 23.05 是否存在此问题。

我是现在openwrt 23.05上发现的问题，然后退回到21.02测试也复现同样的问题。可能是DSA的问题；刷lede，使用的是swconfig没有问题，同样不排除是我硬件的问题。

1715173329, 2024-11-14:
请给 openwrt 上游提交 issue。

zl810881283, 2025-03-30:
红米ax6 在使用dsa时也存在相同问题，看样子是一个普遍问题。
固件版本号 
ImmortalWrt SNAPSHOT r0-29a6ff5 / LuCI Master 25.068.28752~c308c80

wen-long, 2025-04-04:
嵌套一层无 VLAN bridge 可以规避这个问题 https://github.com/openwrt/openwrt/issues/10219
正文里的为例, new 一个 Bridge device 名字叫 br-lan1 , Bridge ports 只包含 br-lan.1, wifi ssid 选择网络 br-lan1

hjchjchjc4352, 2025-04-05:
> 嵌套一层无 VLAN bridge 可以规避这个问题 [openwrt/openwrt#10219](https://github.com/openwrt/openwrt/issues/10219) 正文里的为例, new 一个 Bridge device 名字叫 br-lan1 , Bridge ports 只包含 br-lan.1, wifi ssid 选择网络 br-lan1

这样是可以的，但是不同vlan的wifi漫游又有问题了fdb的学习出了问题https://github.com/openwrt/openwrt/issues/11650
换dsa后坑可真多