immortalwrt/immortalwrt #1574: PVE虚拟机安装openwrt，设置网卡直通，OP识别不到网卡

lysgwl, 2024-11-09:
### Describe the bug

PVE虚拟机安装openwrt，PVE设置PCI直通，并在OP虚拟机中设置PCI网卡直通，但经启动后，OP无法正确识别网卡。以下给出了编译openwrt的配置文件，麻烦大佬帮忙给看看吧，谢谢了。
[seed.txt](https://github.com/user-attachments/files/17688701/seed.txt)

### ImmortalWrt version

25.05.4

### ImmortalWrt release

r28061

### ImmortalWrt target/subtarget

399f9a1db3 

### Device

QEMU Standard PC

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-11-09:
把有线网卡驱动全注释了，是没有你喜欢的驱动吗？
还是用旧版本杂交 istoreos 的东西，自行负责。

closed 2024-11-09 by 1715173329

lysgwl, 2024-11-09:
> 把有线网卡驱动全注释了，是没有你喜欢的驱动吗？ 还是用旧版本杂交 istoreos 的东西，自行负责。

木有，绝对木有。设备是畅网的N5105，网卡是i226v, 应该选哪个驱动呢？