immortalwrt/immortalwrt #1575: 硬路由小米AX3000T刷了  23.05.4 (r28061-399f9a1db3) 链接无线的设备和链接有线的设备无法在路由器互通. 防火墙是都可以互相访问的,

cander0815, 2024-11-10:
### Describe the bug

链接无线的设备和链接有线的设备无法在路由器互通

[image]

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

正常配置 并开启了IPV6,同时启用了lucky做转发.

lan区域的防火墙是可以互相通信的.

```
chain accept_from_lan {
     iifname "br-lan" counter packets 2985 bytes 263949 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
 } 
chain accept_to_lan { 
   oifname "br-lan" counter packets 1858 bytes 98965 accept comment "!fw4: accept lan IPv4/IPv6 traffic" 
}
```

如图接口都在lan中
[image]

无线的客户端隔离也是关闭的
[image]

点击重启lan接口,在重启完成后的几秒内可以互相访问,但是几秒后就不行了
[image]

我昨天通过 `logread -f` 没有任何日志输出

请问这是bug吗,或者需要我提供什么配置好排查这个问题?

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-11-28:
无法重现。据称 lucky 有一些怪异行为，不要使用此软件，请从 immortalwrt 官方源下载固件并重新测试。

C-neon, 2024-12-09:
> 无法重现。据称 lucky 有一些怪异行为，不要使用此软件，请从 immortalwrt 官方源下载固件并重新测试。

请问官方源uboot支持出厂1.0.84，an8855的ax3000t吗（目前刷有[hkint的固件](https://github.com/hkint/xiaomi-ax3000t-immortalwrt-hanwckf-firmware-build)）