immortalwrt/immortalwrt #1579: build image failed for the latest nftables code

yyshell, 2024-11-12:
### Describe the bug

here's the  log:
. /home/adam/immortalwrt/include/shell.sh; xzcat /home/adam/immortalwrt/dl/nftables-1.1.1.tar.xz | tar -C /home/adam/immortalwrt/build_dir/target-x86_64_musl/nftables-json/nftables-1.1.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/adam/immortalwrt/build_dir/target-x86_64_musl/nftables-json/nftables-1.1.1

Applying /home/adam/immortalwrt/package/network/utils/nftables/patches/002-nftables-add-fullcone-expression-support.patch using plaintext:
patching file include/linux/netfilter/nf_tables.h
patching file include/statement.h
patching file src/netlink_delinearize.c
patching file src/netlink_linearize.c
patching file src/parser_bison.y
patching file src/scanner.l
patching file src/statement.c
Aborting.  Reject files found.
make[3]: *** [Makefile:91: /home/adam/immortalwrt/build_dir/target-x86_64_musl/nftables-json/nftables-1.1.1/.prepared_037f66f1a978c57394d6f737bb18c32d_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/adam/immortalwrt/package/network/utils/nftables'
time: package/network/utils/nftables/json/compile#0.27#0.18#0.40
    ERROR: package/network/utils/nftables failed to build (build variant: json).
make[2]: *** [package/Makefile:179: package/network/utils/nftables/compile] Error 1
make[2]: Leaving directory '/home/adam/immortalwrt'
make[1]: *** [package/Makefile:173: /home/adam/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/adam/immortalwrt'
make: *** [/home/adam/immortalwrt/include/toplevel.mk:248: world] Error 2

### ImmortalWrt version

snapshot

### ImmortalWrt release

snapshot

### ImmortalWrt target/subtarget

x86/64

### Device

intel n100

### Image kind

Self-built image

### Steps to reproduce

make -j1 V=s

### Actual behaviour

build failed

### Expected behaviour

build successfully

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

guhill1, 2024-11-12:
+1
I have also encountered the same issue.

lirg-111, 2024-11-12:
把package/network/utils/nftables/patches/002-nftables-add-fullcone-expression-support.patch删掉即可

closed 2024-11-12 by 1715173329