immortalwrt/immortalwrt #1586: [ramips][snapshot] JDCloud RE-SP-01B EMMC broken on SNAPSHOT

everything411, 2024-11-13:
### Describe the bug

mount emmc or mkfs on emmc is not possible, much I/O ERROR in dmesg
```
root@ImmortalWrt:~# dmesg
[    0.000000] Linux version 6.6.59 (buildbot@buildbot.infra.immortalwrt.org) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r32342-1602dae0fe) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Nov  5 13:39:28 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is JDCloud RE-SP-01B
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 12 pages/cpu s19104 r8192 d21856 u49152
[    0.000000] pcpu-alloc: s19104 r8192 d21856 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Writing ErrCtl register=00040a60
[    0.000000] Readback ErrCtl register=00040a60
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113536
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 441160K/458752K available (8373K kernel code, 652K rwdata, 1776K rodata, 1284K init, 229K bss, 17592K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008070] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066228] pid_max: default: 32768 minimum: 301
[    0.079697] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086938] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.103669] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.111884] rcu: Hierarchical SRCU implementation.
[    0.116588] rcu:     Max phase no-delay instances is 1000.
[    0.122670] smp: Bringing up secondary CPUs ...
[    0.128055] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.128083] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.128099] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.128150] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.182116] Synchronize counters for CPU 1: done.
[    0.214930] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.214952] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.214966] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.214997] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.273841] Synchronize counters for CPU 2: done.
[    0.304530] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.304553] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.304567] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.304601] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.359042] Synchronize counters for CPU 3: done.
[    0.388907] smp: Brought up 1 node, 4 CPUs
[    0.400036] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.409855] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.420666] pinctrl core: initialized pinctrl subsystem
[    0.429175] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... 363 lines cut ...]
JDCloud RE-SP-01B

### Image kind

Official downloaded image

### Steps to reproduce

mount emmc or mkfs

### Actual behaviour

emmc broken

### Expected behaviour

emmc works

### Additional info

23.05 works well

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

everything411, 2024-11-14:
remove `kmod-mmc-mtk` and install `kmod-sdhci-mt7620` solves this problem

DragonBluep, 2024-11-15:
在`kmod-sdhci-mt7620`驱动下能达到多高的读取和写入速度？上游驱动好像有些延时的计算与旧平台不太兼容，达不到太高的速率。可以尝试把[时钟频率](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ramips/dts/mt7621.dtsi;h=da992bada4f3b196ab8445c8867afbb9eb4e0c4b;hb=HEAD#l368)调低一些，比如 15 MHz。还有驱动的很多参数都可以通过设备树调的，[参考](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/Documentation/devicetree/bindings/mmc/mmc-controller.yaml?h=next-20241114)。这种读取错误极大概率是时序有问题，小概率是初始化参数不对。我对mmc没有太多研究，你可以试着自己捣鼓一下。

everything411, 2024-11-15:
读写大概都在17MB/s上下
```
root@ImmortalWrt:/mnt# echo 3 >/proc/sys/vm/drop_caches
root@ImmortalWrt:/mnt# time dd if=/dev/zero of=/mnt/test bs=256M count=4
4+0 records in
4+0 records out
real    0m 55.49s
user    0m 0.00s
sys     0m 15.38s
root@ImmortalWrt:/mnt# echo 3 >/proc/sys/vm/drop_caches
root@ImmortalWrt:/mnt# time dd if=/mnt/test of=/dev/null
2097152+0 records in
2097152+0 records out
real    1m 3.60s
user    0m 2.13s
sys     0m 16.18s
```

everything411, 2025-01-05:
close this and discuss in upstream https://github.com/openwrt/openwrt/pull/17409

closed 2025-01-05 by everything411