immortalwrt/immortalwrt #1587: [ramips][snapshot and 23.05 and 24.10] MAC Address is broken on JDCloud RE-SP-01B

everything411, 2024-11-13:
### Describe the bug

The changes of `target/linux/ramips/mt7621/base-files/etc/hotplug.d/ieee80211/10_fix_wifi_mac` and `target/linux/ramips/mt7621/base-files/etc/board.d/02_network` in https://github.com/immortalwrt/immortalwrt/commit/c2dfb7e8b7b34daa930d30eead1fd8cebe25edb7 are missing in 23.05, 24.10 and master tree, which leads to random MAC Address.

### ImmortalWrt version

r32342-1602dae0fe

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

JDCloud RE-SP-01B

### Image kind

Official downloaded image

### Steps to reproduce

/

### Actual behaviour

/

### Expected behaviour

/

### Additional info

/

### Diffconfig

```text
/
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-11-13:
The MAC-s should be set by NVMEM, if not, you may check the mac offset defined in dts.

everything411, 2024-11-13:
the mac address is at 0x4429 of /dev/mtd1 according to the dts. it seems that i used to flash a padavan firmware and it has overwritten this partition to store its config. thanks very much.

everything411, 2024-11-13:
64G版本的offset是0x4429，128G的offset是0x442A

everything411, 2024-11-14:
https://github.com/immortalwrt/immortalwrt/commit/c0c480daa7a5f8f58cefcaebc268422e27ada701 后仍不能正确取得mac。原厂config中有多个重复的mac key，会导致macaddr_canonicalize失败
```
root@ImmortalWrt:~# . /lib/functions/system.sh
root@ImmortalWrt:~# mac_dirty=$(strings "/dev/mtd1" | sed -n 's/^'"mac"'=//p')
root@ImmortalWrt:~# echo $mac_dirty
DCD87C57DB5C DCD87C57DB5C DCD87C57DB5C
root@ImmortalWrt:~# [ -n "$mac_dirty" ] && macaddr_canonicalize "$mac_dirty"
root@ImmortalWrt:~# macaddr_canonicalize "$mac_dirty"
root@ImmortalWrt:~# macaddr_canonicalize $mac_dirty
dc:d8:7c:57:db:5c
```

everything411, 2024-11-14:
fixed by https://github.com/immortalwrt/immortalwrt/commit/7ec8108be5ba99148fd7a4fbb316157695a30449

closed 2024-11-14 by everything411