immortalwrt/immortalwrt #1588: ImmortalWrt 23.05.4 dnsmasp无法启动

chris6868, 2024-11-13:
### Describe the bug

安装好后什么都不改，dnsmasp无法启动，/var/etc目录下也没有生成dnsmasq.conf，生成了dnsmasq.conf.cfg01411c，但是dnsmasp没有引用它，手动执行/usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k，可以成功启动dnsmasp

### ImmortalWrt version

23.05.4

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

23.05.4

### Device

Default string Default string

### Image kind

Official downloaded image

### Steps to reproduce

pve lxc方式安装，安装好后启动容器就是这样

### Actual behaviour

pve lxc方式安装，安装好后启动容器就是这样

### Expected behaviour

希望能像22版本那样，安装好后dnsmasp能正常启动

### Additional info

x86-64

### Diffconfig

```text
/bin/ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

chris6868, 2024-11-14:
> 这好像是上游的问题，dnsmaq版本更新后需要手动去 luci 界面创建配置文件

luci里没找到配置的地方呢

chris6868, 2024-11-14:
> 这好像是上游的问题，dnsmaq版本更新后需要手动去 luci 界面创建配置文件

luci里默认创建的是cfg01411c，也就是/var/etc/dnsmasq.conf.cfg01411c，但是dnsmasp并没有引用它启动，dns和dhcp都无法正常工作

chris6868, 2024-11-17:
[image]

注释掉这几行代码，重启dnsmasq，就可以正常启动、分配ip地址和监听53端口了

14790897, 2024-11-27:
我因为这个浪费了好多时间

14790897, 2024-11-27:
> ![1731821431771](https://private-user-images.githubusercontent.com/22971922/386922462-aed6f5a7-5959-4f6d-aff6-cacb47883ca2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2ODE5MDksIm5iZiI6MTczMjY4MTYwOSwicGF0aCI6Ii8yMjk3MTkyMi8zODY5MjI0NjItYWVkNmY1YTctNTk1OS00ZjZkLWFmZjYtY2FjYjQ3ODgzY2EyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA0MjY0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyNzY1ZDE3YzEwZGE5OTFhMTc2OWJiZjJlMWRjNzczMjEzNWI5NDk4ODhmMWRjNzFlM2YxOGY4YTkwMjAzYzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sdFssCrzNoYZsvHblu7IKfZ65BTfYyEnsihqeNtGYJM)
> 
> 注释掉这几行代码，重启dnsmasq，就可以正常启动、分配ip地址和监听53端口了

大佬这个是在哪里进行注释

14790897, 2024-11-27:
> https://www.right.com.cn/forum/thread-8406680-1-1.html

大佬，我用的是docker，不能用那个方法
https://github.com/14790897/OpenWrt-In-Docker

chris6868, 2024-11-27:
> 大佬，我用的是docker，不能用那个方法 https://github.com/14790897/OpenWrt-In-Docker

那就不知道了，注释代码的文件路径是/etc/init.d/dnsmasq

14790897, 2024-11-29:
> 那就不知道了，注释代码的文件路径是/etc/init.d/dnsmasq

我已成功

1715173329, 2024-12-04:
应该说 docker 和 lxc 不是支持的安装方式。可以开特权模式启动看看，或者直接卸载 procd-ujail。

closed 2024-12-04 by 1715173329

14790897, 2024-12-04:
特权模式没有用