immortalwrt/immortalwrt #1591: 360T7编译阿里云webdav报错

liyiranlab, 2024-11-14:
### Describe the bug

360T7编译alyun webdav 报错

### ImmortalWrt version

r32419-83ad25fcaf

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

 mediatek/filogic

### Device

 Qihoo 360T7

### Image kind

Official downloaded image

### Steps to reproduce

添加alyiun webdav 编译就报错。 

### Additional info

kdir -p /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /home/ran/360T7/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.10.11.1~ecca0e77/.pkgdir/kmod-mt7981-firmware/. /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.kmod-mt7981-firmware_installed
echo "kmod-mt7981-firmware" >> /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/mt76.default.install
make[3]: Leaving directory '/home/ran/360T7/immortalwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#1.32#1.80#5.51
make[3]: Entering directory '/home/ran/360T7/immortalwrt/package/libs/nettle'
mkdir -p /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
echo '8' | cmp -s - /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libnettle.version || { echo '8' > /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libnettle.version;  } 
SHELL= flock /home/ran/360T7/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/nettle-3.9.1/.pkgdir/libnettle/. /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.libnettle_installed
echo "libnettle" >> /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/nettle.default.install
make[3]: Leaving directory '/home/ran/360T7/immortalwrt/package/libs/nettle'
time: package/libs/nettle/compile#0.15#0.25#0.84
make[3]: Entering directory '/home/ran/360T7/immortalwrt/package/libs/uclient'
mkdir -p /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
echo '20201210' | cmp -s - /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libuclient.version || { echo '20201210' > /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libuclient.version;  } 
SHELL= flock /home/ran/360T7/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/uclient-2024.10.22~88ae8f20/.pkgdir/libuclient/. /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.libuclient_installed
echo "libuclient" >> /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/uclient.default.install
mkdir -p /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /home/ran/360T7/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/uclient-2024.10.22~88ae8f20/.pkgdir/uclient-fetch/. /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.uclient-fetch_installed
echo "uclient-fetch" >> /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/uclient.default.install
make[3]: Leaving directory '/home/ran/360T7/immortalwrt/package/libs/uclient'
time: package/libs/uclient/compile#0.12#0.23#0.49
make[3]: Entering directory '/home/ran/360T7/immortalwrt/package/luci-app-aliyundrive-webdav'
mkdir -p /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /home/ran/360T7/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/.pkgdir/luci-app-aliyundrive-webdav/. /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.luci-app-aliyundrive-webdav_installed
rm -rf /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn
mkdir -p /home/ran/360T7/immortalwrt/bin/targets/mediatek/filogic/packages /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo
install -d -m0755 /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/etc/uci-defaults
echo "uci set luci.languages.zh_cn='简体中文 (Chinese Simplified)'; uci commit luci" > /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/etc/uci-defaults/luci-i18n-aliyundrive-webdav-zh-cn
install -d -m0755 /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/usr/lib/lua/luci/i18n
po2lmo /home/ran/360T7/immortalwrt/package/luci-app-aliyundrive-webdav/po/zh_Hans/aliyundrive-webdav.po /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/usr/lib/lua/luci/i18n/aliyundrive-webdav.zh-cn.lmo;
find /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/ran/360T7/immortalwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/ran/360T7/immortalwrt/scripts/strip-kmod.sh" PATCHELF="/home/ran/360T7/immortalwrt/staging_dir/host/bin/patchelf" /home/ran/360T7/immortalwrt/scripts/rstrip.sh /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn
install -d -m0755 /home/ran/360T7/immortalwrt/bin/packages/aarch64_cortex-a53/base/tmp
mkdir -p /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/apk-all/luci-i18n-aliyundrive-webdav-zh-cn/
[... 4 lines cut ...]
if [ -n "" ]; then echo  > /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/lib/apk/packages/luci-i18n-aliyundrive-webdav-zh-cn.rusers; fi;
if [ -n "" ]; then echo  > /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/lib/apk/packages/luci-i18n-aliyundrive-webdav-zh-cn.alternatives; fi;
(cd /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn && find . -type f,l -printf "/%P\n" > /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/lib/apk/packages/luci-i18n-aliyundrive-webdav-zh-cn.list)
# Move conffiles to IDIR and build conffiles_static with csums
if [ -f /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/apk-all/luci-i18n-aliyundrive-webdav-zh-cn/conffiles ]; then mv -f /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/apk-all/luci-i18n-aliyundrive-webdav-zh-cn/conffiles /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/lib/apk/packages/luci-i18n-aliyundrive-webdav-zh-cn.conffiles; for file in $(cat /home/ran/360T7/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/luci-app-aliyundrive-webdav/ipkg-all/luci-i18n-aliyundrive-webdav-zh-cn/lib/apk/packages/luci-i18n-aliyundrive-webdav-zh-cn.conffiles); do [ -f /home [... cut ...]
make[2]: *** [package/Makefile:184: package/luci-app-aliyundrive-webdav/compile] Error 1
make[2]: Leaving directory '/home/ran/360T7/immortalwrt'
make[1]: *** [package/Makefile:178: /home/ran/360T7/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ran/360T7/immortalwrt'
make: *** [/home/ran/360T7/immortalwrt/include/toplevel.mk:248: world] Error 2

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

liyiranlab, 2024-11-14:
另一台AX6编译正常

1715173329, 2024-11-14:
第三方插件自行联系第三方处理。

closed 2024-11-14 by 1715173329

Firsgith, 2024-12-11:
打开源码的makefile，把version的2.3.3-1改成2.3.3。不要有“-”。这样应该就可以编译通过了