immortalwrt/immortalwrt #1593: Nanopi系列机型无法启动

smallprogram, 2024-11-16:
### Describe the bug

从[downloads.immortalwrt.org](https://downloads.immortalwrt.org/)网站下载R4S、R5S等固件刷写道tf卡中可以完美启动。但是通过最新源码snapshots编译的R4S、R5S等，刷入TF卡就无法启动，Lan口无法获取IP，SSH无法连接

### ImmortalWrt version

r32428-a0bab0ccc8

### ImmortalWrt release

snapshots

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Nanopi R4S

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-11-17:
No log = no solution.

smallprogram, 2024-11-17:
> No log = no solution.

@1715173329 大佬，不是我不提供日志，是真没有日志，我借了R5S和R4S，把固件刷到TF卡里，根本就起不来。用lean就固件就没问题，R5S有HDMI视频输出，用immortalwrt的源码编译，连视频输出信号都没有

VIKINGYFY, 2024-11-22:
R4S启动不了应该快有半年了，某次uboot更新后就启动不了了，一模一样的情况，R5C正常，最后把R4S卖了。。。

smallprogram, 2024-11-22:
> R4S启动不了应该快有半年了，某次uboot更新后就启动不了了，一模一样的情况，R5C正常，最后把R4S卖了。。。

R5S也不行啊，并且R5S的HDMI都没有视频输出信号

VIKINGYFY, 2024-11-22:
> > R4S启动不了应该快有半年了，某次uboot更新后就启动不了了，一模一样的情况，R5C正常，最后把R4S卖了。。。
> 
> R5S也不行啊，并且R5S的HDMI都没有视频输出信号

我的是R5C，不是R5S

1715173329, 2024-12-03:
No useful information to debug, closing.

closed 2024-12-03 by 1715173329

ssd2077, 2024-12-16:
R4S的话我记得我之前拿官方的r8168替换，然后正常运行。我今天再试试。

ssd2077, 2024-12-16:
测试能用。不过记得把120-r8168-add-LED-configuration-from-OF.patch补丁加回去。不然lan口只亮一个灯。

ssd2077, 2024-12-16:
而且我发现祖传bug（5.x内核没问题）还在。6.x内核，用ipv6 nat。单线程上行速度异常（lan口到外网）。但是udp传输的话，好像没事。