immortalwrt/immortalwrt #1599: cpufreq无法调频，一直保持在最大频率

VIKINGYFY, 2024-11-22:
### Describe the bug

cpufreq无法调频，一直保持在最大频率，需要reload或者restart一下才能正确的动态调频
/etc/init.d/cpufreq reload

### ImmortalWrt version

b8917e7

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

 qualcommax/ipq60xx

### Device

 JDCloud RE-CS-02

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-11-22:
本项目既不支持这样的设备，也未开放 ipq60xx 目标支持。

xlighting2017, 2024-11-22:
> ### Describe the bug
> 
> cpufreq无法调频，一直保持在最大频率，需要reload或者restart一下才能正确的动态调频 /etc/init.d/cpufreq reload

在你自己 合并/添加 的代码（也就是  `/etc/init.d/S95qca-nss-pbuf` ）中， 默认写死了  `scaling_governor 'performance'`, 这时候 频率是保持最大的

[image]

[image]

cpufreq reload 时，会将 `scaling_governor` 改成 `schedutil` , 这时候才有 min / max freq

所以 commit/merge 的时候，最好还是搞清楚, 代码里面都是些啥...