immortalwrt/immortalwrt #1631: [与openwrt主线兼容性问题]Master分支不兼容mtk驱动

DeepDendrite, 2024-12-15:
### Describe the bug

我试图添加MTK的私有驱动：https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/
该驱动在openwrt的主线分支正常编译
在immortalwrt中，手动处理了几个patch文件合并冲突后，在编译feed中的crypto-eip时，报错找不到stdint.h
immortalwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/crypto-eip/ddk/inc/crypto-eip/ddk/basic_defs.h:79:10: fatal error: stdint.h: No such file or directory
   79 | #include <stdint.h>

### ImmortalWrt version

91cfc2b296c7

### ImmortalWrt release

91cfc2b296c7

### ImmortalWrt target/subtarget

91cfc2b296c7

### Device

WR30U

### Image kind

Self-built image

### Steps to reproduce

1.克隆immortalwrt主线仓库
2.参考https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/的master branch编译方法
3.手动解决fstools等patch的合并冲突
4.编译

### Actual behaviour

编译报错，找不到stdint.h，由于编译未完成，因此可能存在别的问题

### Expected behaviour

正常编译

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-12-15:
问 mtk 去

closed 2024-12-15 by 1715173329

DeepDendrite, 2024-12-15:
> 问 mtk 去

MTK保证了上游openwrt主线的兼容性，因此问题肯定是immortalwrt的某些地方偏离了主线的设置从而产生了问题。

与上游的兼容性不属于该项目的考虑范围吗？

1715173329, 2024-12-15:
这既不是本项目的问题，也不是上游的问题，而是一个不受支持的第三方源的问题。
如果你喜欢厂商发布的驱动，并因此遇到任何问题，请联系厂商解决，而不是开源社区。