immortalwrt/immortalwrt #1641: homeproxy

xiaomayi755259, 2024-12-20:
### Describe the bug

编译完固件运行正常，但homeproxy选择gfwlist模式访问不了google,大陆白名单模式可以，请问大佬是怎么回事？

### ImmortalWrt version

23.05.4

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

23.05.4

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xiaomayi755259, 2024-12-20:
[image]

xiaomayi755259, 2024-12-20:
日志这样，远白名单模式日志也是这样

science2468, 2024-12-21:
你访问下被墙的域名。没有sing-box客户端日志，从你发的图片，home proxy本身没问题。

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

1715173329, 2024-12-21:
去 homeproxy 库发 issue。

closed 2024-12-21 by 1715173329