immortalwrt/immortalwrt #1658: 路由器管理页面的http不能够自动跳转到https

ling0z0, 2025-01-05:
### Describe the bug

路由器管理页面的http不能够自动跳转到https
[image]

### ImmortalWrt version

ImmortalWrt 24.10.0-rc3 r32584-afde822cf7f9 / LuCI openwrt-24.10 branch 24.362.20051~34c1f40

### ImmortalWrt release

ImmortalWrt 24.10.0-rc3 r32584-afde822cf7f9 / LuCI openwrt-24.10 branch 24.362.20051~34c1f40

### ImmortalWrt target/subtarget

sorry,not found

### Device

Microsoft Corporation Virtual Machine   Intel(R) N100 x 2C 2T

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

ling0z0, 2025-01-05:
x86/64 ext4 efi

1715173329, 2025-01-05:
没装 luci-ssl。

closed 2025-01-05 by 1715173329