immortalwrt/immortalwrt #1662: 防火墙运行异常

ciscolive, 2025-01-09:
### Describe the bug

您好，反馈一个BUG：官网镜像加载 p2 后，防火墙无法正常运行，请帮忙看看，谢谢。

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

Section @zone[1] (wan) IPv4 fullcone enabled for zone 'wan'
Section passwall2 option 'reload' is not supported by fw4
Section passwall2_server option 'reload' is not supported by fw4
Section passwall2_server specifies unreachable path '/var/etc/passwall2_server.include', ignoring section
Error: No such file or directory
list chain inet passwall2 mangle_prerouting
                ^^^^^^^^^
Include '/var/etc/passwall2.include' failed with exit code 1

### Actual behaviour

防火墙启动报错，实际没有成功运行
root@OpenWrt:~# /etc/init.d/firewall status
active with no instances

### Expected behaviour

正常运行

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xiaobaishu1, 2025-01-09:
只有luci-app-passwall没有luci-app-passwall2！

1715173329, 2025-01-09:
使用第三方源码造成问题与本项目无关。

closed 2025-01-09 by 1715173329