immortalwrt/immortalwrt #1668: 云编译失败

sclsyin, 2025-01-14:
### Describe the bug

云编译失败，报错：sed: can't read /usr/share/wechatpush/wechatpush: No such file or directory

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Self-built image

### Steps to reproduce

[image]

### Actual behaviour

同样的配置文件，前几天编译正常，今天开始就报错了

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2025-01-14:
自己改的，找别人擦屁股可还行？

[image]

sclsyin, 2025-01-14:
额。。。忘记注释这个了。。谢谢。

closed 2025-01-17 by sclsyin