immortalwrt/immortalwrt #1679: 24.10-rc3 mediatek/filogic 无法正常使用 ImageBuilder

iBug, 2025-01-23:
### Describe the bug

使用 ImageBuilder 报错：

```text
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package block-mount.
 * opkg_install_cmd: Cannot install package ipv6helper.
 * opkg_install_cmd: Cannot install package luci-i18n-opkg-zh-cn.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for block-mount:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package block-mount.
```

### ImmortalWrt version

24.10.0-rc3 (r32584-afde822cf7f9)

### ImmortalWrt release

24.10.0-rc3 (r32584-afde822cf7f9)

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Qihoo 360T7

### Image kind

Self-built image

### Steps to reproduce

下载 <https://downloads.immortalwrt.org/releases/24.10.0-rc3/targets/mediatek/filogic/immortalwrt-imagebuilder-24.10.0-rc3-mediatek-filogic.Linux-x86_64.tar.zst> 并解压进入目录，运行 `make image PROFILE=qihoo_360t7` 即可复现报错。

或者在 Firmware Selector 上选择自定义构建（无需修改软件包列表），也可以复现相同报错，如图：

> [image]
>
> [image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-01-24:
已在 24.10.0-rc4 中修复。

closed 2025-01-24 by 1715173329

iBug, 2025-01-24:
我排查到是上游问题 openwrt/openwrt#17664，在上游的 24.10.0-rc6 中已修复