immortalwrt/immortalwrt #1682: 是不是插件仓库还没有弄好？

ling0z0, 2025-01-26:
### Describe the bug

更新软件仓库列表提示有错误

[image]

[image]

### ImmortalWrt version

ImmortalWrt 24.10.0-rc4

### ImmortalWrt release

r32779-d475f4e4007b

### ImmortalWrt target/subtarget

package-manager

### Device

hype-v x86/64

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-01-26:
等镜像站同步更新即可，等不及可以切官方源。

closed 2025-01-26 by 1715173329

ling0z0, 2025-01-26:
> 等镜像站同步更新即可，等不及可以切官方源。

好的，等的，发现问题先反馈一下

shengmodongxi, 2025-01-26:
如果用官方源的话kmods版本对不上
openwrt官方24.10.0-rc4的kmods版本为6.6.67-1,而immortalwrt的kmods版本为6.6.73-1
这样也会导致opkg update失败，无法安装kmod软件包，会导致依赖kmod的软件安装失败如singbox

initzhu, 2025-01-26:
直接切snapshot 或者2410snapshot 问题解决