immortalwrt/immortalwrt #1683: GNUmakefile:108: abort-due-to-no-makefile] Error 1

SuperKali, 2025-01-26:
I trying compiling ImmortalWRT with github worker but with 24.10.0-rc4 i have some issue.

```bash
2025-01-26T11:11:56.2473953Z  make[3] -C tools/libtool compile
2025-01-26T11:11:57.1629359Z     ERROR: tools/libtool failed to build.
2025-01-26T12:22:15.7957461Z make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2025-01-26T12:22:15.7958797Z make: *** [/home/runner/work/BananaWRT/immortalwrt/include/toplevel.mk:233: world] Error 1
2025-01-26T12:22:23.6287132Z make[2]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/scripts/config'
2025-01-26T12:22:23.6287872Z make[2]: 'conf' is up to date.
2025-01-26T12:22:23.6288492Z make[2]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/scripts/config'
2025-01-26T12:22:25.4292418Z make[1]: Entering directory '/home/runner/work/BananaWRT/immortalwrt'
2025-01-26T12:22:27.9549002Z cd "/home/runner/work/BananaWRT/immortalwrt"; git log --format=%h -1 toolchain > /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check
2025-01-26T12:22:27.9624162Z cmp -s /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check || { \
2025-01-26T12:22:27.9626147Z 	rm -rf /home/runner/work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_musl /home/runner/work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl /home/runner/work/BananaWRT/immortalwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl; \
2025-01-26T12:22:27.9627906Z 	mkdir -p /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp; \
2025-01-26T12:22:27.9629421Z 	mv /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check; \
2025-01-26T12:22:27.9630230Z }
2025-01-26T12:22:27.9764836Z make[2]: Entering directory '/home/runner/work/BananaWRT/immortalwrt'
2025-01-26T12:22:30.4668321Z cd "/home/runner/work/BananaWRT/immortalwrt"; git log --format=%h -1 toolchain > /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check
2025-01-26T12:22:30.4743249Z cmp -s /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check || { \
2025-01-26T12:22:30.4746014Z 	rm -rf /home/runner/work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_musl /home/runner/work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl /home/runner/work/BananaWRT/immortalwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl; \
2025-01-26T12:22:30.4747680Z 	mkdir -p /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp; \
2025-01-26T12:22:30.4748685Z 	mv /home/runner/work/BananaWRT/immortalwrt/tmp/.ver_check /home/runner/work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check; \
2025-01-26T12:22:30.4749391Z }
2025-01-26T12:22:30.4808225Z + mkdir -p /home/runner/work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl
2025-01-26T12:22:30.4819645Z + cd /home/runner/work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl
2025-01-26T12:22:30.4820469Z + mkdir -p bin lib stamp usr/include usr/lib
2025-01-26T12:22:30.4832713Z mkdir -p /home/runner/work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/stamp
2025-01-26T12:22:30.4857168Z touch /home/runner/work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/.prepared
2025-01-26T12:22:30.4898563Z + mkdir -p /home/runner/work/BananaWRT/immortalwrt/staging_dir/host
2025-01-26T12:22:30.4909470Z + cd /home/runner/work/BananaWRT/immortalwrt/staging_dir/host
2025-01-26T12:22:30.4910073Z + mkdir -p bin lib stamp usr/include usr/lib
2025-01-26T12:22:30.4923538Z mkdir -p /home/runner/work/BananaWRT/immortalwrt/build_dir/host/stamp /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/include/sys
2025-01-26T12:22:30.4947675Z install -m0644 /home/runner/work/BananaWRT/immortalwrt/tools/include/*.h /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/include/
2025-01-26T12:22:30.4980297Z install -m0644 /home/runner/work/BananaWRT/immortalwrt/tools/include/sys/*.h /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/include/sys/
2025-01-26T12:22:30.5008475Z ln -snf lib /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/lib64
2025-01-26T12:22:30.5031550Z touch /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/.prepared
2025-01-26T12:22:30.6728485Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/flock'
2025-01-26T12:22:30.7546004Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/flock-2.18 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:30.7575280Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/flock'
2025-01-26T12:22:30.7577508Z time: tools/flock/compile#0.07#0.02#0.09
2025-01-26T12:22:30.9269531Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/libdeflate'
2025-01-26T12:22:31.0076956Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/libdeflate-1.22 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:31.0108147Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/libdeflate'
2025-01-26T12:22:31.0110791Z time: tools/libdeflate/compile#0.06#0.03#0.09
2025-01-26T12:22:31.1797989Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/zstd'
2025-01-26T12:22:31.2607844Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/zstd-1.5.6 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:31.2636975Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/zstd'
2025-01-26T12:22:31.2639530Z time: tools/zstd/compile#0.08#0.02#0.09
2025-01-26T12:22:31.4320823Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/sed'
2025-01-26T12:22:31.5130583Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/sed-4.9 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:31.5160394Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/sed'
2025-01-26T12:22:31.5162990Z time: tools/sed/compile#0.07#0.03#0.09
2025-01-26T12:22:31.6864128Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/patch'
2025-01-26T12:22:31.7679835Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/patch-2.7.6 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:31.7709063Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/patch'
2025-01-26T12:22:31.7711493Z time: tools/patch/compile#0.06#0.03#0.09
2025-01-26T12:22:31.9408308Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/tar'
2025-01-26T12:22:32.0218440Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/tar-1.35 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:32.0250626Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/tar'
2025-01-26T12:22:32.0253123Z time: tools/tar/compile#0.07#0.02#0.09
2025-01-26T12:22:32.2022199Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/ninja'
2025-01-26T12:22:32.2874280Z find /home/runner/work/BananaWRT/immortalwrt/build_dir/host/ninja-1.12.1 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
2025-01-26T12:22:32.2905187Z make[3]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt/tools/ninja'
2025-01-26T12:22:32.2907717Z time: tools/ninja/compile#0.07#0.03#0.09
2025-01-26T12:22:32.4507552Z make[3]: Entering directory '/home/runner/work/BananaWRT/immortalwrt/tools/meson'
[... cut ...]
2025-01-26T12:22:36.6478896Z make[1]: *** [tools/Makefile:229: /home/runner/work/BananaWRT/immortalwrt/staging_dir/host/stamp/.tools_compile_nyyyyyyynnyyyyyynyyyyyyynyyynnynyyyyynyyyyyyynyyyyyyyyyynynynnyyyyynyyy] Error 2
2025-01-26T12:22:36.6483121Z make[1]: Leaving directory '/home/runner/work/BananaWRT/immortalwrt'
2025-01-26T12:22:36.6545006Z make: *** [/home/runner/work/BananaWRT/immortalwrt/include/toplevel.mk:233: world] Error 2
```

All host side dependencies are installed, with version 24.10.0-rc3 I don't have this problem, the strange coast is that compiling on arm64 this problem doesn't occur.

How do I solve it.

Thanks

1715173329, 2025-01-26:
Stop abusing gitbub actions, and compile the source on your own device.

SuperKali, 2025-01-26:
> Stop abusing gitbub actions

Abusing? who is abusing github actions? do they exist why not use them? if I over quote I pay, yes? 

then you tell me if this is the way to approach people who report problems to you...

1715173329, 2025-01-26:
No, this is the way you reporting issues to people.
Of course you don't bother to fill out the issue template then how could you find ToS ever exists.

SuperKali, 2025-01-26:
> No, this is the way you reporting issues to people. Of course you don't bother to fill out the issue template then how could you find ToS ever exists.

Explain to me how can I fill in all the fields of the template? Since the problem is in the build phase? Just to give you confirmation, even in my self-hosted server I have the same problem.

The fields required by the template are mandatory and I can't extract many things, that's why I preferred to open an issue starting from 0

If you want to help me thank you, otherwise I'll try to bang my head and understand the problem

1715173329, 2025-01-26:
So you mean you don't know what branch/target/device and build configuration (diffconfig) you are building until you get the final firmware? Good to know.

SuperKali, 2025-01-26:
> So you mean you don't know

I'm not saying I don't know, but the commands in the template I'm assuming have to be launched directly from the router, anyway I reopened the issue #1684  with the info I was able to find, if you want to help me I'll be grateful, in any case thank you for your time