immortalwrt/immortalwrt #1684: Error compiling 24.10.0-rc4 for BPI R3 mini

SuperKali, 2025-01-26:
### Describe the bug

I’m reopening this hoping everything is correct. I’d like to report that I’m experiencing several issues while compiling the firmware for the board in question. It seems that the process fails when it reaches the compilation of libtool.

I am compiling using my self-hosted runners through GitHub Actions.

### ImmortalWrt version

24.10.0-rc4

### ImmortalWrt release

No response

### ImmortalWrt target/subtarget

No response

### Device

Banana Pi R3 Mini

### Image kind

Self-built image

### Steps to reproduce

The workflow clones the repository installs dependencies if needed and compiles the firmware.
When it gets to the libtool compilation it goes into error.

### Actual behaviour

```bash
[ ! -d ./src/ ] || cp -fpR ./src/* /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/000-relocatable.patch using plaintext: 
patching file libtoolize.in
patching file m4/libtool.m4
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/100-libdir-fixes.patch using plaintext: 
patching file build-aux/ltmain.in
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/110-dont-use-target-dir-for-relinking.patch using plaintext: 
patching file build-aux/ltmain.in
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/120-strip-unsafe-dirs-for-relinking.patch using plaintext: 
patching file build-aux/ltmain.in
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/130-trailingslash.patch using plaintext: 
patching file build-aux/ltmain.in
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/140-don-t-quote-SHELL-in-Makefile.am.patch using plaintext: 
patching file Makefile.am
Applying /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool/patches/200-openwrt-branding.patch using plaintext: 
patching file build-aux/ltmain.in
patching file build-aux/funclib.sh
CFLAGS="-O2 -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CPPFLAGS="-I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CXXFLAGS="-O2 -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " LDFLAGS="-L/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/lib " make  -C /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.  uninstall
make[4]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[4]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[4]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7'
make[3]: [Makefile:68: /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.prepared6d4a1f5d04dba1df3f3041fdaeb49c8a_18f1e190c5d53547fed41a3eaa76e9e9] Error 2 (ignored)
(cd /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/share/aclocal/ && rm -f libtool.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)
( cd /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7; AUTOM4TE=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autom4te AUTOCONF=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf AUTOMAKE=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/automake ACLOCAL=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/aclocal AUTOHEADER=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/libtoolize LIBTOOL=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/libtool M4=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true ./bootstrap --copy --force --skip-git --skip-po --gnulib-srcdir=/home/github-worker/acti
bootstrap:   error: Prerequisite '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf' not found. Please install it, or
bootstrap:          'export AUTOCONF=/path/to//home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf'.
bootstrap:   error: Program    Min_version Homepage                            
bootstrap:          -----------------------------------------------------------
bootstrap:          help2man   1.29        http://www.gnu.org/s/help2man       
bootstrap:          make       3.81        http://www.gnu.org/s/make           
bootstrap:          makeinfo   4.8         http://www.gnu.org/s/texinfo        
bootstrap:          xz         4.999.8beta http://tukaani.org/xz               
bootstrap:          autoconf   2.64        http://www.gnu.org/s/autoconf       
bootstrap:          automake   1.11.1      http://www.gnu.org/s/automake       
[... 316 lines cut ...]
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_ROOTFS_PARTSIZE=7100
CONFIG_USE_LLVM_BUILD=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_DIST="ImmortalWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.immortalwrt.org/releases/24.10.0-rc4"
CONFIG_VERSION_SUPPORT_URL=""
# CONFIG_COLLECT_KERNEL_DEBUG is not set
# CONFIG_LPAC_WITH_UQMI is not set
CONFIG_PACKAGE_default-settings=y
CONFIG_PACKAGE_luci=y
```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2025-01-26:
make sure you are following the "Development/Requirements" section of the Immortalwrt project README

SuperKali, 2025-01-26:
Hi,

Thanks for replying me, 

I think I have everything installed:

```bash
root@servr-server:~# sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
  bzip2 ccache clang cmake cpio curl device-tree-compiler ecj fastjar flex gawk gettext gcc-multilib \
  g++-multilib git gnutls-dev gperf haveged help2man intltool lib32gcc-s1 libc6-dev-i386 libelf-dev \
  libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev libmpfr-dev libncurses-dev libpython3-dev \
  libreadline-dev libssl-dev libtool libyaml-dev libz-dev lld llvm lrzsz mkisofs msmtp nano \
  ninja-build p7zip p7zip-full patch pkgconf python3 python3-pip python3-ply python3-docutils \
  python3-pyelftools qemu-utils re2c rsync scons squashfs-tools subversion swig texinfo uglifyjs \
  upx-ucl unzip vim wget xmlto xxd zlib1g-dev zstd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libgnutls28-dev' instead of 'gnutls-dev'
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'genisoimage' instead of 'mkisofs'
autoconf is already the newest version (2.71-2).
automake is already the newest version (1:1.16.5-1.3).
autopoint is already the newest version (0.21-4ubuntu4).
bison is already the newest version (2:3.8.2+dfsg-1build1).
build-essential is already the newest version (12.9ubuntu3).
bzip2 is already the newest version (1.0.8-5build1).
device-tree-compiler is already the newest version (1.6.1-1).
flex is already the newest version (2.6.4-8build2).
g++-multilib is already the newest version (4:11.2.0-1ubuntu1).
gcc-multilib is already the newest version (4:11.2.0-1ubuntu1).
genisoimage is already the newest version (9:1.1.11-3.2ubuntu1).
gettext is already the newest version (0.21-4ubuntu4).
libelf-dev is already the newest version (0.186-1build1).
libgmp3-dev is already the newest version (2:6.2.1+dfsg-3ubuntu1).
libltdl-dev is already the newest version (2.4.6-15build2).
libmpc-dev is already the newest version (1.2.1-2build1).
libmpfr-dev is already the newest version (4.1.0-3build3).
libreadline-dev is already the newest version (8.1.2-1).
libtool is already the newest version (2.4.6-15build2).
libyaml-dev is already the newest version (0.2.2-1build2).
patch is already the newest version (2.7.6-7build2).
python3-docutils is already the newest version (0.17.1+dfsg-2).
python3-ply is already the newest version (3.11-5).
python3-pyelftools is already the newest version (0.27-1).
re2c is already the newest version (3.0-1).
squashfs-tools is already the newest version (1:4.5-3build1).
zstd is already the newest version (1.4.8+dfsg-3build1).
ack is already the newest version (3.5.0-1).
antlr3 is already the newest version (3.5.2-9).
asciidoc is already the newest version (10.1.2-1).
ccache is already the newest version (4.5.1-1).
clang is already the newest version (1:14.0-55~exp2).
ecj is already the newest version (3.16.0-1).
fastjar is already the newest version (2:0.98-7).
gperf is already the newest version (3.1-1build1).
haveged is already the newest version (1.9.14-1ubuntu1).
help2man is already the newest version (1.49.1).
intltool is already the newest version (0.51.0-6).
lld is already the newest version (1:14.0-55~exp2).
llvm is already the newest version (1:14.0-55~exp2).
lrzsz is already the newest version (0.12.21-10).
msmtp is already the newest version (1.8.16-1).
ninja-build is already the newest version (1.10.1-1).
p7zip is already the newest version (16.02+dfsg-8).
p7zip-full is already the newest version (16.02+dfsg-8).
pkgconf is already the newest version (1.8.0-1).
scons is already the newest version (4.0.1+dfsg-2).
swig is already the newest version (4.0.2-1ubuntu1).
texinfo is already the newest version (6.8-4build1).
uglifyjs is already the newest version (3.15.1-1).
upx-ucl is already the newest version (3.96-3).
xmlto is already the newest version (0.0.28-2.1).
binutils is already the newest version (2.38-4ubuntu2.6).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.2).
cpio is already the newest version (2.13+dfsg-7ubuntu0.1).
curl is already the newest version (7.81.0-1ubuntu1.20).
gawk is already the newest version (1:5.1.0-1ubuntu0.1).
git is already the newest version (1:2.34.1-1ubuntu1.12).
lib32gcc-s1 is already the newest version (12.3.0-1ubuntu1~22.04).
libc6-dev-i386 is already the newest version (2.35-0ubuntu3.8).
libglib2.0-dev is already the newest version (2.72.4-0ubuntu2.4).
libgnutls28-dev is already the newest version (3.7.3-4ubuntu1.5).
libncurses-dev is already the newest version (6.3-2ubuntu0.1).
libpython3-dev is already the newest version (3.10.6-1~22.04.1).
libssl-dev is already the newest version (3.0.2-0ubuntu1.18).
nano is already the newest version (6.2-1ubuntu0.1).
python3 is already the newest version (3.10.6-1~22.04.1).
qemu-utils is already the newest version (1:6.2+dfsg-2ubuntu6.24).
rsync is already the newest version (3.2.7-0ubuntu0.22.04.4).
unzip is already the newest version (6.0-26ubuntu3.2).
vim is already the newest version (2:8.2.3995-1ubuntu2.22).
wget is already the newest version (1.21.2-2ubuntu1.1).
xxd is already the newest version (2:8.2.3995-1ubuntu2.22).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9.2).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
subversion is already the newest version (1.14.1-3ubuntu0.22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

1715173329, 2025-01-26:
Please check if `staging_dir/host/bin/autoconf` exists. If not, run `make tools/autoconf/compile` first then re-run make.

SuperKali, 2025-01-26:
> Please check if `staging_dir/host/bin/autoconf` exists.

```bash
➜  bin git:(v24.10.0-rc4) ✗ ls -latrh autoc*
-rwxr-xr-x 1 github-worker github-worker 16K Jan 26 17:13 autoconf
➜  bin git:(v24.10.0-rc4) ✗ 

```

```bash
➜  immortalwrt git:(v24.10.0-rc4) ✗ make tools/autoconf/compile
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
 make[1] tools/autoconf/compile
 make[2] -C tools/flock compile
 make[2] -C tools/libdeflate compile
 make[2] -C tools/zstd compile
 make[2] -C tools/sed compile
 make[2] -C tools/patch compile
 make[2] -C tools/tar compile
 make[2] -C tools/ninja compile
 make[2] -C tools/meson compile
 make[2] -C tools/pkgconf compile
 make[2] -C tools/libressl compile
 make[2] -C tools/expat compile
 make[2] -C tools/xz compile
 make[2] -C tools/zlib compile
 make[2] -C tools/cmake compile
 make[2] -C tools/ccache compile
 make[2] -C tools/m4 compile
 make[2] -C tools/autoconf compile

```

1715173329, 2025-01-26:
Okay, is `staging_dir/host/bin/autoconf` executable?

1715173329, 2025-01-26:
> But shouldn't it work without doing this extra step?

Yes it should, but based on your log there's something wrong with autoconf.

SuperKali, 2025-01-26:
> Okay, is `staging_dir/host/bin/autoconf` executable?

Yes it is executable, as you can see from my previous message where I attached the ls -latrh

1715173329, 2025-01-26:
I mean if the binary runs normally.

Please paste fresh build log of libtool: `make tools/libtool/{clean,compile} V=s` and if possible tell me your OS version.

SuperKali, 2025-01-26:
> I mean if the binary runs normally.

Yes as you can see below:

```bash
➜  bin git:(v24.10.0-rc4) ✗ ./autoconf --version
autoconf (GNU Autoconf) nightly
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<https://gnu.org/licenses/gpl.html>, <https://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.
➜  bin git:(v24.10.0-rc4) ✗ 
```

> fresh build log of libtool: `make tools/libtool/{clean,compile} V=s`

```bash
➜  immortalwrt git:(v24.10.0-rc4) ✗ make tools/libtool/{clean,compile} V=s
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[1]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
cd "/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt"; git log --format=%h -1 toolchain > /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check
cmp -s /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check || { \
	rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl; \
	mkdir -p /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp; \
	mv /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check; \
}
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool'
CFLAGS="-O2 -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CPPFLAGS="-I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CXXFLAGS="-O2 -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " LDFLAGS="-L/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/lib " make  -C /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.  uninstall
make[3]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[3]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[3]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7'
make[2]: [Makefile:60: host-clean-build] Error 2 (ignored)
(cd /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/share/aclocal/ && rm -f libtool.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)
rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7 /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.built
rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/stamp/.libtool_installed 
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool'
time: tools/libtool/clean#0.11#0.02#0.47
make[1]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[1]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
cd "/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt"; git log --format=%h -1 toolchain > /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check
cmp -s /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check || { \
	rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl; \
	mkdir -p /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp; \
[... cut ...]
time: tools/libtool/compile#0.68#0.13#2.62
    ERROR: tools/libtool failed to build.
make[1]: *** [tools/Makefile:233: tools/libtool/compile] Error 1
make[1]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
make: *** [/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/include/toplevel.mk:233: tools/libtool/compile] Error 2

```

> if possible tell me your OS version

I'm using Ubuntu 22.04 on my worker node, i tried on my PC with ubuntu 24.04 and i have the same issue.

SuperKali, 2025-01-26:
[image]

This is currently a worker compiling immortalwrt v24.10.0-rc3 always in my runner, the problem does not occur

1715173329, 2025-01-26:
It's weird as there's no changes for it between 24.10.0 rc3 and rc4, I will try to reproduce the issue on Ubuntu 22.04 later today.

1715173329, 2025-01-27:
Unfortunately I was unable to reproduce this issue on my host (Ubuntu 22.04).

Please apply this patch:
```patch
--- a/tools/libtool/Makefile
+++ b/tools/libtool/Makefile
@@ -29,6 +29,7 @@ define Host/Bootstrap
 		$(AM_TOOL_PATHS) \
 		./bootstrap \
 			--copy \
+			--debug \
 			--force \
 			--skip-git \
 			--skip-po \
```

then re-run `make tools/libtool/{clean,compile} V=s`

SuperKali, 2025-01-27:
> Unfortunately I was unable to reproduce this issue on my host (Ubuntu 22.04).

Strange, also because I tried again from a diffent host with debian 11 and the problem remains the same

> then re-run `make tools/libtool/{clean,compile} V=s`

```bash
github-worker@it-manu-worker-x86:~/actions-runner/_work/BananaWRT/immortalwrt$ make tools/libtool/{clean,compile} V=s
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/scripts/config'
make[1]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
cd "/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt"; git log --format=%h -1 toolchain > /home/gith                                                                                                                           ub-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check
cmp -s /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check /home/github-worker/actions-r                                                                                                                           unner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check || { \
        rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/target-aarch64_cortex-a53_                                                                                                                           musl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl /ho                                                                                                                           me/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_mu                                                                                                                           sl /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.                                                                                                                           0_musl; \
        mkdir -p /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cort                                                                                                                           ex-a53_gcc-13.3.0_musl/stamp; \
        mv /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tmp/.ver_check /home/github-worker/actio                                                                                                                           ns-runner/_work/BananaWRT/immortalwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/stamp/.ver_check; \
}
make[2]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool'
CFLAGS="-O2 -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CPPFLAGS="                                                                                                                           -I/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " CXXFLAGS="-O2 -I/home/                                                                                                                           github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/include " LDFLAGS="-L/home/github-worker                                                                                                                           /actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/lib " make  -C /home/github-worker/actions-runner/_wo                                                                                                                           rk/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.  uninstall
make[3]: Entering directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool                                                                                                                           -2.4.7'
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
make[3]: *** [GNUmakefile:108: abort-due-to-no-makefile] Error 1
make[3]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-                                                                                                                           2.4.7'
make[2]: [Makefile:60: host-clean-build] Error 2 (ignored)
(cd /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/share/aclocal/ && rm -f libtoo                                                                                                                           l.m4 ltdl.m4 lt~obsolete.m4 ltoptions.m4 ltsugar.m4 ltversion.m4)
rm -rf /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7 /home/github-wo                                                                                                                           rker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.built
[... cut ...]
make[2]: *** [Makefile:66: /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/build_dir/host/libtool-2.4.7/.prepared0353fec309512c82c80f727c8b083eca_18f1e190c5d53547fed41a3eaa76e9e9] Error 1
make[2]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/tools/libtool'
time: tools/libtool/compile#0.21#0.16#0.66
    ERROR: tools/libtool failed to build.
make[1]: *** [tools/Makefile:233: tools/libtool/compile] Error 1
make[1]: Leaving directory '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt'
make: *** [/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/include/toplevel.mk:233: tools/libtool/compile] Error 2
```

1715173329, 2025-01-27:
please apply this patch to OpenWrt buildroot

SuperKali, 2025-01-27:
> please apply this patch to OpenWrt buildroot

Sorry, I'm pretty new on these things, inside the immortalwrt folder?

1715173329, 2025-01-27:
> Sorry, I'm pretty new on these things, inside the immortalwrt folder?

Yes

1715173329, 2025-01-27:
So this is the whole detection logic:
```
func_get_version ()
{
    $debug_cmd

    _G_app=$1

    # Rather than uncomment the sed script in-situ, strip the comments
    # programatically before passing the result to $SED for evaluation.
    sed_get_version=`$ECHO '# extract version within line
          s|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          # extract version at start of line
          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          # the following essentially does s|5.005|5.5|
          s|\.0*\([1-9]\)|.\1|g
          p
          q' \
    |$SED '/^[	 ]*#.*$/d'`

    func_tool_version_output $_G_app >/dev/null
    _G_status=$?

    test 0 -ne "$_G_status" \
      || $_G_app --version 2>&1 |$SED -n "$sed_get_version"

    (exit $_G_status)
}

func_check_versions ()
{
    $debug_cmd

    func_check_versions_result=:

    while test $# -gt 0; do
      _G_app=$1; shift
      _G_reqver=$1; shift
      _G_url=$1; shift

      # Diagnose bad buildreq formatting.
      case $_G_url in
      [a-z]*://*) ;; # looks like a url
      *) func_fatal_error "\
'$_G_url' from the buildreq table in
'bootstrap.conf' does not look like the URL for downloading
$_G_app. Please ensure that buildreq is a strict newline
delimited list of triples; 'program min-version url'."
        ;;
      esac

      # Honor $APP variables ($TAR, $AUTOCONF, etc.)
      _G_appvar=`echo $_G_app |tr '[a-z]' '[A-Z]'`
      test TAR = "$_G_appvar" && _G_appvar=AMTAR
      eval "_G_app=\${$_G_appvar-$_G_app}"

      # Fail if no version specified, but the program can't be found.
      if test x- = "x$_G_reqver"; then
        func_check_tool $_G_app
	if test -z "$func_check_tool_result"; then
	  func_error "Prerequisite '$_G_app' not not found. Please install it, or
'export $_G_appvar=/path/to/$_G_app'."
          func_check_versions_result=false
	else
	  func_verbose "found '$func_check_tool_result' for $_G_appvar."
	fi
      else
        _G_instver=`func_get_version $_G_app`

        # Fail if --version didn't work.
        if test -z "$_G_instver"; then
          func_error "Prerequisite '$_G_app' not found. Please install it, or
'export $_G_appvar=/path/to/$_G_app'."
          func_check_versions_result=false

        # Fail if a newer version than what we have is required.
        else
          func_verbose "found '$_G_app' version $_G_instver."

	  case $_G_reqver in
            =*)
              # If $buildreq version starts with '=', version must
	      # match the installed program exactly.
	      test "x$_G_reqver" = "x=$_G_instver" || {
	        func_error "\
  '$_G_app' version == $_G_instver is too old
  'exactly $_G_app-$_G_reqver is required"
                func_check_versions_result=false
              }
	      ;;
	    *)
              # Otherwise, anything that is not older is a match.
              func_lt_ver "$_G_reqver" "$_G_instver" || {
                func_error "\
  '$_G_app' version == $_G_instver is too old
  '$_G_app' version >= $_G_reqver is required"
            func_check_versions_result=false
              }
	      ;;
	  esac
        fi
      fi
    done
}
```

there could be something wrong with `test -x ...` or `sed` command.

SuperKali, 2025-01-27:
i put the patch here:

[image]

1715173329, 2025-01-27:
Please apply the patch with `patch -p1 < debug-libtool.patch`

1715173329, 2025-01-27:
the log shows sed failed to extract the autoconf version:
```
+ func_tool_version_output /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf
+ set -x
+ _G_cmd=/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf
+ _G_fatal_error_msg=
+ _G_status=0
+ test 0 -ne 0
+ exit 0
+ _G_status=0
+ test 0 -ne 0
+ /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf --version
+ /home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/sed -n           s|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          s|\.0*\([1-9]\)|.\1|g
          p
          q
+ exit 0
+ _G_instver=
+ test -z 
+ func_error Prerequisite '/home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf' not found. Please install it, or
'export AUTOCONF=/path/to//home/github-worker/actions-runner/_work/BananaWRT/immortalwrt/staging_dir/host/bin/autoconf'.
+ set -x
+ func_require_term_colors
+ set -x
+ test -t 1
```

now check the version of sed:
```
staging_dir/host/bin/sed --version
```

and try to run the same command to extract autoconf version:
```
staging_dir/host/bin/autoconf --version | staging_dir/host/bin/sed -n 's|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          s|\.0*\([1-9]\)|.\1|g
          p
          q'
```

SuperKali, 2025-01-27:
```bash
➜  immortalwrt git:(v24.10.0-rc4) ✗ staging_dir/host/bin/sed --version
staging_dir/host/bin/sed (GNU sed) 4.9
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Jay Fenlason, Tom Lord, Ken Pizzini,
Paolo Bonzini, Jim Meyering, and Assaf Gordon.

This sed program was built without SELinux support.

GNU sed home page: <https://www.gnu.org/software/sed/>.
General help using GNU software: <https://www.gnu.org/gethelp/>.
E-mail bug reports to: <bug-sed@gnu.org>.
➜  immortalwrt git:(v24.10.0-rc4) ✗ staging_dir/host/bin/autoconf --version | staging_dir/host/bin/sed -n 's|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          s|\.0*\([1-9]\)|.\1|g
          p
          q'
➜  immortalwrt git:(v24.10.0-rc4) ✗

```

SuperKali, 2025-01-27:
- workflow on arm64: https://github.com/SuperKali/BananaWRT/actions/runs/12986004010/job/36212047502
- workflow on x86: https://github.com/SuperKali/BananaWRT/actions/runs/12986121631/job/36212289421

both fails with rc4, but randomically fails, i didn't understand why, but this issue on rc3 not occur.

1715173329, 2025-01-27:
and what if using host sed?

```
staging_dir/host/bin/autoconf --version | /usr/bin/sed -n 's|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          s|\.0*\([1-9]\)|.\1|g
          p
          q'
```

SuperKali, 2025-01-27:
```bash
github-worker@it-manu-worker-x86:~/actions-runner/_work/BananaWRT/immortalwrt$ staging_dir/host/bin/autoconf --version | /usr/bin/sed -n 's|.*[v ]\{1,\}\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          s|^\([0-9]\{1,\}\.[.a-z0-9-]*\).*|\1|
          t done

          d

          :done
          s|\.0*\([1-9]\)|.\1|g
          p
          q'

```

1715173329, 2025-01-27:
```
autoconf (GNU Autoconf) nightly
```

so this is the cause.

SuperKali, 2025-01-27:
yes, it's fixable?

1715173329, 2025-01-27:
main problem is how you get this... it's not even a part of the source code

SuperKali, 2025-01-27:
I’m a idiot.. maybe this is the issue

SuperKali, 2025-01-27:
The error was related to the RELEASE variable, sorry for wasting your time, but I did not know that this variable was used elsewhere.

Thank you very much for your support

closed 2025-01-27 by SuperKali