immortalwrt/immortalwrt #1700: JDCloud RE-CP-03可以获取到pd/60但是下面的设备没有ipv6

aYangQT, 2025-02-07:
### Describe the bug

[内核日志.txt](https://github.com/user-attachments/files/18706889/default.txt)
[系统日志.txt](https://github.com/user-attachments/files/18706890/default.txt)

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

JDCloud RE-CP-03

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

aYangQT, 2025-02-07:
[内核日志.txt](https://github.com/user-attachments/files/18706964/default.txt)
[系统日志.txt](https://github.com/user-attachments/files/18706963/default.txt)

aYangQT, 2025-02-07:
[内核日志.txt](https://github.com/user-attachments/files/18706978/default.txt)
[系统日志.txt](https://github.com/user-attachments/files/18706977/default.txt)

aYangQT, 2025-02-08:
23.05.4下面设备有ipv6，24.10.0没有，路由器有ipv6,前缀/60，应该可以往下面分，不知道哪里的问题

aYangQT, 2025-02-08:
> [@aYangQT](https://github.com/aYangQT) Wireshark抓包，看下ICMPv6的信息

回去试试

aYangQT, 2025-02-08:
> 23.05.4下面设备有ipv6，24.10.0没有，路由器有ipv6,前缀/60，应该可以往下面分，不知道哪里的问题

只有内网ipv6，没有公网ipv6

xiaoyangdkj, 2025-02-08:
> 23.05.4下面设备有ipv6，24.10.0没有，路由器有ipv6,前缀/60，应该可以往下面分，不知道哪里的问题

[image]

如果使用的是从[downloads.immortalwrt.org](https://downloads.immortalwrt.org/releases/24.10.0/targets/mediatek/filogic/)下载的[固件](https://downloads.immortalwrt.org/releases/24.10.0/targets/mediatek/filogic/immortalwrt-24.10.0-mediatek-filogic-jdcloud_re-cp-03-squashfs-sysupgrade.itb)的话，请确保【RA 服务】（24.10-rc4的时候默认是禁用状态）

aYangQT, 2025-02-08:
> > 23.05.4下面设备有ipv6，24.10.0没有，路由器有ipv6,前缀/60，应该可以往下面分，不知道哪里的问题
> 
> [image]
> 
> 如果使用的是从[downloads.immortalwrt.org](https://downloads.immortalwrt.org/releases/24.10.0/targets/mediatek/filogic/)下载的[固件](https://downloads.immortalwrt.org/releases/24.10.0/targets/mediatek/filogic/immortalwrt-24.10.0-mediatek-filogic-jdcloud_re-cp-03-squashfs-sysupgrade.itb)的话，请确保【RA 服务】已启用（24.10-rc4的时候默认是禁用状态）

[image] 看了下昨天的截图，确实是关着的，晚上回去开启试试

aYangQT, 2025-02-08:
我是路由器拨号，不知道有影响吗

1715173329, 2025-02-08:
12bd0c55114816e07e7e34fcc709089f6c79f6b6

closed 2025-02-08 by 1715173329

aYangQT, 2025-02-11:
@1715173329 大佬这个版本，我使用这几天，发现几个小bug，1，2.4G设置40mhz频宽后，使用信道检测还是20mhz. 2,gtp给根分区分区110G后，显示默认占用接近1G，不知道有没有什么影响。

[image]

[image]

aYangQT, 2025-02-11:
[image]
都是小问题，目前不影响使用

aYangQT, 2025-02-19:
> [@1715173329](https://github.com/1715173329) 大佬，我用wol网络唤醒功能，之前电脑关机后，网卡协商速率是10Mbs，现在好像是100Mbps，不知道哪里的问题，不过不影响使用

路由器重启后，好像又变成10mbps了

everything411, 2025-02-19:
默认占用接近1G是文件系统的元数据和保留区域，2.4G跑不到40Mhz可能是因为周围的其他路由器太多，找不到空闲信道。WoL时的网卡速率取决于网卡型号和驱动，只要功能正常就可以了，不一定统一

aYangQT, 2025-02-20:
> 默认占用接近1G是文件系统的元数据和保留区域，2.4G跑不到40Mhz可能是因为周围的其他路由器太多，找不到空闲信道。WoL时的网卡速率取决于网卡型号和驱动，只要功能正常就可以了，不一定统一

基本同意，不过wifi分析就是20mhz，之前版本是40mhz，应该不是周围设备影响，wol网卡速率偶现100Mhz，近期重启都是10mhz,影响不大，近期使用zerotier感觉远程经常失败，设备都在国内，感觉zerotier还有点问题，可能要后面版本优化了