immortalwrt/immortalwrt #1701: 部分 ramips/mt7621 设备的 802.11k/v 设置消失导致 AP 无法启动

iBug, 2025-02-08:
### Describe the bug

从 23.05.4 保留配置升级 24.10.0 后 AP 无法启动，日志节选如下：

```text
Sat Feb  8 17:44:41 2025 daemon.notice netifd: radio0 (22929): sh: acs_survey: out of range
Sat Feb  8 17:44:41 2025 daemon.notice netifd: radio0 (22929): sh: acs_survey: out of range
Sat Feb  8 17:44:41 2025 daemon.notice wpa_supplicant[1670]: Set new config for phy phy0
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: Restart interface for phy phy0
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: Remove interface 'phy0'
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Sat Feb  8 17:44:41 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Sat Feb  8 17:44:41 2025 daemon.notice hostapd: nl80211: Failed to remove interface phy0-ap0 from bridge br-lan: Invalid argument
Sat Feb  8 17:44:42 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CN ieee80211d=1 hw_mode=g vendor_vht=1 supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 stationary_ap=1 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=0 vht_oper_centr_freq_seg0_idx= vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3] channel=acs_survey  interface=phy0-ap0 bssid=c8:bf:4c:01:31:5c ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 ocv=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=diansanlou420 wpa_psk_file=/var/run/hosta
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 50: unknown configuration item 'eap_server'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 51: unknown configuration item 'wps_state'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 52: unknown configuration item 'device_type'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 53: unknown configuration item 'device_name'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 54: unknown configuration item 'manufacturer'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 55: unknown configuration item 'config_methods'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 56: unknown configuration item 'wps_independent'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Line 61: unknown configuration item 'bss_transition'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: 8 errors found in configuration file '<inline>'
Sat Feb  8 17:44:42 2025 daemon.err hostapd: Failed to set up interface with data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CN ieee80211d=1 hw_mode=g vendor_vht=1 supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 stationary_ap=1 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=0 vht_oper_centr_freq_seg0_idx= vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3] channel=acs_survey  interface=phy0-ap0 bssid=c8:bf:4c:01:31:5c ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 ocv=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=diansanlou420 wpa_psk_file=/v
Sat Feb  8 17:44:42 2025 daemon.notice hostapd: hostapd.add_iface failed for phy phy0 ifname=phy0-ap0
Sat Feb  8 17:44:42 2025 daemon.notice netifd: Wireless device 'radio0' is now up
Sat Feb  8 17:44:44 2025 kern.info kernel: [82019.525196] br-lan: left allmulticast mode
Sat Feb  8 17:44:44 2025 kern.info kernel: [82019.683634] br-lan: entered allmulticast mode
Sat Feb  8 17:44:44 2025 kern.info kernel: [82019.689123] mt7530-mdio mdio-bus:1f wan: left allmulticast mode
Sat Feb  8 17:44:44 2025 kern.info kernel: [82019.803670] mt7530-mdio mdio-bus:1f wan: entered allmulticast mode
```

手动编辑 `/etc/config/wireless` 删除 `wps_pushbutton` 和 `bss_transition` 之后可以正常启动 AP，并且设置界面 WPS 按钮选项也消失了（虽然这个设备上并没有按钮）：

> [image]

漫游设置里 k 和 v 的选项也没了，只剩下 r 了：

> [image]

手上三台不同的 MT7621 的 Wi-Fi 5 路由器均有此现象，不知道是不是 24.10 的预期变化。

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

xiaomi_mi-router-4a-gigabit-v2 / xiaomi_mi-router-ac2100 / xiaomi_redmi-router-ac2100

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-02-08:
如果不是用值守式系统更新升级的，需要手动卸载 wpad-basic-openssl 并安装 wpad-openssl。

iBug, 2025-02-08:
感谢提醒，检查发现 24.10 的分支少了 4d80ab42d3faa1edf0dd54bfd840e120c86061f4，所以变回 OpenWrt 默认的 wpad basic 了。

请问是否考虑 cherry-pick 上述 commit，或者另有其他想法？

1715173329, 2025-02-08:
因为 24.10 内核体积进一步增大，所以有意切换成 wpad-basic 以减少固件大小。不过刚刚看了一下实际上也没有小于 8MB 的固件被生成，所以用 basic 大概也没什么意义了。会在下个版本切回去，感谢提醒。

closed 2025-02-08 by 1715173329