immortalwrt/immortalwrt #1706: immortalwrt编译问题

xiaomayi755259, 2025-02-10:
### Describe the bug

编译结尾出现WARNING: Applying padding in /home/dh/immortalwrt/bin/targets/x86/64/packages/Packages to workaround usign SHA-512 bug! 怎么回事？怎么解决？23.05.4正常

### ImmortalWrt version

24.10.0

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

24.10.0

### Device

x86/64

### Image kind

Official downloaded image

### Steps to reproduce

编译结尾出现WARNING: Applying padding in /home/dh/immortalwrt/bin/targets/x86/64/packages/Packages to workaround usign SHA-512 bug! 怎么回事？怎么解决？23.05.4正常

### Actual behaviour

编译结尾出现WARNING: Applying padding in /home/dh/immortalwrt/bin/targets/x86/64/packages/Packages to workaround usign SHA-512 bug! 怎么回事？怎么解决？23.05.4正常

### Expected behaviour

编译结尾出现WARNING: Applying padding in /home/dh/immortalwrt/bin/targets/x86/64/packages/Packages to workaround usign SHA-512 bug! 怎么回事？怎么解决？23.05.4正常

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-02-10:
你觉得这是 expected behavior，那就是。

closed 2025-02-10 by 1715173329

initzhu, 2025-02-10:
没炸就不管 能启动且能用就完事

xiaomayi755259, 2025-02-10:
谢谢大佬啊，不懂，小白，只是跟着教程一步一步编译过来的。结果能用。

xiaomayi755259, 2025-02-10:
> 你觉得这是预期的行为，那就是。

谢谢大佬啊，不懂，小白，只要按照教程一步一步编译过来的。结果能用。

xiaomayi755259, 2025-02-10:
> 没炸就不管能启动且能用就完事

嗯嗯，结果能用。