immortalwrt/immortalwrt #1723: HC5962 最新的24.10的MAC不正常

xieqiyou, 2025-02-18:
### Describe the bug

23.05.4时候，路由的lan的mac是读取bdinfo的。是正常的。
在24.10时候，lan和wan的不知道为什么不读取bdinfo的。重置一次就变一次路由的lan口和wan口mac。wifi的倒是正常的。

### ImmortalWrt version

24.10

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

ramips/mt7621

### Device

HC 5962

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-03-01:
估计是 551e04f3c9c0 造成的，执行下下面的命令看看
```
. /lib/functions/system.sh
mtd_get_mac_ascii bdinfo "Vfac_mac "; echo
mtd_get_mac_ascii bdinfo "Vfac_mac"; echo
```

xieqiyou, 2025-03-02:
> 估计是 [551e04f](https://github.com/immortalwrt/immortalwrt/commit/551e04f3c9c063d885ca7a91db36f27f684a3a5c) 造成的，执行下下面的命令看看
> 
> ```
> . /lib/functions/system.sh
> mtd_get_mac_ascii bdinfo "Vfac_mac "; echo
> mtd_get_mac_ascii bdinfo "Vfac_mac"; echo
> ```
 对。就相差一个空格，后面的能正常获取到MAC，前面的不行。前面的

[image]

closed 2025-03-21 by 1715173329