immortalwrt/immortalwrt #1726: *** [/workdir/openwrt/include/toplevel.mk:233: world] Error 2

woni928, 2025-02-20:
### Describe the bug

昨天下午编译正常，今天一大早开始就出错了。
请知道问题的同学帮忙解答下，感谢！

### ImmortalWrt version

23.05

### ImmortalWrt release

23.05

### ImmortalWrt target/subtarget

23.05

### Device

N1

### Image kind

Official downloaded image

### Steps to reproduce

到现在为止，只要编译就报错。

### Actual behaviour

[image]

### Expected behaviour

系统编译正常

### Additional info

完整的日志

[未命名 2.txt](https://github.com/user-attachments/files/18884223/2.txt)

### Diffconfig

```text
CONFIG_TARGET_armsr=y
CONFIG_TARGET_armsr_armv8=y
CONFIG_TARGET_armsr_armv8_DEVICE_generic=y
CONFIG_TARGET_ROOTFS_INITRAMFS=n
CONFIG_TARGET_ROOTFS_CPIOGZ=n
CONFIG_TARGET_ROOTFS_EXT4FS=n
CONFIG_TARGET_ROOTFS_SQUASHFS=n

# ipv6
CONFIG_PACKAGE_ipv6helper=y

# kmod
CONFIG_PACKAGE_kmod-amazon-ena=n
CONFIG_PACKAGE_kmod-ata-ahci=y
CONFIG_PACKAGE_kmod-atlantic=n
CONFIG_PACKAGE_kmod-bcmgenet=n
CONFIG_PACKAGE_kmod-dwmac-imx=n
CONFIG_PACKAGE_kmod-dwmac-rockchip=n
CONFIG_PACKAGE_kmod-dwmac-sun8i=n
CONFIG_PACKAGE_kmod-e1000e=n
CONFIG_PACKAGE_kmod-fsl-dpaa1-net=n
CONFIG_PACKAGE_kmod-fsl-dpaa2-net=n
CONFIG_PACKAGE_kmod-fsl-enetc-net=n
CONFIG_PACKAGE_kmod-fsl-fec=n
CONFIG_PACKAGE_kmod-gpio-pca953x=n
CONFIG_PACKAGE_kmod-i2c-mux-pca954x=n
CONFIG_PACKAGE_kmod-mvneta=n
CONFIG_PACKAGE_kmod-mvpp2=n
CONFIG_PACKAGE_kmod-octeontx2-net=n
CONFIG_PACKAGE_kmod-phy-aquantia=n
CONFIG_PACKAGE_kmod-phy-broadcom=n
CONFIG_PACKAGE_kmod-phy-marvell=n
CONFIG_PACKAGE_kmod-phy-marvell-10g=n
CONFIG_PACKAGE_kmod-rtc-rx8025=n
CONFIG_PACKAGE_kmod-sfp=n
CONFIG_PACKAGE_kmod-thunderx-net=n
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-vmxnet3=n
CONFIG_PACKAGE_kmod-wdt-sp805=n

# luci								   
CONFIG_PACKAGE_luci-app-filetransfer=n
#CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_PACKAGE_luci-app-nlbwmon=y
CONFIG_PACKAGE_luci-app-wechatpush=y

# themes
#CONFIG_PACKAGE_luci-theme-argon=y

# other packages
CONFIG_PACKAGE_autocore=y
CONFIG_PACKAGE_fstrim=y
CONFIG_PACKAGE_ppp=n
CONFIG_PACKAGE_zoneinfo-asia=y

# necessary
CONFIG_BTRFS_PROGS_ZSTD=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_bsdtar=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_PACKAGE_chattr=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_gawk=y
CONFIG_PACKAGE_getopt=y
CONFIG_PACKAGE_lsattr=y
CONFIG_PACKAGE_perl=y
CONFIG_PACKAGE_perlbase-getopt=y
CONFIG_PACKAGE_perlbase-unicode=y
CONFIG_PACKAGE_perl-http-date=y
CONFIG_PACKAGE_pigz=y
CONFIG_PACKAGE_tar=y
CONFIG_PACKAGE_xfs-fsck=y
CONFIG_PACKAGE_xfs-mkfs=y

# remove wifi
CONFIG_PACKAGE_hostapd-common=n
CONFIG_PACKAGE_iw=n
CONFIG_PACKAGE_kmod-brcmfmac=n
CONFIG_PACKAGE_kmod-brcmutil=n
CONFIG_PACKAGE_kmod-cfg80211=n
CONFIG_PACKAGE_kmod-mac80211=n
CONFIG_PACKAGE_wpa-cli=n
CONFIG_PACKAGE_wpad-basic=n
```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-02-20:
引入非官方源码造成的问题自行负责。

closed 2025-02-20 by 1715173329

woni928, 2025-02-20:
> 引入非官方源码造成的问题由自己负责。
可以请教一下怎么解决吗？

1715173329, 2025-02-20:
移除所有非官方支持的软件包再重新编译即可。

woni928, 2025-02-21:
> 移除所有非官方支持的软件包再重新编译即可。

感谢，第三方已经修复